osrf/homebrew-simulation #1800: Rebuild ignition-common*, gazebo* after ffmpeg 5.0

scpeters, 2022-02-03:
A new major version of ffmpeg (5.0) was just merged in https://github.com/Homebrew/homebrew-core/pull/94423, which has broken our bottles.

* [ ] Disable all broken bottles
* [ ] Rebuild ignition-common1 up to gazebo9
* [ ] Rebuild ignition-common3 up to citadel, gazebo11
* [ ] Rebuild ignition-common4 up to edifice, fortress

carlocab, 2022-02-03:
Not hugely important, but ffmpeg 5.0 was actually merged in https://github.com/Homebrew/homebrew-core/pull/93167 (in case someone wants to look it up later).

scpeters, 2022-02-04:
> Not hugely important, but ffmpeg 5.0 was actually merged in [Homebrew/homebrew-core#93167](https://github.com/Homebrew/homebrew-core/pull/93167) (in case someone wants to look it up later).

whoops, thanks @carlocab

traversaro, 2022-02-14:
Just a small note to anyone  trying to compile downstream projects using the formulas that use `ffmpeg@4`. Due to an upstream gazebo issue (more and less tracked in https://github.com/osrf/gazebo/issues/2580) the ffmpeg include directories (that should be passed to downstream as they are included in public headers) are not passed to downstream projects, so you may get errors such as:
~~~
2022-02-09T03:04:59.4240130Z [10/106] Linking CXX shared library lib/libgazebo_yarp_rpc_worldinterface.4.1.1.dylib
2022-02-09T03:04:59.4340680Z [11/106] Creating library symlink lib/libgazebo_yarp_rpc_worldinterface.dylib
2022-02-09T03:04:59.4441780Z [12/106] Building CXX object libraries/singleton/CMakeFiles/gazebo_yarp_singleton.dir/src/ConfHelpers.cc.o
2022-02-09T03:04:59.4542640Z FAILED: libraries/singleton/CMakeFiles/gazebo_yarp_singleton.dir/src/ConfHelpers.cc.o 
2022-02-09T03:04:59.4649190Z /Applications/Xcode_12.4.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -DBOOST_ALL_NO_LIB -DBOOST_DATE_TIME_DYN_LINK -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_TEST_DYN_LINK -DBOOST_THREAD_DYN_LINK -DLIBBULLET_VERSION=3.20 -DLIBBULLET_VERSION_GT_282 -Dgazebo_yarp_singleton_EXPORTS -I/Users/runner/work/robotology-superbuild/robotology-superbuild/src/GazeboYARPPlugins/libraries/singleton/include/GazeboYarpPlugins -I/Users/runner/work/robotology-superbuild/robotology-superbuild/src/GazeboYARPPlugins/libraries/singleton/include -isystem /usr/local/include -isystem /usr/local/include/gazebo-11 -isystem /usr/local/Cellar/bullet/3.21/include/bullet -isystem /usr/local/Cellar/bullet/3.21/include -isystem /usr/local/include/simbody -isystem /usr/local/Cellar/sdformat9/9.7.0/include/sdformat-9.7 -isystem /usr/local/include/ignition/math6 -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/include/OGRE -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/include -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/include/OGRE/Terrain -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/include/OGRE/Paging -isystem /usr/local/include/ignition/transport8 -isystem /usr/local/include/ignition/msgs5 -isystem /usr/local/include/ignition/common3 -isystem /usr/local/include/ignition/fuel_tools4 -isystem /Users/runner/work/robotology-superbuild/robotology-superbuild/build/install/include -isystem /usr/local/include/eigen3 -isystem /usr/local/Cellar/fcl/0.7.0/include -isystem /usr/local/Cellar/assimp/5.2.1/include -isystem /usr/local/include/sdformat-9.7/sdf/.. -isystem /usr/local/include/ignition/cmake2 -O3 -DNDEBUG -isysroot /Applications/Xcode_12.4.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.1.sdk -mmacosx-version-min=10.15 -fPIC -F/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/System/Library/Frameworks  -std=gnu++17 -MD -MT libraries/singleton/CMakeFiles/gazebo_yarp_singleton.dir/src/ConfHelpers.cc.o -MF libraries/singleton/CMakeFiles/gazebo_yarp_singleton.dir/src/ConfHelpers.cc.o.d -o libraries/singleton/CMakeFiles/gazebo_yarp_singleton.dir/src/ConfHelpers.cc.o -c /Users/runner/work/robotology-superbuild/robotology-superbuild/src/GazeboYARPPlugins/libraries/singleton/src/ConfHelpers.cc
2022-02-09T03:04:59.4653440Z In file included from /Users/runner/work/robotology-superbuild/robotology-superbuild/src/GazeboYARPPlugins/libraries/singleton/src/ConfHelpers.cc:13:
2022-02-09T03:04:59.4754170Z In file included from /usr/local/include/gazebo-11/gazebo/common/common.hh:53:
2022-02-09T03:04:59.4855370Z /usr/local/include/gazebo-11/gazebo/common/ffmpeg_inc.h:32:10: fatal error: 'libavcodec/avcodec.h' file not found
2022-02-09T03:04:59.4955600Z #include <libavcodec/avcodec.h>
2022-02-09T03:04:59.5056670Z          ^~~~~~~~~~~~~~~~~~~~~~
2022-02-09T03:04:59.5157140Z 1 error generated.
2022-02-09T03:04:59.5258120Z [13/106] Building CXX object libraries/singleton/CMakeFiles/gazebo_yarp_singleton.dir/src/Handler.cc.o
2022-02-09T03:04:59.5358430Z [14/106] Building CXX object plugins/clock/CMakeFiles/gazebo_yarp_clock.dir/src/Clock.cc.o
2022-02-09T03:04:59.5459660Z [15/106] Building CXX object plugins/camera/CMakeFiles/gazebo_yarp_camera.dir/src/Camera.cc.o
2022-02-09T03:04:59.5559980Z [16/106] Building CXX object plugins/camera/CMakeFiles/gazebo_yarp_camera.dir/src/CameraDriver.cpp.o
2022-02-09T03:04:59.5661260Z ninja: build stopped: subcommand failed.
~~~

To solve them, you need to manually add to the `CPATH` environment variable the include directories of ffmpeg, exactly as the gazebo formula actually does in https://github.com/osrf/homebrew-simulation/blob/bde5964693982a47de8a35be18d3045334fcb3c6/Formula/gazebo11.rb#L105 .

closed 2022-02-25 by scpeters