osrf/homebrew-simulation #1818: Ignition launch 4/5 cannot be installed in M1 mac big sur

tmefmd, 2022-02-22:
I could not install the ignition launch 4/5 on my M1 mac now.

https://gist.github.com/51809fd423d3704ad62d4fb7e36b164b

I think the key error/warning messages are below associated with doxygen. 

/CMakeFiles/ignition-launch4.dir/ign.cc.o -MF CMakeFiles/ignition-launch4.dir/ign.cc.o.d -o CMakeFiles/ignition-launch4.dir/ign.cc.o -c /tmp/ignition-launch4-20220221-179-1i43bwe/ignition-launch-4.1.0/src/ign.cc
warning: Tag 'TCL_SUBST' at line 237 of file '/tmp/ignition-launch4-20220221-179-1i43bwe/ignition-launch-4.1.0/api_tagfile.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'COLS_IN_ALPHA_INDEX' at line 1044 of file '/tmp/ignition-launch4-20220221-179-1i43bwe/ignition-launch-4.1.0/api_tagfile.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'LATEX_SOURCE_CODE' at line 1757 of file '/tmp/ignition-launch4-20220221-179-1i43bwe/ignition-launch-4.1.0/api_tagfile.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'RTF_SOURCE_CODE' at line 1839 of file '/tmp/ignition-launch4-20220221-179-1i43bwe/ignition-launch-4.1.0/api_tagfile.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'DOCBOOK_PROGRAMLISTING' at line 1937 of file '/tmp/ignition-launch4-20220221-179-1i43bwe/ignition-launch-4.1.0/api_tagfile.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'CLASS_DIAGRAMS' at line 2133 of file '/tmp/ignition-launch4-20220221-179-1i43bwe/ignition-launch-4.1.0/api_tagfile.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
In file included from /tmp/ignition-launch4-20220221-179-1i43bwe/ignition-launch-4.1.0/src/Manager.cc:51:
/tmp/ignition-launch4-20220221-179-1i43bwe/ignition-launch-4.1.0/src/vendor/backward.hpp:3753:57: error: member reference type 'struct __darwin_mcontext64 *' is a pointer; did you mean to use '->'?
                error_addr = reinterpret_cast<void*>(uctx->uc_mcontext.pc);
                                                     ~~~~~~~~~~~~~~~~~^
                                                                      ->
/tmp/ignition-launch4-20220221-179-1i43bwe/ignition-launch-4.1.0/src/vendor/backward.hpp:3753:58: error: no member named 'pc' in '__darwin_mcontext64'
                error_addr = reinterpret_cast<void*>(uctx->uc_mcontext.pc);
                                                     ~~~~~~~~~~~~~~~~~ ^
2 errors generated.
make[2]: *** [src/CMakeFiles/ignition-launch4.dir/Manager.cc.o] Error 1
make[1]: *** [src/CMakeFiles/ignition-launch4.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/opt/homebrew/bin/doxygen /tmp/ignition-launch4-20220221-179-1i43bwe/ignition-launch-4.1.0/api.dox
warning: Tag 'TCL_SUBST' at line 237 of file '/tmp/ignition-launch4-20220221-179-1i43bwe/ignition-launch-4.1.0/api.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'COLS_IN_ALPHA_INDEX' at line 1044 of file '/tmp/ignition-launch4-20220221-179-1i43bwe/ignition-launch-4.1.0/api.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'LATEX_SOURCE_CODE' at line 1757 of file '/tmp/ignition-launch4-20220221-179-1i43bwe/ignition-launch-4.1.0/api.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'RTF_SOURCE_CODE' at line 1839 of file '/tmp/ignition-launch4-20220221-179-1i43bwe/ignition-launch-4.1.0/api.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'DOCBOOK_PROGRAMLISTING' at line 1937 of file '/tmp/ignition-launch4-20220221-179-1i43bwe/ignition-launch-4.1.0/api.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'CLASS_DIAGRAMS' at line 2133 of file '/tmp/ignition-launch4-20220221-179-1i43bwe/ignition-launch-4.1.0/api.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
[ 25%] Built target doc
make: *** [all] Error 2

Any help will be appreciated!

iche033, 2022-06-21:
a fix targeted at ign-launch6 for Garden was merged: https://github.com/gazebosim/gz-launch/pull/161

jasonbyun8, 2022-06-21:
Yes, @iche033's fix worked for me, I will close this until further issues are coming up

closed 2022-06-21 by jasonbyun8