osrf/homebrew-simulation #1872: Pybind11 CMake error when installing ignition-math6 with homebrew

varunagrawal, 2022-04-12:
When running `brew install ignition-math6`, the build fails with the error message:

```
CMake Error at /home/linuxbrew/.linuxbrew/share/cmake/pybind11/pybind11NewTools.cmake:207 (python3_add_library):
  Unknown CMake command "python3_add_library".
Call Stack (most recent call first):
  src/python_pybind11/CMakeLists.txt:10 (pybind11_add_module)
```

The full build log is
```                                                                                                                                    6s
==> Downloading https://osrf-distributions.s3.amazonaws.com/ign-math/releases/ignition-math6-6.10.0.tar.bz2
Already downloaded: /home/varun/.cache/Homebrew/downloads/1aa43810b9fe3dcdd09d760e2894acea7540299272a1193754ee959a2767d70e--ignition-math6-6.10.0.tar.bz2
==> Verifying checksum for '1aa43810b9fe3dcdd09d760e2894acea7540299272a1193754ee959a2767d70e--ignition-math6-6.10.0.tar.bz2'
==> Installing ignition-math6 from osrf/simulation
tar --extract --no-same-owner --file /home/varun/.cache/Homebrew/downloads/1aa43810b9fe3dcdd09d760e2894acea7540299272a1193754ee959a2767d70e--ignition-math6-6.10.0.tar.bz2 --directory /tmp/d20220412-1161017-1mv0jhw
cp -pR /tmp/d20220412-1161017-1mv0jhw/ignition-math-6.10.0/. /tmp/ignition-math6-20220412-1161017-18z728/ignition-math-6.10.0
chmod -Rf +w /tmp/d20220412-1161017-1mv0jhw
==> cmake . -DCMAKE_INSTALL_PREFIX=/home/linuxbrew/.linuxbrew/Cellar/ignition-math6/6.10.0 -DCMAKE_INSTALL_LIBDIR=lib -DCMAKE_BUILD_TYPE=Release -DCMAKE_FIND_FRAMEWORK=LAST -DCMAKE_VERBOSE_MAKEFILE=ON -Wno-dev -DBUILD_TESTING=OFF -DBUILD_TESTING=Off -DCMAKE_INSTALL_RPATH='$ORIGIN/../lib'
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/linuxbrew/.linuxbrew/Homebrew/Library/Homebrew/shims/linux/super/gcc-11 - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/linuxbrew/.linuxbrew/Homebrew/Library/Homebrew/shims/linux/super/g++-11 - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- ignition-math6 version 6.10.0
-- Operating system is Linux
-- Found CPack generators: DEB
-- Looking for eigen3 - found

-- Searching for swig - found version 4.0.2.
-- Searching for Ruby - found version 3.1.1.
-- Searching for Python - found version 3.8.10.
-- Performing Test HAS_FLTO
-- Performing Test HAS_FLTO - Success
-- Searching for pybind11 - found version 2.9.2.
-- 
-- Searching for host SSE information
-- SSE2 found
-- SSE3 found
-- SSE4.1 found
-- SSE4.2 found
-- Configuring library: ignition-math6
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
-- Testing is disabled -- skipping UNIT tests
-- Testing is disabled -- skipping UNIT tests
-- Building pybind11 interfaces
CMake Error at /home/linuxbrew/.linuxbrew/share/cmake/pybind11/pybind11NewTools.cmake:207 (python3_add_library):
  Unknown CMake command "python3_add_library".
Call Stack (most recent call first):
  src/python_pybind11/CMakeLists.txt:10 (pybind11_add_module)

-- Configuring incomplete, errors occurred!
See also "/tmp/ignition-math6-20220412-1161017-18z728/ignition-math-6.10.0/CMakeFiles/CMakeOutput.log".

==> Formula
Tap: osrf/simulation
Path: /home/linuxbrew/.linuxbrew/Homebrew/Library/Taps/osrf/homebrew-simulation/Formula/ignition-math6.rb
==> Configuration
HOMEBREW_VERSION: 3.4.6
ORIGIN: https://github.com/Homebrew/brew
HEAD: 56415326bb8c9eccc6e53f3997b4e062b36402af
Last commit: 3 days ago
Core tap ORIGIN: https://github.com/Homebrew/homebrew-core
Core tap HEAD: 37f7375c6df3d7d2e97c1719b5bce137fc782cb3
Core tap last commit: 3 hours ago
Core tap branch: master
HOMEBREW_PREFIX: /home/linuxbrew/.linuxbrew
HOMEBREW_CASK_OPTS: []
HOMEBREW_CORE_GIT_REMOTE: https://github.com/Homebrew/homebrew-core
HOMEBREW_DISPLAY: :1
HOMEBREW_EDITOR: /home/varun/.oh-my-zsh/plugins/emacs/emacsclient.sh
HOMEBREW_MAKE_JOBS: 8
Homebrew Ruby: 2.6.8 => /home/linuxbrew/.linuxbrew/Homebrew/Library/Homebrew/vendor/portable-ruby/2.6.8/bin/ruby
CPU: octa-core 64-bit skylake
Clang: 13.0.0
Git: 2.25.1 => /bin/git
Curl: 7.68.0 => /bin/curl
Kernel: Linux 5.13.0-39-generic x86_64 GNU/Linux
OS: Ubuntu 20.04.3 LTS (focal)
Host glibc: 2.31
/usr/bin/gcc: 9.4.0
/usr/bin/ruby: 2.7.0
glibc: N/A
gcc@5: N/A
xorg: N/A
==> ENV
HOMEBREW_CC: gcc-11
HOMEBREW_CXX: g++-11
MAKEFLAGS: -j8
CMAKE_PREFIX_PATH: /home/linuxbrew/.linuxbrew
PKG_CONFIG_PATH: /home/linuxbrew/.linuxbrew/opt/ncurses/lib/pkgconfig:/home/linuxbrew/.linuxbrew/opt/openssl@1.1/lib/pkgconfig:/home/linuxbrew/.linuxbrew/opt/ignition-cmake2/lib/pkgconfig:/home/linuxbrew/.linuxbrew/opt/libyaml/lib/pkgconfig:/home/linuxbrew/.linuxbrew/opt/readline/lib/pkgconfig:/home/linuxbrew/.linuxbrew/opt/libffi/lib/pkgconfig:/home/linuxbrew/.linuxbrew/opt/zlib/lib/pkgconfig:/home/linuxbrew/.linuxbrew/opt/ruby/lib/pkgconfig:/home/linuxbrew/.linuxbrew/opt/eigen/share/pkgconfig
HOMEBREW_GIT: git
ACLOCAL_PATH: /home/linuxbrew/.linuxbrew/share/aclocal
PATH: /home/linuxbrew/.linuxbrew/Homebrew/Library/Homebrew/shims/linux/super:/home/linuxbrew/.linuxbrew/opt/cmake/bin:/home/linuxbrew/.linuxbrew/opt/doxygen/bin:/home/linuxbrew/.linuxbrew/opt/pybind11/bin:/home/linuxbrew/.linuxbrew/opt/ncurses/bin:/home/linuxbrew/.linuxbrew/opt/openssl@1.1/bin:/home/linuxbrew/.linuxbrew/opt/pkg-config/bin:/home/linuxbrew/.linuxbrew/opt/ruby/bin:/home/linuxbrew/.linuxbrew/opt/binutils/bin:/usr/bin:/bin:/usr/sbin:/sbin:/home/linuxbrew/.linuxbrew/opt/gcc/bin

Error: osrf/simulation/ignition-math6 6.10.0 did not build
Logs:
     /home/varun/.cache/Homebrew/Logs/ignition-math6/00.options.out
     /home/varun/.cache/Homebrew/Logs/ignition-math6/01.cmake
     /home/varun/.cache/Homebrew/Logs/ignition-math6/01.cmake.cc
     /home/varun/.cache/Homebrew/Logs/ignition-math6/CMakeFiles
     /home/varun/.cache/Homebrew/Logs/ignition-math6/CMakeCache.txt
If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
  https://github.com/osrf/homebrew-simulation/issues
```

`brew gist-logs ignition-math6` returns the message `Error: Not found`.

It seems like something is messed up in the CMake config. Can someone help please?

### Environment

```
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.3 LTS
Release:	20.04
Codename:	focal

C++ compiler: g++ 9.4.0
```

varunagrawal, 2022-04-12:
Update: This issue isn't present in the latest version on Github. Looks like a bad version was uploaded to AWS?

abhicmraja, 2022-04-14:
Hi Varun,
Even I'm having the same problem that you had before. I'm tried to install gazebo on my m1 mac today, and I'm running into this error. How do I go about solving this?
Thank you in advance!

abhicmraja, 2022-04-14:
> Update: This issue isn't present in the latest version on Github. Looks like a bad version was uploaded to AWS?

Hi Varun,
Even I'm having the same problem that you had before. I tried to install gazebo on my m1 mac today, and I'm running into this error. How do I go about solving this?
Thank you in advance!

varunagrawal, 2022-04-16:
It seems like this issue has been resolved. If @abhicmraja can confirm, I'll close this.

JPJSea, 2022-04-20:
@varunagrawal I am having this issue today on Mac M1

dlech, 2022-05-01:
I wonder if https://github.com/ignitionrobotics/ign-math/pull/402 will help?

scpeters, 2024-11-22:
I believe this has been fixed; please reopen if the issue persists

closed 2024-11-22 by scpeters