osrf/homebrew-simulation #2069: Rebuild gazebo9/11 bottles for boost 1.80

scpeters, 2022-10-03:
Boost 1.80 was merged recently, and the gazebo bottles need to be rebuild:

* [image] https://build.osrfoundation.org/job/gazebo-install-one_liner-homebrew-amd64/2561/

~~~
dyld: lazy symbol binding failed: Symbol not found: __ZN5boost10filesystem6detail28directory_iterator_constructERNS0_18directory_iteratorERKNS0_4pathEjPNS_6system10error_codeE
  Referenced from: /usr/local/opt/gazebo11/lib/libgazebo_common.11.dylib
  Expected in: /usr/local/opt/boost/lib/libboost_filesystem-mt.dylib

dyld: Symbol not found: __ZN5boost10filesystem6detail28directory_iterator_constructERNS0_18directory_iteratorERKNS0_4pathEjPNS_6system10error_codeE
  Referenced from: /usr/local/opt/gazebo11/lib/libgazebo_common.11.dylib
  Expected in: /usr/local/opt/boost/lib/libboost_filesystem-mt.dylib
~~~

scpeters, 2022-10-03:
disabling broken bottles in https://github.com/osrf/homebrew-simulation/pull/2070

scpeters, 2022-10-03:
bottles were rebuilt in #2071 and #2072

closed 2022-10-03 by scpeters

ecmnet, 2023-01-12:
@scpeters Seems as if this is again necessary for Boost 1.81:

`dyld[22664]: Symbol not found: __ZN5boost10filesystem4path9append_v3ERKS1_
  Referenced from: <4EEE96F2-F741-3745-ABD7-45A919969C02> /usr/local/Cellar/gazebo11/11.12.0_3/lib/libgazebo.11.12.0.dylib
  Expected in:     <E292A764-4926-33BD-B438-FCFB244A96E4> /usr/local/Cellar/boost/1.81.0/lib/libboost_filesystem-mt.dylib
zsh: abort      gazebo
`