osrf/homebrew-simulation #2109: gazebo7 can't load libraries

Ratatinator97, 2022-10-28:
Hello
I just installed on my M1 the `ignition-gazebo7` package and I have the following errors:
```
➜  gz plugin                                                                               
dyld[3862]: Library not loaded: '@rpath/libgz-plugin2-loader.2.dylib'
  Referenced from: '/opt/homebrew/Cellar/gz-plugin2/2.0.0/libexec/gz/plugin2/gz-plugin'
  Reason: tried: '/opt/homebrew/Cellar/gz-plugin2/2.0.0/libexec/gz/plugin2/../lib/libgz-plugin2-loader.2.dylib' (no such file), '/opt/homebrew/Cellar/gz-plugin2/2.0.0/libexec/gz/plugin2/../lib/libgz-plugin2-loader.2.dylib' (no such file), '/usr/local/lib/libgz-plugin2-loader.2.dylib' (no such file), '/usr/lib/libgz-plugin2-loader.2.dylib' (no such file)
Error: Version mismatch. Your configuration file version is
            [2.0.0] but /opt/homebrew/Cellar/gz-plugin2/2.0.0/libexec/gz/plugin2/gz-plugin version is [].
```
Same issue when I run a SDF file with `gz sim -s MySDF.sdf`:
```
Error while loading the library [/opt/homebrew/Cellar/gz-sim7/7.0.0_1/lib/gz-sim-7/plugins/libgz-sim-physics-system.dylib]: dlopen(/opt/homebrew/Cellar/gz-sim7/7.0.0_1/lib/gz-sim-7/plugins/libgz-sim-physics-system.dylib, 0x0085): Library not loaded: '@rpath/libgz-sim7.7.dylib'
  Referenced from: '/opt/homebrew/Cellar/gz-sim7/7.0.0_1/lib/gz-sim-7/plugins/libgz-sim7-physics-system.7.0.0.dylib'
  Reason: tried: '/opt/homebrew/Cellar/gz-sim7/7.0.0_1/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-sim7/7.0.0_1/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-plugin2/2.0.0/lib/../lib/libgz-sim7.7.dylib' (no such file), '/usr/local/lib/libgz-sim7.7.dylib' (no such file), '/usr/lib/libgz-sim7.7.dylib' (no such file)
```
and for `gz-sim-physics-system`, `gz-sim-user-commands-system`, `gz-sim-scene-broadcaster-system`, `gz-sim-particle-emitter-system` ... 

```
➜  ls /opt/homebrew/Cellar/gz-sim7/7.0.0_1/lib/gz-sim-7/plugins/                              
gui                                                       libgz-sim7-joint-position-controller-system.7.dylib
libgz-sim-ackermann-steering-system.dylib                 libgz-sim7-joint-position-controller-system.dylib
libgz-sim-air-pressure-system.dylib                       libgz-sim7-joint-state-publisher-system.7.0.0.dylib
libgz-sim-altimeter-system.dylib                          libgz-sim7-joint-state-publisher-system.7.dylib
libgz-sim-apply-joint-force-system.dylib                  libgz-sim7-joint-state-publisher-system.dylib
libgz-sim-apply-link-wrench-system.dylib                  libgz-sim7-joint-trajectory-controller-system.7.0.0.dylib
libgz-sim-breadcrumbs-system.dylib                        libgz-sim7-joint-trajectory-controller-system.7.dylib
libgz-sim-buoyancy-engine-system.dylib                    libgz-sim7-joint-trajectory-controller-system.dylib
libgz-sim-buoyancy-system.dylib                           libgz-sim7-kinetic-energy-monitor-system.7.0.0.dylib
libgz-sim-camera-video-recorder-system.dylib              libgz-sim7-kinetic-energy-monitor-system.7.dylib
libgz-sim-collada-world-exporter-system.dylib             libgz-sim7-kinetic-energy-monitor-system.dylib
libgz-sim-comms-endpoint-system.dylib                     libgz-sim7-label-system.7.0.0.dylib
libgz-sim-contact-system.dylib                            libgz-sim7-label-system.7.dylib
libgz-sim-detachable-joint-system.dylib                   libgz-sim7-label-system.dylib
libgz-sim-diff-drive-system.dylib                         libgz-sim7-lift-drag-system.7.0.0.dylib
libgz-sim-elevator-system.dylib                           libgz-sim7-lift-drag-system.7.dylib
libgz-sim-follow-actor-system.dylib                       libgz-sim7-lift-drag-system.dylib
libgz-sim-forcetorque-system.dylib                        libgz-sim7-linearbatteryplugin-system.7.0.0.dylib
libgz-sim-hydrodynamics-system.dylib                      libgz-sim7-linearbatteryplugin-system.7.dylib
libgz-sim-imu-system.dylib                                libgz-sim7-linearbatteryplugin-system.dylib
libgz-sim-joint-controller-system.dylib                   libgz-sim7-log-system.7.0.0.dylib
libgz-sim-joint-position-controller-system.dylib          libgz-sim7-log-system.7.dylib
libgz-sim-joint-state-publisher-system.dylib              libgz-sim7-log-system.dylib
libgz-sim-joint-trajectory-controller-system.dylib        libgz-sim7-log-video-recorder-system.7.0.0.dylib
libgz-sim-kinetic-energy-monitor-system.dylib             libgz-sim7-log-video-recorder-system.7.dylib
libgz-sim-label-system.dylib                              libgz-sim7-log-video-recorder-system.dylib
libgz-sim-lift-drag-system.dylib                          libgz-sim7-logical-camera-system.7.0.0.dylib
libgz-sim-linearbatteryplugin-system.dylib                libgz-sim7-logical-camera-system.7.dylib
libgz-sim-log-system.dylib                                libgz-sim7-logical-camera-system.dylib
libgz-sim-log-video-recorder-system.dylib                 libgz-sim7-logicalaudiosensorplugin-system.7.0.0.dylib
libgz-sim-logical-camera-system.dylib                     libgz-sim7-logicalaudiosensorplugin-system.7.dylib
libgz-sim-logicalaudiosensorplugin-system.dylib           libgz-sim7-logicalaudiosensorplugin-system.dylib
libgz-sim-magnetometer-system.dylib                       libgz-sim7-magnetometer-system.7.0.0.dylib
libgz-sim-mecanum-drive-system.dylib                      libgz-sim7-magnetometer-system.7.dylib
libgz-sim-model-photo-shoot-system.dylib                  libgz-sim7-magnetometer-system.dylib
libgz-sim-multicopter-control-system.dylib                libgz-sim7-mecanum-drive-system.7.0.0.dylib
libgz-sim-multicopter-motor-model-system.dylib            libgz-sim7-mecanum-drive-system.7.dylib
libgz-sim-navsat-system.dylib                             libgz-sim7-mecanum-drive-system.dylib
libgz-sim-odometry-publisher-system.dylib                 libgz-sim7-model-photo-shoot-system.7.0.0.dylib
libgz-sim-opticaltactileplugin-system.dylib               libgz-sim7-model-photo-shoot-system.7.dylib
libgz-sim-particle-emitter-system.dylib                   libgz-sim7-model-photo-shoot-system.dylib
libgz-sim-particle-emitter2-system.dylib                  libgz-sim7-multicopter-control-system.7.0.0.dylib
libgz-sim-perfect-comms-system.dylib                      libgz-sim7-multicopter-control-system.7.dylib
libgz-sim-performer-detector-system.dylib                 libgz-sim7-multicopter-control-system.dylib
libgz-sim-physics-system.dylib                            libgz-sim7-multicopter-motor-model-system.7.0.0.dylib
libgz-sim-pose-publisher-system.dylib                     libgz-sim7-multicopter-motor-model-system.7.dylib
libgz-sim-rf-comms-system.dylib                           libgz-sim7-multicopter-motor-model-system.dylib
libgz-sim-scene-broadcaster-system.dylib                  libgz-sim7-navsat-system.7.0.0.dylib
libgz-sim-sensors-system.dylib                            libgz-sim7-navsat-system.7.dylib
libgz-sim-shader-param-system.dylib                       libgz-sim7-navsat-system.dylib
[... 114 lines cut ...]
  /Users/alex/.pyenv/shims/python3.7m-config
  /Users/alex/.pyenv/shims/python3.9-config
  /Users/alex/.pyenv/shims/python-config
  /Users/alex/.pyenv/shims/python3-config
  /Library/Frameworks/Python.framework/Versions/3.7/bin/python3.7-config
  /Library/Frameworks/Python.framework/Versions/3.7/bin/python3.7m-config
  /Library/Frameworks/Python.framework/Versions/3.7/bin/python3-config

Warning: Unbrewed dylibs were found in /usr/local/lib.
If you didn't put them there on purpose they could cause problems when
building Homebrew formulae, and may need to be deleted.

Unexpected dylibs:
  /usr/local/lib/librtlsdr.0.0.0.dylib
  /usr/local/lib/librtlsdr.0.dylib
  /usr/local/lib/librtlsdr.dylib
  /usr/local/lib/libusb-1.0.0.dylib
```
My Rbenv config:
```
➜  ~ which ruby
/Users/alex/.rbenv/shims/ruby
➜  ~ ruby --version
ruby 3.1.2p20 (2022-04-12 revision 4491bb740a) [arm64-darwin21]
```

scpeters, 2022-11-05:
I haven't yet had a chance to build bottles for arm64, so I'm not sure how much of gazebo works on that architecture

Ratatinator97, 2022-11-07:
I installed `gz-garden` and the problem persists:
```
➜  ~ gz gui          
Error while loading the library [/opt/homebrew/Cellar/gz-gui7/7.0.0/lib/gz-gui-7/plugins/libCameraTracking.dylib]: dlopen(/opt/homebrew/Cellar/gz-gui7/7.0.0/lib/gz-gui-7/plugins/libCameraTracking.dylib, 0x0085): Library not loaded: '@rpath/libgz-gui7.7.dylib'
  Referenced from: '/opt/homebrew/Cellar/gz-gui7/7.0.0/lib/gz-gui-7/plugins/libCameraTracking.dylib'
  Reason: tried: '/opt/homebrew/Cellar/gz-gui7/7.0.0/lib/gz-gui-7/plugins/../lib/libgz-gui7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-gui7/7.0.0/lib/gz-gui-7/plugins/../lib/libgz-gui7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-plugin2/2.0.0/lib/../lib/libgz-gui7.7.dylib' (no such file), '/usr/local/lib/libgz-gui7.7.dylib' (no such file), '/usr/lib/libgz-gui7.7.dylib' (no such file)
[GUI] [Err] [Application.cc:495] Failed to load plugin [CameraTracking] : couldn't load library on path [/opt/homebrew/Cellar/gz-gui7/7.0.0/lib/gz-gui-7/plugins/libCameraTracking.dylib].
➜  ~ ls /opt/homebrew/Cellar/gz-gui7/7.0.0/lib/gz-gui-7/plugins/
libCameraTracking.dylib         libMinimalScene.dylib           libTeleop.dylib
libGrid3D.dylib                 libNavSatMap.dylib              libTopicEcho.dylib
libGridConfig.dylib             libPointCloud.dylib             libTopicViewer.dylib
libImageDisplay.dylib           libPublisher.dylib              libTransportPlotting.dylib
libInteractiveViewControl.dylib libScreenshot.dylib             libTransportSceneManager.dylib
libKeyPublisher.dylib           libShutdownButton.dylib         libWorldControl.dylib
libMarkerManager.dylib          libTapeMeasure.dylib            libWorldStats.dylib
➜  ~ ruby --version
ruby 3.1.2p20 (2022-04-12 revision 4491bb740a) [arm64-darwin21]
```

Another example: 
```
/opt/homebrew/opt/ruby/bin/ruby $(which gz) gui             
Error while loading the library [/opt/homebrew/Cellar/gz-gui7/7.0.0/lib/gz-gui-7/plugins/libCameraTracking.dylib]: dlopen(/opt/homebrew/Cellar/gz-gui7/7.0.0/lib/gz-gui-7/plugins/libCameraTracking.dylib, 0x0085): Library not loaded: '@rpath/libgz-gui7.7.dylib'
  Referenced from: '/opt/homebrew/Cellar/gz-gui7/7.0.0/lib/gz-gui-7/plugins/libCameraTracking.dylib'
  Reason: tried: '/opt/homebrew/Cellar/gz-gui7/7.0.0/lib/gz-gui-7/plugins/../lib/libgz-gui7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-gui7/7.0.0/lib/gz-gui-7/plugins/../lib/libgz-gui7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-plugin2/2.0.0/lib/../lib/libgz-gui7.7.dylib' (no such file), '/usr/local/lib/libgz-gui7.7.dylib' (no such file), '/usr/lib/libgz-gui7.7.dylib' (no such file)
[GUI] [Err] [Application.cc:495] Failed to load plugin [CameraTracking] : couldn't load library on path [/opt/homebrew/Cellar/gz-gui7/7.0.0/lib/gz-gui-7/plugins/libCameraTracking.dylib].
```

Here are the rights of the differents libraries:
```
➜  ~ ls -al /opt/homebrew/Cellar/gz-gui7/7.0.0/lib/gz-gui-7/plugins/
total 8216
drwxr-xr-x  23 alex  admin     736 Nov  7 10:47 .
drwxr-xr-x   3 alex  admin      96 Nov  7 10:47 ..
-r--r--r--   1 alex  admin  245392 Nov  7 10:47 libCameraTracking.dylib
-r--r--r--   1 alex  admin  176720 Nov  7 10:47 libGrid3D.dylib
-r--r--r--   1 alex  admin  176720 Nov  7 10:47 libGridConfig.dylib
-r--r--r--   1 alex  admin  197904 Nov  7 10:47 libImageDisplay.dylib
-r--r--r--   1 alex  admin  204400 Nov  7 10:47 libInteractiveViewControl.dylib
-r--r--r--   1 alex  admin  130960 Nov  7 10:47 libKeyPublisher.dylib
-r--r--r--   1 alex  admin  285664 Nov  7 10:47 libMarkerManager.dylib
-r--r--r--   1 alex  admin  277904 Nov  7 10:47 libMinimalScene.dylib
-r--r--r--   1 alex  admin  174816 Nov  7 10:47 libNavSatMap.dylib
-r--r--r--   1 alex  admin  285840 Nov  7 10:47 libPointCloud.dylib
-r--r--r--   1 alex  admin  151776 Nov  7 10:47 libPublisher.dylib
-r--r--r--   1 alex  admin  201344 Nov  7 10:47 libScreenshot.dylib
-r--r--r--   1 alex  admin  128896 Nov  7 10:47 libShutdownButton.dylib
-r--r--r--   1 alex  admin  211328 Nov  7 10:47 libTapeMeasure.dylib
-r--r--r--   1 alex  admin  152592 Nov  7 10:47 libTeleop.dylib
-r--r--r--   1 alex  admin  175968 Nov  7 10:47 libTopicEcho.dylib
-r--r--r--   1 alex  admin  184768 Nov  7 10:47 libTopicViewer.dylib
-r--r--r--   1 alex  admin  128416 Nov  7 10:47 libTransportPlotting.dylib
-r--r--r--   1 alex  admin  286240 Nov  7 10:47 libTransportSceneManager.dylib
-r--r--r--   1 alex  admin  216944 Nov  7 10:47 libWorldControl.dylib
-r--r--r--   1 alex  admin  175424 Nov  7 10:47 libWorldStats.dylib
```

briankthsmth, 2023-04-26:
I have same problem when try to run `gz sim -s`. I noticed that when brew installs gz-sim7 it runs ` cmake .. -DCMAKE_INSTALL_RPATH=@loader_path/../lib`. You can see it in gz-sim7.rb file and the other gz-*.rb files.

The error is consistent with the linker  setting the path to libgz-sim7.7.dylib in each plugin's dylib to `@loader_path/../lib`, as the dynamic linker is searching for  libgz-sim7.7.dylib one directory below in a folder which doesn't exist. 

`@loader_path` would be the plugins path, _/opt/homebrew/Cellar/gz-sim7/7.4.0_2/lib/gz-sim-7/plugins_, but  libgz-sim7.7.dylib  would be in _/opt/homebrew/Cellar/gz-sim7/7.4.0_2/lib_. So, the rpath in the cmake call could be ` cmake .. -DCMAKE_INSTALL_RPATH=@loader_path/../../../lib` which would let the dynamic loader find libgz-sim7.7.dylib. This might cause problems for libgz-sim7.7.dylib if it has any dependencies in the _/opt/homebrew/Cellar/gz-sim7/7.4.0_2/lib_ folder. So, it might be better have a separate rpath for plugins for this. This is a similar problem for all of the gz-* modules that have plugins like gz-gui.

I wonder if this is only on Mac silicon with homebrew as homebrew no longer install to _/usr/local_, and now everything is in _/opt/homebrew_ on Mac silicon. Is homebrew  still adding links to the various libgz-*.dylib in _/usr/local/lib_ on other systems which the dynamic linker includes in its search path?

scpeters, 2023-09-21:
I believe I have fixed the RPATH issues for garden in #2409 and #2410

closed 2023-09-21 by scpeters