osrf/homebrew-simulation #2179: Gazebo 11 fails to install

ecmnet, 2023-01-14:
Installing Gazebo11 fails with

Last 15 lines from /Users/ecmnet/Library/Logs/Homebrew/gazebo11/02.make:
      std::array<std::string, 2> sensing =
                                 ^
/Library/Developer/CommandLineTools/SDKs/MacOSX13.sdk/usr/include/c++/v1/__tuple:219:64: note: template is declared here
template <class _Tp, size_t _Size> struct _LIBCPP_TEMPLATE_VIS array;
                                                               ^
/tmp/gazebo11-20230114-26075-1tcpwqi/gazebo-11.12.0/gazebo/msgs/msgs.cc:2023:34: error: implicit instantiation of undefined template 'std::array<std::string, 2>'
      std::array<std::string, 2> dimensions = {{"horizontal", "vertical"}};
                                 ^
/Library/Developer/CommandLineTools/SDKs/MacOSX13.sdk/usr/include/c++/v1/__tuple:219:64: note: template is declared here
template <class _Tp, size_t _Size> struct _LIBCPP_TEMPLATE_VIS array;
                                                               ^
4 errors generated.
make[2]: *** [gazebo/msgs/CMakeFiles/gazebo_msgs.dir/msgs.cc.o] Error 1
make[1]: *** [gazebo/msgs/CMakeFiles/gazebo_msgs.dir/all] Error 2
make: *** [all] Error 2

laborosh, 2023-01-25:
Sorry for no answer. I too have the same problem on macOS 13, Intel version.

tdwsNi, 2023-01-25:
#2181

scpeters, 2023-01-26:
> #2181

yes, I opened a PR to rebuild bottles, but the monterey bottles failed to build. I'm investigating on my local machine

scpeters, 2023-01-26:
the compilation error should be fixed by https://github.com/gazebosim/gazebo-classic/pull/3290

tdwsNi, 2023-01-26:
> 
thnaks @scpeters 
confirmed. 
fyi i also needed to downgrade qt to qt5 and qwt to qwt-qt5 to compile it sucessfully.

scpeters, 2023-01-26:
trying a bottle rebuild with that patch 🤞

ecmnet, 2023-01-28:
Seems to work partially. GZClient now fails due to missing billet 3.24. The recipe installs version 3.25.

scpeters, 2023-01-30:
this is now failing due to https://github.com/Homebrew/homebrew-core/pull/121687

scpeters, 2023-01-30:
the bottle rebuild in https://github.com/osrf/homebrew-simulation/pull/2181 didn't last very long

scpeters, 2023-01-30:
rebuilding in #2189

ecmnet, 2023-01-31:
Latest bottle worked for me

closed 2023-01-31 by ecmnet