osrf/homebrew-simulation #2220: Failed to build gazebo11

Sitin, 2023-02-19:
Condition:

~~~
HOMEBREW_VERSION: 4.0.1
ORIGIN: https://github.com/Homebrew/brew
HEAD: 17c872fb5275d87922a56416587cb439a5064354
Last commit: 3 days ago
Core tap origin: https://github.com/Homebrew/homebrew-core
Core tap HEAD: 1d8c58a0aad7953501bd69df9f5014a6dfbf199e
Core tap last commit: 3 days ago
Core tap branch: master
Core tap JSON: 19 Feb 09:50 UTC
HOMEBREW_PREFIX: /opt/homebrew
HOMEBREW_CASK_OPTS: []
HOMEBREW_DISPLAY: /private/tmp/com.apple.launchd.vD5aQBdL4k/org.xquartz:0
HOMEBREW_GITHUB_API_TOKEN: set
HOMEBREW_MAKE_JOBS: 10
Homebrew Ruby: 2.6.10 => /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/bin/ruby
CPU: 10-core 64-bit arm_firestorm_icestorm
Clang: 14.0.0 build 1400
Git: 2.37.1 => /Applications/Xcode.app/Contents/Developer/usr/bin/git
Curl: 7.86.0 => /usr/bin/curl
macOS: 13.2.1-arm64
CLT: N/A
Xcode: 14.2
Rosetta 2: false
~~~

More details in the [autogenerated gist](https://gist.github.com/Sitin/f46e51945265fb574f2a07033ee1d81a).

DhruvilROS, 2024-04-11:
any solutions for this issue?

gmerlino, 2024-04-12:
> any solutions for this issue?

It looks like it's just a matter of changing the dependency from `ffmpeg` (which now default to version 7.x) to `ffmpeg@6`.
I just did it this way, and it built successfully now.

DhruvilROS, 2024-04-13:
Last 15 lines from /Users/dhruvil/Library/Logs/Homebrew/gazebo11/02.make:
cd /tmp/gazebo11-20240413-7006-6b99ap/gazebo-11.14.0/build/deps/opende/OPCODE && /opt/homebrew/Library/Homebrew/shims/mac/super/clang++ -DHAVE_OPENGL -DIGN_PROFILER_ENABLE=0 -DLIBBULLET_VERSION=3.25 -DLIBBULLET_VERSION_GT_282 -DMAC_OS_X_VERSION=1080 -Dgazebo_opcode_EXPORTS -I/tmp/gazebo11-20240413-7006-6b99ap/gazebo-11.14.0/test/gtest/include -I/tmp/gazebo11-20240413-7006-6b99ap/gazebo-11.14.0 -I/tmp/gazebo11-20240413-7006-6b99ap/gazebo-11.14.0/build -I/tmp/gazebo11-20240413-7006-6b99ap/gazebo-11.14.0/build/deps/opende -I/tmp/gazebo11-20240413-7006-6b99ap/gazebo-11.14.0/deps/opende/include -I/tmp/gazebo11-20240413-7006-6b99ap/gazebo-11.14.0/deps/opende/ou/include -I/tmp/gazebo11-20240413-7006-6b99ap/gazebo-11.14.0/deps/opende/OPCODE -I/tmp/gazebo11-20240413-7006-6b99ap/gazebo-11.14.0/deps/opende/GIMPACT/include -I/tmp/gazebo11-20240413-7006-6b99ap/gazebo-11.14.0/deps/threadpool -I/opt/homebrew/Cellar/tinyxml2/10.0.0/include -I/opt/homebrew/Cellar/ffmpeg/7.0/include -I/opt/homebrew/include/ignition/msgs5 -I/opt/homebrew/include/ignition/transport8 -I/opt/homebrew/include/ignition/common3 -I/opt/homebrew/include/ignition/fuel_tools4 -I/opt/homebrew/Cellar/libccd/2.1_1/include -I/opt/homebrew/Cellar/bullet/3.25/include/bullet -I/opt/homebrew/Cellar/bullet/3.25/include -isystem /tmp/gazebo11-20240413-7006-6b99ap/gazebo-11.14.0/build/deps/opende/OPCODE -isystem /tmp/gazebo11-20240413-7006-6b99ap/gazebo-11.14.0/deps/opende/OPCODE/Ice -isystem /opt/homebrew/include/ignition/math6 -stdlib=libc++  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wfloat-equal -Wshadow -Wswitch-default -Wmissing-include-dirs -pedantic  -DBUILDING_DLL_OPCODE  -s -O3 -DNDEBUG  -std=c++11 -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk -fPIC -MD -MT deps/opende/OPCODE/CMakeFiles/gazebo_opcode.dir/Ice/IceTriangle.cpp.o -MF CMakeFiles/gazebo_opcode.dir/Ice/IceTriangle.cpp.o.d -o CMakeFiles/gazebo_opcode.dir/Ice/IceTriangle.cpp.o -c /tmp/gazebo11-20240413-7006-6b99ap/gazebo-11.14.0/deps/opende/OPCODE/Ice/IceTriangle.cpp
[  6%] Building CXX object deps/opende/OPCODE/CMakeFiles/gazebo_opcode.dir/Ice/IceUtils.cpp.o
cd /tmp/gazebo11-20240413-7006-6b99ap/gazebo-11.14.0/build/deps/opende/OPCODE && /opt/homebrew/Library/Homebrew/shims/mac/super/clang++ -DHAVE_OPENGL -DIGN_PROFILER_ENABLE=0 -DLIBBULLET_VERSION=3.25 -DLIBBULLET_VERSION_GT_282 -DMAC_OS_X_VERSION=1080 -Dgazebo_opcode_EXPORTS -I/tmp/gazebo11-20240413-7006-6b99ap/gazebo-11.14.0/test/gtest/include -I/tmp/gazebo11-20240413-7006-6b99ap/gazebo-11.14.0 -I/tmp/gazebo11-20240413-7006-6b99ap/gazebo-11.14.0/build -I/tmp/gazebo11-20240413-7006-6b99ap/gazebo-11.14.0/build/deps/opende -I/tmp/gazebo11-20240413-7006-6b99ap/gazebo-11.14.0/deps/opende/include -I/tmp/gazebo11-20240413-7006-6b99ap/gazebo-11.14.0/deps/opende/ou/include -I/tmp/gazebo11-20240413-7006-6b99ap/gazebo-11.14.0/deps/opende/OPCODE -I/tmp/gazebo11-20240413-7006-6b99ap/gazebo-11.14.0/deps/opende/GIMPACT/include -I/tmp/gazebo11-20240413-7006-6b99ap/gazebo-11.14.0/deps/threadpool -I/opt/homebrew/Cellar/tinyxml2/10.0.0/include -I/opt/homebrew/Cellar/ffmpeg/7.0/include -I/opt/homebrew/include/ignition/msgs5 -I/opt/homebrew/include/ignition/transport8 -I/opt/homebrew/include/ignition/common3 -I/opt/homebrew/include/ignition/fuel_tools4 -I/opt/homebrew/Cellar/libccd/2.1_1/include -I/opt/homebrew/Cellar/bullet/3.25/include/bullet -I/opt/homebrew/Cellar/bullet/3.25/include -isystem /tmp/gazebo11-20240413-7006-6b99ap/gazebo-11.14.0/build/deps/opende/OPCODE -isystem /tmp/gazebo11-20240413-7006-6b99ap/gazebo-11.14.0/deps/opende/OPCODE/Ice -isystem /opt/homebrew/include/ignition/math6 -stdlib=libc++  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wfloat-equal -Wshadow -Wswitch-default -Wmissing-include-dirs -pedantic  -DBUILDING_DLL_OPCODE  -s -O3 -DNDEBUG  -std=c++11 -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk -fPIC -MD -MT deps/opende/OPCODE/CMakeFiles/gazebo_opcode.dir/Ice/IceUtils.cpp.o -MF CMakeFiles/gazebo_opcode.dir/Ice/IceUtils.cpp.o.d -o CMakeFiles/gazebo_opcode.dir/Ice/IceUtils.cpp.o -c /tmp/gazebo11-20240413-7006-6b99ap/gazebo-11.14.0/deps/opende/OPCODE/Ice/IceUtils.cpp
[  6%] Linking CXX shared library libgazebo_opcode.dylib
cd /tmp/gazebo11-20240413-7006-6b99ap/gazebo-11.14.0/build/deps/opende/OPCODE && /opt/homebrew/Cellar/cmake/3.29.2/bin/cmake -E cmake_link_script CMakeFiles/gazebo_opcode.dir/link.txt --verbose=1
/opt/homebrew/Library/Homebrew/shims/mac/super/clang++       -stdlib=libc++  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wfloat-equal -Wshadow -Wswitch-default -Wmissing-include-dirs -pedantic  -DBUILDING_DLL_OPCODE  -s -O3 -DNDEBUG  -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk -dynamiclib -Wl,-headerpad_max_install_names    -Wl,-undefined -Wl,dynamic_lookup -compatibility_version 11.0.0 -current_version 11.14.0 -o libgazebo_opcode.11.14.0.dylib -install_name @rpath/libgazebo_opcode.11.dylib CMakeFiles/gazebo_opcode.dir/OPC_AABBCollider.cpp.o CMakeFiles/gazebo_opcode.dir/OPC_AABBTree.cpp.o CMakeFiles/gazebo_opcode.dir/OPC_BaseModel.cpp.o CMakeFiles/gazebo_opcode.dir/OPC_Collider.cpp.o CMakeFiles/gazebo_opcode.dir/OPC_Common.cpp.o CMakeFiles/gazebo_opcode.dir/OPC_HybridModel.cpp.o CMakeFiles/gazebo_opcode.dir/OPC_LSSCollider.cpp.o CMakeFiles/gazebo_opcode.dir/OPC_MeshInterface.cpp.o CMakeFiles/gazebo_opcode.dir/OPC_Model.cpp.o CMakeFiles/gazebo_ [... cut ...] rfactory.cpp.o CMakeFiles/gzqtpropertybrowser.dir/qtvariantproperty.cpp.o CMakeFiles/gzqtpropertybrowser.dir/qttreepropertybrowser.cpp.o CMakeFiles/gzqtpropertybrowser.dir/qtbuttonpropertybrowser.cpp.o CMakeFiles/gzqtpropertybrowser.dir/qtgroupboxpropertybrowser.cpp.o CMakeFiles/gzqtpropertybrowser.dir/qtpropertybrowserutils.cpp.o CMakeFiles/gzqtpropertybrowser.dir/qrc_qtpropertybrowser.cpp.o
/usr/bin/ranlib libgzqtpropertybrowser.a
[  6%] Built target gzqtpropertybrowser
make: *** [all] Error 2

DhruvilROS, 2024-04-13:
I am receiving this error. please reach out to me if you have any solution for this.

gmerlino, 2024-04-14:
https://github.com/osrf/homebrew-simulation/pull/2636 (defaulting to ffmpeg 7 but patching accordingly) is merged now, and should have solved @DhruvilROS

scpeters, 2024-04-16:
the error from the original post is in [02.make](https://gist.github.com/Sitin/f46e51945265fb574f2a07033ee1d81a/raw/16afba6008273255868ce00b5dbb615800fe3b62/02.make):

~~~
make[2]: *** No rule to make target `/Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk/usr/lib/libm.tbd', needed by `deps/opende/libgazebo_ode.11.12.0.dylib'.  Stop.
~~~

this seems like an issue with the version of Xcode / command-line tools

scpeters, 2024-04-16:
> Last 15 lines from /Users/dhruvil/Library/Logs/Homebrew/gazebo11/02.make:

@DhruvilROS this section of the log doesn't show the actual error; I recommend looking for `***` in the referenced log file

scpeters, 2024-05-15:
I can't this so I'm going to close until more information is provided

closed 2024-05-15 by scpeters