osrf/homebrew-simulation #2280: Broken libraries paths in `gz-garden`

Sitin, 2023-06-10:
Hi, first of all, thank you for the great work! Meanwhile...

### Gazebo Sim can't load plugins due to broken library paths

This happens both for `gz sim -s` and `gz sim -g`.

For example (while running `gz sim -s`):

```
...
Error while loading the library [/opt/homebrew/Cellar/gz-gui7/7.2.0_1/lib/gz-gui-7/plugins/libMinimalScene.dylib]: dlopen(/opt/homebrew/Cellar/gz-gui7/7.2.0_1/lib/gz-gui-7/plugins/libMinimalScene.dylib, 0x0085): Library not loaded: @rpath/libgz-gui7.7.dylib
  Referenced from: <9B0838E7-B717-3727-A10F-C87E39958AEF> /opt/homebrew/Cellar/gz-gui7/7.2.0_1/lib/gz-gui-7/plugins/libMinimalScene.dylib
  Reason: tried: '/opt/homebrew/Cellar/gz-gui7/7.2.0_1/lib/gz-gui-7/plugins/../lib/libgz-gui7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-gui7/7.2.0_1/lib/gz-gui-7/plugins/../lib/libgz-gui7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-plugin2/2.0.1/lib/../lib/libgz-gui7.7.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS@rpath/libgz-gui7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-gui7/7.2.0_1/lib/gz-gui-7/plugins/../lib/libgz-gui7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-gui7/7.2.0_1/lib/gz-gui-7/plugins/../lib/libgz-gui7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-plugin2/2.0.1/lib/../lib/libgz-gui7.7.dylib' (no such file), '/usr/lib/libgz-gui7.7.dylib' (no such file, not in dyld cache), (security policy does not allow @ path expansion)
[GUI] [Err] [Application.cc:506] Failed to load plugin [MinimalScene] : couldn't load library on path [/opt/homebrew/Cellar/gz-gui7/7.2.0_1/lib/gz-gui-7/plugins/libMinimalScene.dylib].
...
```

The list of errors is long. I've put everything to [gist](https://gist.github.com/Sitin/cab4eb81cabaacdcc3b833c16be479d0).

This can be fixed with a workaround:

```shell
ln -s /opt/homebrew/Cellar/gz-sim7/7.5.0_1/lib /opt/homebrew/Cellar/gz-sim7/7.5.0_1/lib/gz-sim-7/lib
ln -s /opt/homebrew/Cellar/gz-sim7/7.5.0_1/lib /opt/homebrew/Cellar/gz-sim7/7.5.0_1/lib/gz-sim-7/plugins/gui/lib
ln -s /opt/homebrew/Cellar/gz-sim7/7.5.0_1/lib /opt/homebrew/Cellar/gz-sim7/7.5.0_1/lib/gz-sim-7/plugins/lib
ln -s /opt/homebrew/Cellar/gz-physics6/6.4.0_1/lib /opt/homebrew/Cellar/gz-physics6/6.4.0_1/lib/gz-physics-6/lib
ln -s /opt/homebrew/Cellar/gz-gui7/7.2.0_1/lib /opt/homebrew/Cellar/gz-gui7/7.2.0_1/lib/gz-gui-7/lib
ln -s /opt/homebrew/Cellar/gz-rendering7/7.4.0/lib/libgz-rendering7.7.dylib /opt/homebrew/Cellar/gz-plugin2/2.0.1/lib/libgz-rendering7.7.dylib
```

Looks like involved packages are `gz-sim-7`, `gz-physics6`, `gz-gui7`, and `gz-plugin2`.

### Brew configuration

~~~
$ brew config
HOMEBREW_VERSION: 4.0.21
ORIGIN: https://github.com/Homebrew/brew
HEAD: 9d0b7361d1a197949dbb47b761405075f83983b7
Last commit: 4 days ago
Core tap JSON: 10 Jun 05:58 UTC
HOMEBREW_PREFIX: /opt/homebrew
HOMEBREW_CASK_OPTS: []
HOMEBREW_DISPLAY: /private/tmp/com.apple.launchd.vD5aQBdL4k/org.xquartz:0
HOMEBREW_GITHUB_API_TOKEN: set
HOMEBREW_MAKE_JOBS: 10
Homebrew Ruby: 2.6.10 => /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/bin/ruby
CPU: 10-core 64-bit arm_firestorm_icestorm
Clang: 14.0.3 build 1403
Git: 2.39.2 => /Applications/Xcode.app/Contents/Developer/usr/bin/git
Curl: 7.86.0 => /usr/bin/curl
macOS: 13.2.1-arm64
CLT: 14.3.1.0.1.1683849156
Xcode: 14.3.1
Rosetta 2: false
~~~

### Brew linkage test is green

Brew tells that the entire `osrf/simulation/*` is linked correctly.

~~~
$brew list -1 --full-name | grep osrf/simulation/ | xargs brew linkage --test
==> Checking gz-cmake3 linkage
==> Checking gz-common5 linkage
==> Checking gz-fuel-tools8 linkage
==> Checking gz-garden linkage
==> Checking gz-gui7 linkage
==> Checking gz-launch6 linkage
==> Checking gz-math7 linkage
==> Checking gz-msgs9 linkage
==> Checking gz-physics6 linkage
==> Checking gz-plugin2 linkage
==> Checking gz-rendering7 linkage
==> Checking gz-sensors7 linkage
==> Checking gz-sim7 linkage
==> Checking gz-tools2 linkage
==> Checking gz-transport12 linkage
==> Checking gz-utils2 linkage
==> Checking ogre1.9 linkage
==> Checking ogre2.3 linkage
==> Checking sdformat13 linkage
~~~

Just in case, `brew doctor` have no complains as well:

~~~
$brew doctor
Your system is ready to brew.
~~~

Sitin, 2023-06-10:
Probably related to #1827

scpeters, 2023-06-11:
I'm currently trying to fix bottles broken by protobuf (tracked in #2274). I'll take a look at this when I have a chance

Sitin, 2023-06-11:
Thank you, @scpeters.

HiceS, 2023-06-22:
I also encountered a error with `gz-fuel-tools8`. Looks like there is a protobuf mismatch that is mentioned above. 

For anyone that is looking into this and wants more information I am running on M1 with Ventura and here is the output of the verbose pour. https://gist.github.com/HiceS/cc2b2fa2f7800f7648345bd6a14d2281

Locally I use protobuf 23.3 from brew but 21 was successfully pulled down in an earlier step.

scpeters, 2023-09-19:
I think this is fixed for the garden formulae by the RPATH changes in #2409 and #2410

closed 2023-09-21 by scpeters