osrf/homebrew-simulation #2289: Fail to install gazebo as installing errors

kevinuav, 2023-06-13:
Please supply your brew configuration:

~~~
$ brew config
HOMEBREW_VERSION: 4.0.22-6-gf71294d
ORIGIN: https://github.com/Homebrew/brew
HEAD: f71294d483e1ce9920fd10d8169b1419c0ba41fb
Last commit: 5 hours ago
Core tap origin: https://github.com/Homebrew/homebrew-core
Core tap HEAD: 8e196e546044e33c3246ce89b0603bdbf774374b
Core tap last commit: 6 hours ago
Core tap branch: master
Core tap JSON: 13 Jun 20:42 UTC
HOMEBREW_PREFIX: /usr/local
HOMEBREW_CASK_OPTS: []
HOMEBREW_DISPLAY: /private/tmp/com.apple.launchd.UopL9MA43V/org.xquartz:0
HOMEBREW_MAKE_JOBS: 8
Homebrew Ruby: 2.6.10 => /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/bin/ruby
CPU: octa-core 64-bit kabylake
Clang: 14.0.0 build 1400
Git: 2.37.1 => /Applications/Xcode.app/Contents/Developer/usr/bin/git
Curl: 7.87.0 => /usr/bin/curl
macOS: 12.6.6-x86_64
CLT: 14.2.0.0.1.1668646533
Xcode: 14.2
~~~

If you are reporting a build failure, please supply the build logs
with the `--verbose` flag:

~~~
% brew install px4-sim-gazebo --verbose
..............
In file included from /tmp/gazebo11-20230614-11471-yopqbe/gazebo-11.13.0/gazebo/msgs/msgs.cc:28:
In file included from /tmp/gazebo11-20230614-11471-yopqbe/gazebo-11.13.0/gazebo/msgs/msgs.hh:32:
/usr/local/include/ignition/msgs5/ignition/msgs/color.pb.h:13:2: error: "This file was generated by a newer version of protoc which is"
#error "This file was generated by a newer version of protoc which is"
 ^
/usr/local/include/ignition/msgs5/ignition/msgs/color.pb.h:14:2: error: "incompatible with your Protocol Buffer headers. Please update"
#error "incompatible with your Protocol Buffer headers. Please update"
 ^
/usr/local/include/ignition/msgs5/ignition/msgs/color.pb.h:15:2: error: "your headers."
#error "your headers."
 ^
In file included from /tmp/gazebo11-20230614-11471-yopqbe/gazebo-11.13.0/gazebo/msgs/msgs.cc:28:
In file included from /tmp/gazebo11-20230614-11471-yopqbe/gazebo-11.13.0/gazebo/msgs/msgs.hh:32:
In file included from /usr/local/include/ignition/msgs5/ignition/msgs/color.pb.h:34:
/usr/local/include/ignition/msgs5/ignition/msgs/header.pb.h:13:2: error: "This file was generated by a newer version of protoc which is"
#error "This file was generated by a newer version of protoc which is"
 ^
/usr/local/include/ignition/msgs5/ignition/msgs/header.pb.h:14:2: error: "incompatible with your Protocol Buffer headers. Please update"
#error "incompatible with your Protocol Buffer headers. Please update"
 ^
/usr/local/include/ignition/msgs5/ignition/msgs/header.pb.h:15:2: error: "your headers."
#error "your headers."
 ^
In file included from /tmp/gazebo11-20230614-11471-yopqbe/gazebo-11.13.0/gazebo/msgs/msgs.cc:28:
In file included from /tmp/gazebo11-20230614-11471-yopqbe/gazebo-11.13.0/gazebo/msgs/msgs.hh:32:
In file included from /usr/local/include/ignition/msgs5/ignition/msgs/color.pb.h:34:
In file included from /usr/local/include/ignition/msgs5/ignition/msgs/header.pb.h:34:
/usr/local/include/ignition/msgs5/ignition/msgs/time.pb.h:13:2: error: "This file was generated by a newer version of protoc which is"
#error "This file was generated by a newer version of protoc which is"
 ^
/usr/local/include/ignition/msgs5/ignition/msgs/time.pb.h:14:2: error: "incompatible with your Protocol Buffer headers. Please update"
#error "incompatible with your Protocol Buffer headers. Please update"
 ^
/usr/local/include/ignition/msgs5/ignition/msgs/time.pb.h:15:2: error: "your headers."
#error "your headers."
 ^
/usr/local/include/ignition/msgs5/ignition/msgs/time.pb.h:195:12: error: no member named 'absl' in the global namespace
[... 64 lines cut ...]
HOMEBREW_CC: clang
HOMEBREW_CXX: clang++
MAKEFLAGS: -j8
CMAKE_PREFIX_PATH: /usr/local/opt/icu4c:/usr/local/opt/openblas:/usr/local/opt/openssl@1.1:/usr/local/opt/libarchive:/usr/local/opt/m4:/usr/local/opt/readline:/usr/local/opt/sqlite:/usr/local/opt/ruby:/usr/local/opt/protobuf@21:/usr/local/opt/qt@5:/usr/local/opt/qwt-qt5:/usr/local
CMAKE_INCLUDE_PATH: /Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk/System/Library/Frameworks/OpenGL.framework/Versions/Current/Headers
CMAKE_LIBRARY_PATH: /Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk/System/Library/Frameworks/OpenGL.framework/Versions/Current/Libraries
CMAKE_FRAMEWORK_PATH: /usr/local/opt/python@3.11/Frameworks:/usr/local/opt/qt@5/Frameworks
PKG_CONFIG_PATH: /usr/local/opt/protobuf-c/lib/pkgconfig:/usr/local/opt/icu4c/lib/pkgconfig:/usr/local/opt/xz/lib/pkgconfig:/usr/local/opt/lz4/lib/pkgconfig:/usr/local/opt/zstd/lib/pkgconfig:/usr/local/opt/bullet/lib/pkgconfig:/usr/local/opt/assimp/lib/pkgconfig:/usr/local/opt/libccd/lib/pkgconfig:/usr/local/opt/octomap/lib/pkgconfig:/usr/local/opt/fcl/lib/pkgconfig:/usr/local/opt/gmp/lib/pkgconfig:/usr/local/opt/isl/lib/pkgconfig:/usr/local/opt/mpfr/lib/pkgconfig:/usr/local/opt/flann/lib/pkgconfig:/usr/local/opt/fmt/lib/pkgconfig:/usr/local/opt/openblas/lib/pkgconfig:/usr/local/opt/ipopt/lib/pkgconfig:/usr/local/opt/nlopt/lib/pkgconfig:/usr/local/opt/ode/lib/pkgconfig:/usr/local/opt/libpng/lib/pkgconfig:/usr/local/opt/freetype/lib/pkgconfig:/usr/local/opt/fontconfig/lib/pkgconfig:/usr/local/opt/jpeg-turbo/lib/pkgconfig:/usr/local/opt/sdl2/lib/pkgconfig:/usr/local/opt/open-scene-graph/lib/pkgconfig:/usr/local/opt/spdlog/lib/pkgconfig:/usr/local/opt/tinyxml2/lib/pkgconfig:/usr/local/opt/console_bridge/lib/pkgconfig:/usr/local/opt/tinyxml/lib/pkgconfig:/usr/local/opt/urdfdom_headers/lib/pkgconfig:/usr/local/opt/urdfdom/lib/pkgconfig:/usr/local/opt/dartsim/lib/pkgconfig:/usr/local/opt/brotli/lib/pkgconfig:/usr/local/opt/highway/lib/pkgconfig:/usr/local/opt/imath/lib/pkgconfig:/usr/local/opt/libtiff/lib/pkgconfig:/usr/local/opt/little-cms2/lib/pkgconfig:/usr/local/opt/openexr/lib/pkgconfig:/usr/local/opt/webp/lib/pkgconfig:/usr/local/opt/jpeg-xl/lib/pkgconfig:/usr/local/opt/libvmaf/lib/pkgconfig:/usr/local/opt/aom/lib/pkgconfig:/usr/local/opt/aribb24/lib/pkgconfig:/usr/local/opt/dav1d/lib/pkgconfig:/usr/local/opt/frei0r/lib/pkgconfig:/usr/local/opt/libidn2/lib/pkgconfig:/usr/local/opt/libtasn1/lib/pkgconfig:/usr/local/opt/nettle/lib/pkgconfig:/usr/local/opt/p11-kit/lib/pkgconfig:/usr/local/opt/openssl@1.1/lib/pkgconfig:/usr/local/opt/libevent/lib/pkgconfig:/usr/local/opt/libnghttp2/lib/pkgconfig:/usr/local/opt/unbound/lib/pkgconfig:/usr/local/opt/gnutls/lib/pkgconfig:/usr/local/opt/fribidi/lib/pkgconfig:/usr/local/opt/pcre2/lib/pkgconfig:/usr/local/opt/glib/lib/pkgconfig:/usr/local/opt/libxau/lib/pkgconfig:/usr/local/opt/libxdmcp/lib/pkgconfig:/usr/local/opt/libxcb/lib/pkgconfig:/usr/local/opt/libx11/lib/pkgconfig:/usr/local/opt/libxext/lib/pkgconfig:/usr/local/opt/libxrender/lib/pkgconfig:/usr/local/opt/lzo/lib/pkgconfig:/usr/local/opt/pixman/lib/pkgconfig:/usr/local/opt/cairo/lib/pkgconfig:/usr/local/opt/graphite2/lib/pkgconfig:/usr/local/opt/harfbuzz/lib/pkgconfig:/usr/local/opt/libunibreak/lib/pkgconfig:/usr/local/opt/libass/lib/pkgconfig:/usr/local/opt/libbluray/lib/pkgconfig:/usr/local/opt/cjson/lib/pkgconfig:/usr/local/opt/librist/lib/pkgconfig:/usr/local/opt/libsoxr/lib/pkgconfig:/usr/local/opt/libvidstab/lib/pkgconfig:/usr/local/opt/libogg/lib/pkgconfig:/usr/local/opt/libvorbis/lib/pkgconfig:/usr/local/opt/libvpx/lib/pkgconfig:/usr/local/opt/opencore-amr/lib/pkgconfig:/usr/local/opt/openjpeg/lib/pkgconfig:/usr/local/opt/opus/lib/pkgconfig:/usr/local/opt/rav1e/lib/pkgconfig:/usr/local/opt/libsamplerate/lib/pkgconfig:/usr/local/opt/flac/lib/pkgconfig:/usr/local/opt/mpg123/lib/pkgconfig:/usr/local/opt/libsndfile/lib/pkgconfig:/usr/local/opt/rubberband/lib/pkgconfig:/usr/local/opt/speex/lib/pkgconfig:/usr/local/opt/srt/lib/pkgconfig:/usr/local/opt/svt-av1/lib/pkgconfig:/usr/local/opt/leptonica/lib/pkgconfig:/usr/local/opt/libb2/lib/pkgconfig:/usr/local/opt/libarchive/lib/pkgconfig:/usr/local/opt/pango/lib/pkgconfig:/usr/local/opt/tesseract/lib/pkgconfig:/usr/local/opt/theora/lib/pkgconfig:/usr/local/opt/x264/lib/pkgconfig:/usr/local/opt/x265/lib/pkgconfig:/usr/local/opt/libsodium/lib/pkgconfig:/usr/local/opt/zeromq/lib/pkgconfig:/usr/local/opt/zimg/lib/pkgconfig:/usr/local/opt/ffmpeg/lib/pkgconfig:/usr/local/opt/libavif/lib/pkgconfig:/usr/local/opt/gd/lib/pkgconfig:/usr/local/opt/jasper/lib/pkgconfig:/usr/local/opt/netpbm/lib/pkgconfig:/usr/local/opt/gts/lib/pkgconfig:/usr/local/opt/gdk-pixbuf/lib/pkgconfig:/usr/local/opt/librsvg/lib/pkgconfig:/usr/local/opt/graphviz/lib/pkgconfig:/usr/local/opt/ignition-cmake2/lib/pkgconfig:/usr/local/opt/readline/lib/pkgconfig:/usr/local/opt/sqlite/lib/pkgconfig [... cut ...] c/bin:/usr/bin:/bin:/usr/sbin:/sbin

Error: osrf/simulation/gazebo11 11.13.0 did not build
Logs:
     /Users/kevin/Library/Logs/Homebrew/gazebo11/00.options.out
     /Users/kevin/Library/Logs/Homebrew/gazebo11/01.cmake.cc
     /Users/kevin/Library/Logs/Homebrew/gazebo11/01.cmake
     /Users/kevin/Library/Logs/Homebrew/gazebo11/02.make.cc
     /Users/kevin/Library/Logs/Homebrew/gazebo11/build
     /Users/kevin/Library/Logs/Homebrew/gazebo11/02.make
~~~

then include the link in the issue. Thanks!

Onicc, 2023-06-14:
hi, i have the same problem.

kevinuav, 2023-06-14:
> 
Emm. I think it is a new bug since I have installed it successfully no long ago. But it fails this time. 
Just as the brew config says, it has been committed just 5hours ago.

HOMEBREW_VERSION: 4.0.22-6-gf71294d
ORIGIN: https://github.com/Homebrew/brew
HEAD: f71294d483e1ce9920fd10d8169b1419c0ba41fb
Last commit: 5 hours ago

scpeters, 2023-06-15:
gazebo11 was just rebuilt in https://github.com/osrf/homebrew-simulation/pull/2295, so this should be fixed now

closed 2023-06-15 by scpeters