osrf/homebrew-simulation #2339: Rebuild bottles after protobuf 24.1 is in homebrew-core

scpeters, 2023-08-23:
Placeholder issue for when https://github.com/Homebrew/homebrew-core/pull/139954/ is merged

Please supply your brew configuration:

### Brew configuration

~~~
$ brew config

~~~

### Brew linkage test result

~~~
$ brew linkage --test gz-common5

~~~

### Checklist

- [ ] Identify change in homebrew-core that broke the bottle https://github.com/Homebrew/homebrew-core/pull/139954
- [ ] Identify list of affected formulae
    - [ ] gz-*
- [ ] Remove broken bottles: for each affected formula, run `brew bump-revision --remove-bottle-block`
- [ ] Rebuild broken bottles

scpeters, 2023-08-30:
the subsequent update of abseil in https://github.com/Homebrew/homebrew-core/pull/139025 has broken these formulae again. I'll remove broken formulae again

j-rivero, 2023-09-05:
@scpeters we probably can close this one, right?

scpeters, 2023-09-05:
yes, these bottles are rebuilt

closed 2023-09-05 by scpeters