osrf/homebrew-simulation #2347: ignition-fortress  install fail on apple m2

devashishRaj, 2023-08-29:
Please supply your brew configuration:

~~~
$ brew config

HOMEBREW_VERSION: 4.1.7
ORIGIN: https://github.com/Homebrew/brew
HEAD: d4444b563e24ac7c05a93121c464c02dfa04d44f
Last commit: 35 hours ago
Core tap JSON: 29 Aug 06:58 UTC
HOMEBREW_PREFIX: /opt/homebrew
HOMEBREW_CASK_OPTS: []
HOMEBREW_DISPLAY: /private/tmp/com.apple.launchd.ymSidsSit8/org.xquartz:0
HOMEBREW_MAKE_JOBS: 10
Homebrew Ruby: 2.6.10 => /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/bin/ruby
CPU: 10-core 64-bit arm_blizzard_avalanche
Clang: 14.0.3 build 1403
Git: 2.42.0 => /opt/homebrew/bin/git
Curl: 8.1.2 => /usr/bin/curl
macOS: 13.5.1-arm64
CLT: 14.3.1.0.1.1683849156
Xcode: 14.3.1
Rosetta 2: false

~~~

If you are reporting a build failure, please supply the build logs
with the `--verbose` flag:

https://gist.github.com/devashishRaj/80664d4fd355a9c974a85159d159528a

Error: osrf/simulation/ogre2.2 2.2.6+20211021~312bf40 did not build
Logs:
     /Users/username/Library/Logs/Homebrew/ogre2.2/00.options.out
     /Users/username/Library/Logs/Homebrew/ogre2.2/01.cmake.cc
     /Users/username/Library/Logs/Homebrew/ogre2.2/01.cmake
     /Users/username/Library/Logs/Homebrew/ogre2.2/02.make.cc
     /Users/username/Library/Logs/Homebrew/ogre2.2/build
     /Users/username/Library/Logs/Homebrew/ogre2.2/02.make

~~~
brew install ignition-fortress --verbose
brew gist-logs --new-issue ignition-fortress                                                                                                    
Error: No logs.

i put stuff from loggs files here 

https://gist.github.com/devashishRaj/337fa855dad83e9bfbb24914f5c33d1e

Do mention if i missed something 
~~~

then include the link in the issue. Thanks!

scpeters, 2023-09-22:
can you try again? I just merged some fixes for fortress in #1823, #2414, #2415

scpeters, 2023-09-23:
> can you try again? I just merged some fixes for fortress in #1823, #2414, #2415

you should be able to compile fortress now, but as noted in the description of https://github.com/osrf/homebrew-simulation/pull/2415, I don't expect `ign gazebo -g` to work yet

devashishRaj, 2023-09-23:
harmonic is working pretty good via brew binary , just a UI bug that text/values are not visible unless i select/click on them and i don't know on which repo to report to

closed 2023-09-24 by devashishRaj