osrf/homebrew-simulation #2478: gz-harmonic failing to build on macOS arm, due to ogre failure

henriquenunez, 2023-10-19:
HOMEBREW_VERSION: 4.1.16
ORIGIN: https://github.com/Homebrew/brew
HEAD: c0c8a4dd4dd19ec16deb429abd50cdf82ea33b27
Last commit: 5 days ago
Core tap JSON: 19 Oct 12:20 UTC
HOMEBREW_PREFIX: /opt/homebrew
HOMEBREW_CASK_OPTS: []
HOMEBREW_DISPLAY: /private/tmp/com.apple.launchd.01KX0mygsH/org.xquartz:0
HOMEBREW_MAKE_JOBS: 8
HOMEBREW_SORBET_RUNTIME: set
Homebrew Ruby: 2.6.10 => /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/bin/ruby
CPU: octa-core 64-bit arm_firestorm_icestorm
Clang: 14.0.3 build 1403
Git: 2.39.2 => /Applications/Xcode.app/Contents/Developer/usr/bin/git
Curl: 7.84.0 => /usr/bin/curl
macOS: 13.0.1-arm64
CLT: 14.3.1.0.1.1683849156
Xcode: 14.3.1
Rosetta 2: false

Error (I couldnt gist the logs):
```

Undefined symbols for architecture arm64:
  "_FreeImage_AcquireMemory", referenced from:
      Ogre::FreeImageCodec::encode(Ogre::SharedPtr<Ogre::MemoryDataStream>&, Ogre::SharedPtr<Ogre::Codec::CodecData>&) const in OgreFreeImageCodec.cpp.o
  "_FreeImage_AllocateT", referenced from:
      Ogre::FreeImageCodec::encodeBitmap(Ogre::SharedPtr<Ogre::MemoryDataStream>&, Ogre::SharedPtr<Ogre::Codec::CodecData>&) const in OgreFreeImageCodec.cpp.o
  "_FreeImage_CloseMemory", referenced from:
      Ogre::FreeImageCodec::encode(Ogre::SharedPtr<Ogre::MemoryDataStream>&, Ogre::SharedPtr<Ogre::Codec::CodecData>&) const in OgreFreeImageCodec.cpp.o
      Ogre::FreeImageCodec::decode(Ogre::SharedPtr<Ogre::DataStream>&) const in OgreFreeImageCodec.cpp.o
      Ogre::FreeImageCodec::magicNumberToFileExt(char const*, unsigned long) const in OgreFreeImageCodec.cpp.o
  "_FreeImage_ConvertTo24Bits", referenced from:
      Ogre::FreeImageCodec::decode(Ogre::SharedPtr<Ogre::DataStream>&) const in OgreFreeImageCodec.cpp.o
  "_FreeImage_ConvertTo32Bits", referenced from:
      Ogre::FreeImageCodec::decode(Ogre::SharedPtr<Ogre::DataStream>&) const in OgreFreeImageCodec.cpp.o
  "_FreeImage_ConvertToGreyscale", referenced from:
      Ogre::FreeImageCodec::encodeBitmap(Ogre::SharedPtr<Ogre::MemoryDataStream>&, Ogre::SharedPtr<Ogre::Codec::CodecData>&) const in OgreFreeImageCodec.cpp.o
      Ogre::FreeImageCodec::decode(Ogre::SharedPtr<Ogre::DataStream>&) const in OgreFreeImageCodec.cpp.o
  "_FreeImage_DeInitialise", referenced from:
      Ogre::FreeImageCodec::shutdown() in OgreFreeImageCodec.cpp.o
  "_FreeImage_FIFSupportsExportBPP", referenced from:
      Ogre::FreeImageCodec::encodeBitmap(Ogre::SharedPtr<Ogre::MemoryDataStream>&, Ogre::SharedPtr<Ogre::Codec::CodecData>&) const in OgreFreeImageCodec.cpp.o
  "_FreeImage_FIFSupportsExportType", referenced from:
      Ogre::FreeImageCodec::encodeBitmap(Ogre::SharedPtr<Ogre::MemoryDataStream>&, Ogre::SharedPtr<Ogre::Codec::CodecData>&) const in OgreFreeImageCodec.cpp.o
  "_FreeImage_GetBPP", referenced from:
      Ogre::FreeImageCodec::decode(Ogre::SharedPtr<Ogre::DataStream>&) const in OgreFreeImageCodec.cpp.o
  "_FreeImage_GetBits", referenced from:
      Ogre::FreeImageCodec::encodeBitmap(Ogre::SharedPtr<Ogre::MemoryDataStream>&, Ogre::SharedPtr<Ogre::Codec::CodecData>&) const in OgreFreeImageCodec.cpp.o
      Ogre::FreeImageCodec::decode(Ogre::SharedPtr<Ogre::DataStream>&) const in OgreFreeImageCodec.cpp.o
  "_FreeImage_GetColorType", referenced from:
      Ogre::FreeImageCodec::decode(Ogre::SharedPtr<Ogre::DataStream>&) const in OgreFreeImageCodec.cpp.o
  "_FreeImage_GetCopyrightMessage", referenced from:
      Ogre::FreeImageCodec::startup() in OgreFreeImageCodec.cpp.o
  "_FreeImage_GetFIFCount", referenced from:
      Ogre::FreeImageCodec::startup() in OgreFreeImageCodec.cpp.o
  "_FreeImage_GetFIFExtensionList", referenced from:
      Ogre::FreeImageCodec::startup() in OgreFreeImageCodec.cpp.o
  "_FreeImage_GetFileTypeFromMemory", referenced from:
      Ogre::FreeImageCodec::magicNumberToFileExt(char const*, unsigned long) const in OgreFreeImageCodec.cpp.o
  "_FreeImage_GetFormatFromFIF", referenced from:
      Ogre::FreeImageLoadErrorHandler(FREE_IMAGE_FORMAT, char const*) in OgreFreeImageCodec.cpp.o
      Ogre::FreeImageCodec::magicNumberToFileExt(char const*, unsigned long) const in OgreFreeImageCodec.cpp.o
  "_FreeImage_GetGreenMask", referenced from:
      Ogre::FreeImageCodec::decode(Ogre::SharedPtr<Ogre::DataStream>&) const in OgreFreeImageCodec.cpp.o
  "_FreeImage_GetHeight", referenced from:
      Ogre::FreeImageCodec::decode(Ogre::SharedPtr<Ogre::DataStream>&) const in OgreFreeImageCodec.cpp.o
  "_FreeImage_GetImageType", referenced from:
      Ogre::FreeImageCodec::decode(Ogre::SharedPtr<Ogre::DataStream>&) const in OgreFreeImageCodec.cpp.o
  "_FreeImage_GetPitch", referenced from:
[... 78 lines cut ...]
      Ogre::ZipDataStream::read(void*, unsigned long) in OgreZip.cpp.o
  "_zzip_error", referenced from:
      Ogre::ZipArchive::open(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, bool) const in OgreZip.cpp.o
  "_zzip_file_close", referenced from:
      Ogre::ZipDataStream::~ZipDataStream() in OgreZip.cpp.o
      Ogre::ZipDataStream::close() in OgreZip.cpp.o
  "_zzip_file_open", referenced from:
      Ogre::ZipArchive::open(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, bool) const in OgreZip.cpp.o
  "_zzip_file_read", referenced from:
      Ogre::ZipDataStream::read(void*, unsigned long) in OgreZip.cpp.o
  "_zzip_seek", referenced from:
      Ogre::ZipDataStream::skip(long) in OgreZip.cpp.o
      Ogre::ZipDataStream::seek(unsigned long) in OgreZip.cpp.o
  "_zzip_strerror_of", referenced from:
      Ogre::ZipDataStream::read(void*, unsigned long) in OgreZip.cpp.o
  "_zzip_tell", referenced from:
      Ogre::ZipDataStream::tell() const in OgreZip.cpp.o
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [lib/macosx/libOgreMain.1.9.0.dylib] Error 1
make[1]: *** [OgreMain/CMakeFiles/OgreMain.dir/all] Error 2
make: *** [all] Error 2
```

I tried changing the `-DCMAKE_OSX_ARCHITECTURES` flag to `arm64` instead of the default `x86-64`, but no success with that. I also saw a couple other issues with trouble building on apple silicon, but they seemed unrelated.

scpeters, 2023-11-20:
Maybe you can try without ogre1.9? Does it work if you comment out the following line?

https://github.com/osrf/homebrew-simulation/blob/master/Formula/gz-rendering8.rb#L26

rlrs, 2024-01-03:
Ogre2.3 also fails:

```
[ 94%] Built target RenderSystem_GL3Plus
make: *** [all] Error 2

HOMEBREW_VERSION: 4.2.2
ORIGIN: https://github.com/Homebrew/brew
HEAD: 705d2564ddfb9c915ee9425dd9ed408bfac61005
Last commit: 2 days ago
Core tap JSON: 03 Jan 10:26 UTC
Core cask tap JSON: 03 Jan 10:26 UTC
HOMEBREW_PREFIX: /opt/homebrew
HOMEBREW_CASK_OPTS: []
HOMEBREW_DISPLAY: /private/tmp/com.apple.launchd.JcsNn6Qqu3/org.xquartz:0
HOMEBREW_MAKE_JOBS: 10
HOMEBREW_NO_BOOTSNAP: set
Homebrew Ruby: 3.1.4 => /opt/homebrew/Library/Homebrew/vendor/portable-ruby/3.1.4/bin/ruby
CPU: 10-core 64-bit arm_firestorm_icestorm
Clang: 15.0.0 build 1500
Git: 2.39.3 => /Applications/Xcode.app/Contents/Developer/usr/bin/git
Curl: 8.4.0 => /usr/bin/curl
macOS: 14.2.1-arm64
CLT: 15.1.0.0.1.1700200546
Xcode: 15.1
Rosetta 2: false
```

piaosn, 2024-04-07:
I had the same issue when I try to install gz-garden.

"_zzip_file_read", referenced from:
      Ogre::ZipDataStream::read(void*, unsigned long) in OgreZip.cpp.o
  "_zzip_seek", referenced from:
      Ogre::ZipDataStream::skip(long) in OgreZip.cpp.o
      Ogre::ZipDataStream::seek(unsigned long) in OgreZip.cpp.o
  "_zzip_strerror_of", referenced from:
      Ogre::ZipDataStream::read(void*, unsigned long) in OgreZip.cpp.o
  "_zzip_tell", referenced from:
      Ogre::ZipDataStream::tell() const in OgreZip.cpp.o
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [lib/macosx/libOgreMain.1.9.0.dylib] Error 1
make[1]: *** [OgreMain/CMakeFiles/OgreMain.dir/all] Error 2
make: *** [all] Error 2

The same result with both x86_64 and arm64.

henriquenunez, 2024-04-11:
> I had the same issue when I try to install gz-garden.
> 
> "_zzip_file_read", referenced from: Ogre::ZipDataStream::read(void*, unsigned long) in OgreZip.cpp.o "_zzip_seek", referenced from: Ogre::ZipDataStream::skip(long) in OgreZip.cpp.o Ogre::ZipDataStream::seek(unsigned long) in OgreZip.cpp.o "_zzip_strerror_of", referenced from: Ogre::ZipDataStream::read(void*, unsigned long) in OgreZip.cpp.o "_zzip_tell", referenced from: Ogre::ZipDataStream::tell() const in OgreZip.cpp.o ld: symbol(s) not found for architecture arm64 clang: error: linker command failed with exit code 1 (use -v to see invocation) make[2]: *** [lib/macosx/libOgreMain.1.9.0.dylib] Error 1 make[1]: *** [OgreMain/CMakeFiles/OgreMain.dir/all] Error 2 make: *** [all] Error 2
> 
> The same result with both x86_64 and arm64.
> 
> However, the installation was successful when I used iTerm2 with Rosetta 2 enabled.

But I guess we want to unlock full native speed specially when simulating, don't we?

palahmd, 2025-02-16:
Could anyone find a fix yet? I am also trying to install ogre1.9 on my M1 Pro.

ingmarvdsande, 2025-04-08:
> Could anyone find a fix yet? I am also trying to install ogre1.9 on my M1 Pro.

Yes, actually. If you're still having issues, switch your terminal to Rosetta and use Brew (you'll need to install brew for X86 installations) to install ogre1.9 and ogre2.3. Then you can switch it back to arm64 to install Gazebo (or whatever else you're gonna use Ogre for)

m0sviatoslav, 2025-04-14:
@ingmarvdsande I installed ogre1.9 and ogre2.3 under rosseta. 

But it required to install under x86_64
```
❯ brew install ogre1.9
Error: Cannot install under Rosetta 2 in ARM default prefix (/opt/homebrew)!
To rerun under ARM use:
    arch -arm64 brew install ...
To install under x86_64, install Homebrew into /usr/local.
```

and so I did `❯ /bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/HEAD/install.sh)"`

and after successfully installing ogre1.9 and ogre2.3. 

But when I switch back to arm terminal, brew tries to install ogre1.9 and ogre2.3 

in rosetta terminal
```
❯ brew list ogre1.9
/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/bin/OgreMeshUpgrader
/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/bin/OgreXMLConverter
/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/include/OGRE/ (525 files)
/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreMain.1.9.0.dylib
/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreOverlay.1.9.0.dylib
/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgrePaging.1.9.0.dylib
/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreProperty.1.9.0.dylib
/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreRTShaderSystem.1.9.0.dylib
/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreTerrain.1.9.0.dylib
/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreVolume.1.9.0.dylib
/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/OGRE/ (24 files)
/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/pkgconfig/ (8 files)
/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/ (7 other files)
/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/sbom.spdx.json
/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/share/OGRE/ (11 files)
```

ingmarvdsande, 2025-04-15:
@mOsviatoslav 

Try running 'brew install gz-ionic' in X86 as well. That should fix it

scpeters, 2025-08-25:
we now have arm bottles for ogre