osrf/homebrew-simulation #2549: gz-transport13 fails to find Python3

rgov, 2024-02-05:
On macOS Sonoma 14.3,  installing gz-transport13 fails. If I'm understanding the error, maybe it's because it doesn't think it's the "development" package of Python?

```
-- Searching for Python - found version 3.11.7.
...
CMake Error at /opt/homebrew/Cellar/cmake/3.28.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Python3 (missing: Development Development.Module
  Development.Embed) (found version "3.12")
Call Stack (most recent call first):
  /opt/homebrew/Cellar/cmake/3.28.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  /opt/homebrew/Cellar/cmake/3.28.2/share/cmake/Modules/FindPython/Support.cmake:3867 (find_package_handle_standard_args)
  /opt/homebrew/Cellar/cmake/3.28.2/share/cmake/Modules/FindPython3.cmake:545 (include)
  CMakeLists.txt:151 (find_package)
```

```
HOMEBREW_VERSION: 4.2.6-119-gd947721
ORIGIN: https://github.com/Homebrew/brew
HEAD: d94772171f8be4885ce2f9ef018989c9955b324d
Last commit: 10 hours ago
Core tap HEAD: 96429d0c4d7356e72396b83523ea4c74a4eb3253
Core tap last commit: 9 months ago
Core tap branch: colmap
Core tap JSON: 05 Feb 05:16 UTC
Core cask tap HEAD: a771466c1d4ecce5ed03d1af1708147de31aecce
Core cask tap last commit: 1 year, 4 months ago
Core cask tap branch: bump-foxglove-studio-1.27.0
Core cask tap JSON: 05 Feb 05:16 UTC
HOMEBREW_PREFIX: /opt/homebrew
HOMEBREW_CASK_OPTS: []
HOMEBREW_DISPLAY: /private/tmp/com.apple.launchd.oCiqzjI50D/org.xquartz:0
HOMEBREW_GITHUB_API_TOKEN: set
HOMEBREW_MAKE_JOBS: 10
HOMEBREW_SORBET_RUNTIME: set
Homebrew Ruby: 3.1.4 => /opt/homebrew/Library/Homebrew/vendor/portable-ruby/3.1.4/bin/ruby
CPU: 10-core 64-bit arm_firestorm_icestorm
Clang: 15.0.0 build 1500
Git: 2.39.3 => /Applications/Xcode.app/Contents/Developer/usr/bin/git
Curl: 8.4.0 => /usr/bin/curl
macOS: 14.3-arm64
CLT: 15.1.0.0.1.1700200546
Xcode: 15.1
Rosetta 2: false
```

scpeters, 2024-02-06:
I think this is more an issue with how gz-transport is finding python and that it should be fixed by https://github.com/gazebosim/gz-transport/pull/472 without any additional changes to the homebrew formulae

rgov, 2024-02-06:
We ought to verify that CMake is correctly finding the python@3.11 keg's development headers/library and not defaulting to another Python like python@3.12.

rgov, 2024-02-06:
I applied https://github.com/gazebosim/gz-transport/pull/472 alone, without #2550, and ran CMake from within a Homebrew debug shell:

```
$ cmake .. -DCMAKE_INSTALL_PREFIX=/opt/homebrew/Cellar/gz-transport14/13.999.999-0-20231016 -DCMAKE_INSTALL_LIBDIR=lib -DCMAKE_BUILD_TYPE=Release -DCMAKE_FIND_FRAMEWORK=LAST -DCMAKE_VERBOSE_MAKEFILE=ON -Wno-dev -DBUILD_TESTING=OFF -DCMAKE_OSX_SYSROOT=/Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk -DBUILD_TESTING=OFF -DPython3_EXECUTABLE=/opt/homebrew/opt/python@3.11/bin/python3
...
-- ====== Finding Dependencies ======
-- Found Python3: /opt/homebrew/opt/python@3.11/bin/python3 (found version "3.11.7") found components: Interpreter Development Development.Module Development.Embed 
...
-- Python3_INCLUDE_DIRS = /opt/homebrew/opt/python@3.11/Frameworks/Python.framework/Versions/3.11/include/python3.11
-- Python3_LIBRARIES = /opt/homebrew/opt/python@3.11/Frameworks/Python.framework/Versions/3.11/lib/libpython3.11.dylib
-- Python3_LIBRARY_DIRS = /opt/homebrew/opt/python@3.11/Frameworks/Python.framework/Versions/3.11/lib
```

closed 2024-02-06 by rgov

rgov, 2024-02-12:
Hi @maisternia. If you use my fork, you'll want to check out the `rzg/python3` branch. Like:

```
$ cd $(brew --prefix)/Library/Taps/rgov/homebrew-simulation
$ git checkout rzg/python3
```

However please note I will not be keeping my fork in sync with upstream, which will eventually resolve this issue, so I recommend this only as a short term fix. When you want to switch back to the upstream, please untap and re-tap.

maisternia, 2024-02-13:
That worked! Also I had to `brew --HEAD sdformat14` as well, but stuck on **gz-physics7** with `fatal error: 'sdf/Joint.hh' file not found`. But this is another kind of animal, will to dig further.

Thank you so much @rgov!