osrf/homebrew-simulation #2555: Failed to install Gazebo Harmonic on macOS

atinfinity, 2024-02-12:
I read <https://gazebosim.org/docs/harmonic/install_osx> to install `Gazebo Harmonic` on macOS.
And, I add `ulimit -n unlimited` to avoid <https://github.com/gazebosim/gz-sim/issues/2215>.

```shell
ulimit -n unlimited
brew tap osrf/simulation
brew install gz-harmonic
```

As a result, the following error occured.

```shell
Last 15 lines from /Users/dandelion/Library/Logs/Homebrew/gz-transport13/01.cmake:
-- Performing Test HAS_FLTO
-- Performing Test HAS_FLTO - Success
-- Performing Test HAS_FLTO_THIN
-- Performing Test HAS_FLTO_THIN - Success
CMake Error at /opt/homebrew/Cellar/cmake/3.28.3/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Python3 (missing: Development Development.Module
  Development.Embed) (found version "3.12")
Call Stack (most recent call first):
  /opt/homebrew/Cellar/cmake/3.28.3/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  /opt/homebrew/Cellar/cmake/3.28.3/share/cmake/Modules/FindPython/Support.cmake:3867 (find_package_handle_standard_args)
  /opt/homebrew/Cellar/cmake/3.28.3/share/cmake/Modules/FindPython3.cmake:545 (include)
  CMakeLists.txt:151 (find_package)

-- Configuring incomplete, errors occurred!

If reporting this issue please do so at (not Homebrew/brew or Homebrew/homebrew-core):
  https://github.com/osrf/homebrew-simulation/issues
```

## brew configuration

```shell
$ brew config
HOMEBREW_VERSION: 4.2.7
ORIGIN: https://github.com/Homebrew/brew
HEAD: 6170b6297c61ef0b68814c708eeff2199d8b0d11
Last commit: 7 days ago
Core tap JSON: 12 Feb 16:40 UTC
Core cask tap JSON: 12 Feb 16:24 UTC
HOMEBREW_PREFIX: /opt/homebrew
HOMEBREW_CASK_OPTS: []
HOMEBREW_DISPLAY: /private/tmp/com.apple.launchd.U1q1efnuAt/org.xquartz:0
HOMEBREW_MAKE_JOBS: 8
Homebrew Ruby: 3.1.4 => /opt/homebrew/Library/Homebrew/vendor/portable-ruby/3.1.4/bin/ruby
CPU: octa-core 64-bit arm_firestorm_icestorm
Clang: 15.0.0 build 1500
Git: 2.39.3 => /Applications/Xcode.app/Contents/Developer/usr/bin/git
Curl: 8.4.0 => /usr/bin/curl
macOS: 14.3-arm64
CLT: 15.1.0.0.1.1700200546
Xcode: 15.2
Rosetta 2: false
```

scpeters, 2024-02-13:
I just added patches to fix finding python in #2563. Can you try again?

cirocavani, 2024-02-14:
Hi @scpeters 

I had the same error yesterday. It's working for me now (10GMT). But some projects still find 3.12 instead of 3.11 (even with the `-DPython3_EXECUTABLE`. `gz-math7` for example). But as yesterday, it's out of skill right know. So thank you very much for this work, it helps a lot for beginners like me.

The problem I found trying to understand the error yesterday was to look at the cmake logs and some of then still have 3.12 as python found.

```sh
grep "3\.12" -r ~/Library/Logs/Homebrew/gz-*/01.cmake

# /Users/ciro.cavani/Library/Logs/Homebrew/gz-fuel-tools9/01.cmake:-- Found Python3: /opt/homebrew/bin/python3.12 (found version "3.12.2") found components: Interpreter 
# /Users/ciro.cavani/Library/Logs/Homebrew/gz-gui8/01.cmake:-- Found Python3: /opt/homebrew/bin/python3.12 (found version "3.12.2") found components: Interpreter 
# /Users/ciro.cavani/Library/Logs/Homebrew/gz-harmonic/01.cmake:-- Found Python3: /opt/homebrew/bin/python3.12 (found version "3.12.2") found components: Interpreter 
# /Users/ciro.cavani/Library/Logs/Homebrew/gz-launch7/01.cmake:-- Found Python3: /opt/homebrew/bin/python3.12 (found version "3.12.2") found components: Interpreter 
# /Users/ciro.cavani/Library/Logs/Homebrew/gz-math7/01.cmake:-- Searching for Python - found version 3.12.2.
# /Users/ciro.cavani/Library/Logs/Homebrew/gz-sensors8/01.cmake:-- Found Python3: /opt/homebrew/bin/python3.12 (found version "3.12.2") found components: Interpreter 
```

How I tested (as today):

```sh
brew reinstall python@3.12
brew reinstall python@3.11
brew link --overwrite python@3.11
which python3
# /opt/homebrew/bin/python3
python3 --version
# Python 3.11.7
which python
# /opt/homebrew/bin/python
python --version
# Python 3.11.7

for f in gz-launch7 gz-sim8 gz-sensors8 gz-physics7 sdformat14 gz-gui8 gz-transport13 gz-rendering8 ogre2.3 ogre1.9 gz-plugin2 gz-fuel-tools9 gz-msgs10 gz-tools2 gz-common5 gz-math7 gz-utils2 gz-cmake3 gz-harmonic
do
  echo $f
  brew uninstall --force --ignore-dependencies $f 2> /dev/null
  brew cleanup -s $f
  find ~/Library/Caches/Homebrew -name "*$f*" -delete
  rm -rf ~/Library/Logs/Homebrew/$f
done

brew list | grep gz
# empty

brew tap osrf/simulation
brew update && brew upgrade
brew install gz-harmonic

# ==> Fetching dependencies for osrf/simulation/gz-harmonic: osrf/simulation/gz-cmake3, osrf/simulation/gz-utils2, osrf/simulation/gz-math7, osrf/simulation/gz-common5, osrf/simulation/gz-tools2, osrf/simulation/gz-msgs10, osrf/simulation/gz-fuel-tools9, osrf/simulation/gz-plugin2, osrf/simulation/ogre1.9, osrf/simulation/ogre2.3, osrf/simulation/gz-rendering8, osrf/simulation/gz-transport13, osrf/simulation/gz-gui8, osrf/simulation/sdformat14, osrf/simulation/gz-physics7, osrf/simulation/gz-sensors8, osrf/simulation/gz-sim8 and osrf/simulation/gz-launch7
#
# ...
#
# ==> Installing osrf/simulation/gz-harmonic
# ==> cmake ..
# ==> make install
# ==> /opt/homebrew/opt/python@3.11/libexec/bin/python -m venv --system-site-packages --without-pip /opt/homebrew/Cellar/gz-harmonic/1.0.0_3/libexec
# ==> /opt/homebrew/opt/python@3.11/libexec/bin/python -m pip --python=/opt/homebrew/Cellar/gz-harmonic/1.0.0_3/libexec/bin/python install PyYAML
# ==> /opt/homebrew/opt/python@3.11/libexec/bin/python -m pip --python=/opt/homebrew/Cellar/gz-harmonic/1.0.0_3/libexec/bin/python install vcstool
# 🍺  /opt/homebrew/Cellar/gz-harmonic/1.0.0_3: 93 files, 627.5KB, built in 1 minute 13 seconds
# ==> Running `brew cleanup gz-harmonic`...
# Disable this behaviour by setting HOMEBREW_NO_INSTALL_CLEANUP.
# Hide these hints with HOMEBREW_NO_ENV_HINTS (see `man brew`).
```

atinfinity, 2024-02-14:
@scpeters I was able to install Gazebo Harmonic. And, I also encountered <https://github.com/osrf/homebrew-simulation/issues/2555#issuecomment-1943564011>.

atinfinity, 2024-02-15:
@scpeters Thank you for your explanation. I understood. So, I closed this issue.

closed 2024-02-15 by atinfinity

scpeters, 2024-02-15:
glad it's working for you!

skol101, 2024-09-10:
Fails on M2 with very strange doxygen error. I'm running standard Macos binary installation instruction
```
==> Installing gz-harmonic from osrf/simulation
==> Installing dependencies for osrf/simulation/gz-harmonic: osrf/simulation/gz-math7, ossp-uuid, tinyxml2, osrf/simulation/gz-common5, osrf/simulation/gz-tools2, osrf/simulation/gz-msgs10, jsoncpp, libzip, osrf/simulation/gz-fuel-tools9, osrf/simulation/gz-plugin2, boost, libzzip, hwloc, tbb, osrf/simulation/ogre1.9, osrf/simulation/ogre2.3, osrf/simulation/gz-rendering8, cppzmq, osrf/simulation/gz-transport13, qt@5, osrf/simulation/gz-gui8, bullet, libccd, octomap, fcl, flann, fmt, ampl-mp, ipopt, nlopt, ode, open-scene-graph, spdlog, console_bridge, urdfdom_headers, urdfdom, dartsim, google-benchmark, osrf/simulation/sdformat14, osrf/simulation/gz-physics7, osrf/simulation/gz-sensors8, osrf/simulation/gz-sim8 and osrf/simulation/gz-launch7
==> Installing osrf/simulation/gz-harmonic dependency: osrf/simulation/gz-math7
==> cmake .. -DCMAKE_INSTALL_RPATH=@loader_path/../lib -DPython3_EXECUTABLE=/opt/homebrew/opt/python@3.12/bin/python3
==> make install
Last 15 lines from /Users/funkymonkey/Library/Logs/Homebrew/gz-math7/02.make:
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
```

scpeters, 2024-11-10:
@skol101 I don't think the error is from doxygen; perhaps there was another error earlier on the console log?