osrf/homebrew-simulation #2580: Use python@3.12 for formulae with python bindings

scpeters, 2024-02-23:
Now that `python@3.12` is the default `python3` formula (see https://github.com/Homebrew/homebrew-core/pull/150390), we should update our formulae to use python@3.12 for the formulae with python bindings.

* [ ] Ionic
* [ ] Harmonic
* [ ] Garden

scpeters, 2024-04-15:
> Is it some relationship between this and [PX4/PX4-Autopilot#22826](https://github.com/PX4/PX4-Autopilot/issues/22826) ?

gazebo-classic doesn't use python bindings, so I don't think this is related to your PX4 issue

mengchaoheng, 2024-04-16:
> gazebo-classic doesn't use python bindings,

I mean, as a px4 user, when I use gazebo11 for simulation, it can work normally even if my python3 environment is not python3.12, right? @scpeters 

Sorry, I am currently troubled by px4 gazebo simulation problems. All compilation tasks are completed but gazebo fails.

scpeters, 2024-04-16:
> I mean, as a px4 user, when I use gazebo11 for simulation, it can work normally even if my python3 environment is not python3.12, right? @scpeters
>
> Sorry, I am currently troubled by px4 gazebo [...]

gazebo11 does not directly use python. Unfortunately I can't answer questions about px4, so I recommend asking in a more relevant forum.

closed 2024-09-03 by scpeters