osrf/homebrew-simulation #2585: osrf/simulation/gz-physics7 7.0.0 did not build

e-taha, 2024-03-08:
I was trying to install Gazebo-harmonic on my Macbook Air M1. I run the commands like in the documentation:

~~~
brew tap osrf/simulation
brew install gz-harmonic
~~~

But then I received an error: 

The maximum number of open files on this system has been reached. Use `ulimit -n` to increase this limit.

I wrote `ulimit -n 512`  and re-run the `brew install gz-harmonic` command, but I received an error. The last 3 lines were:
~~~
[ 98%] Built target gz-physics7-tpe-plugin
make[1]: *** [dartsim/CMakeFiles/gz-physics7-dartsim-plugin.dir/all] Error 2
make: *** [all] Error 2
~~~

Then I re-run the command with verbose flag:
~~~
brew install gz-harmonic --verbose
~~~

It gave me the following error:

`Error: osrf/simulation/gz-physics7 7.0.0 did not build`

Brew configuration:

~~~
HOMEBREW_VERSION: 4.2.11-94-g815db06
ORIGIN: https://github.com/Homebrew/brew
HEAD: 815db0637a350db412fb951d74d2ad1c15906704
Last commit: 4 hours ago
Core tap JSON: 08 Mar 13:02 UTC
Core cask tap JSON: 08 Mar 13:02 UTC
HOMEBREW_PREFIX: /opt/homebrew
HOMEBREW_CASK_OPTS: []
HOMEBREW_DISPLAY: /private/tmp/com.apple.launchd.eor5c72yC2/org.xquartz:0
HOMEBREW_MAKE_JOBS: 8
HOMEBREW_SORBET_RUNTIME: set
Homebrew Ruby: 3.1.4 => /opt/homebrew/Library/Homebrew/vendor/portable-ruby/3.1.4/bin/ruby
CPU: octa-core 64-bit arm_firestorm_icestorm
Clang: 15.0.0 build 1500
Git: 2.39.1 => /opt/homebrew/bin/git
Curl: 8.4.0 => /usr/bin/curl
macOS: 14.3.1-arm64
CLT: 15.3.0.0.1.1708646388
Xcode: 15.2
Rosetta 2: false
~~~

Output link of `brew gist-logs gz-physics7` command (`brew gist-logs gz-harmonic` didn't create anything)

https://gist.github.com/e-taha/274dfb45b1ad23fc61564231cc41d538

scpeters, 2024-03-11:
the following two lines stand out to me:

https://gist.github.com/e-taha/274dfb45b1ad23fc61564231cc41d538#file-02-make-L223-L224

~~~
make[2]: *** No rule to make target `/Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk/usr/lib/libm.tbd', needed by `lib/libgz-physics7-dartsim-plugin.7.0.0.dylib'.  Stop.
make[2]: *** Waiting for unfinished jobs....
~~~

Especially seeing `MacOSX12.sdk` in the file path, while your `brew config` shows macOS 14. Perhaps your command-lint tools installation is out of date?

e-taha, 2024-03-12:
I re-installed the command line tools, but I get the same error. I checked the `/Library/Developer/CommandLineTools/SDKs` file and `MacOSX12.sdk` folder doesn't exist there. The folders that exist are MacOSX14.sdk, MacOSX14.4.sdk, MacOSX13.sdk, MacOSX13.3.sdk and MacOSX.sdk. Do I need to install an older version of sdk?

scpeters, 2024-04-09:
after reinstalling, do the error logs still reference that MacOSX12.sdk path?

scpeters, 2024-05-13:
it's been a while since the last response; I'll close this, but reopen if the issue persists

closed 2024-05-13 by scpeters