osrf/homebrew-simulation #2638: osrf/simulation/gazebo11 build error

RobertoD91, 2024-04-13:
Build error with gazebo11 installation

~~~
$ brew config
HOMEBREW_VERSION: 4.2.17-92-gcfadc89
ORIGIN: https://github.com/Homebrew/brew.git
HEAD: cfadc894e52236483328185f676a747e30439473
Last commit: 24 minutes ago
Core tap HEAD: 77ff6055363300271a18ea94e00ddda8a2c9cab3
Core tap last commit: 16 hours ago
Core tap JSON: 12 Apr 15:19 UTC
Core cask tap HEAD: 8e46538763d7958c3ba5467ab1d5184933e1526e
Core cask tap last commit: 16 hours ago
Core cask tap JSON: 12 Apr 15:22 UTC
HOMEBREW_PREFIX: /usr/local
HOMEBREW_CASK_OPTS: []
HOMEBREW_DISPLAY: /private/tmp/com.apple.launchd.SdnaszHhDE/org.xquartz:0
HOMEBREW_EDITOR: nano
HOMEBREW_GITHUB_API_TOKEN: set
HOMEBREW_MAKE_JOBS: 16
HOMEBREW_SORBET_RUNTIME: set
Homebrew Ruby: 3.1.4 => /usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/3.1.4/bin/ruby
CPU: 16-core 64-bit kabylake
Clang: 15.0.0 build 1500
Git: 2.44.0 => /usr/local/bin/git
Curl: 8.4.0 => /usr/bin/curl
macOS: 14.4.1-x86_64
CLT: 15.3.0.0.1.1708646388
Xcode: 15.3
~~~

~~~
brew install px4-sim --verbose # it install osrf/simulation/gazebo11 as dep
brew gist-logs osrf/simulation/gazebo11
~~~

https://gist.github.com/RobertoD91/b7ec90b4dd256cf93e44fc6e41abffd2

mengchaoheng, 2024-04-15:
Any suggestions for https://github.com/PX4/PX4-Autopilot/issues/22826 ?

scpeters, 2024-04-15:
I looked in your gist and see the error messages related to an incompatibility with ffmpeg 7 that was recently patched in gazebo11:

~~~
/tmp/gazebo11-20240412-53945-tnwy5q/gazebo-11.14.0/gazebo/common/AudioDecoder.cc:146:27: error: no member named 'channels' in 'AVCodecContext'
          this->codecCtx->channels;
          ~~~~~~~~~~~~~~  ^
/tmp/gazebo11-20240412-53945-tnwy5q/gazebo-11.14.0/gazebo/common/AudioDecoder.cc:340:19: error: no member named 'channel_layout' in 'AVCodecContext'
  this->codecCtx->channel_layout   = par->channel_layout;
  ~~~~~~~~~~~~~~  ^
/tmp/gazebo11-20240412-53945-tnwy5q/gazebo-11.14.0/gazebo/common/AudioDecoder.cc:340:43: error: no member named 'channel_layout' in 'AVCodecParameters'
  this->codecCtx->channel_layout   = par->channel_layout;
                                     ~~~  ^
/tmp/gazebo11-20240412-53945-tnwy5q/gazebo-11.14.0/gazebo/common/AudioDecoder.cc:341:19: error: no member named 'channels' in 'AVCodecContext'
  this->codecCtx->channels         = par->channels;
  ~~~~~~~~~~~~~~  ^
/tmp/gazebo11-20240412-53945-tnwy5q/gazebo-11.14.0/gazebo/common/AudioDecoder.cc:341:43: error: no member named 'channels' in 'AVCodecParameters'
  this->codecCtx->channels         = par->channels;
                                     ~~~  ^
5 errors generated.
~~~

* https://github.com/gazebosim/gazebo-classic/pull/3380

the gazebo11 bottles were rebuilt in https://github.com/osrf/homebrew-simulation/pull/2636, so I believe this is now fixed.

closed 2024-04-15 by scpeters

RobertoD91, 2024-04-15:
> I believe this is now fixed.

It is, thank you.

mengchaoheng, 2024-04-16:
thank you !!!  @scpeters 
May I ask if I have used the latest fixed version?
```Console
➜  ~ brew install px4-sim-gazebo

==> Auto-updating Homebrew...
Adjust how often this is run with HOMEBREW_AUTO_UPDATE_SECS or disable with
HOMEBREW_NO_AUTO_UPDATE. Hide these hints with HOMEBREW_NO_ENV_HINTS (see `man brew`).
==> Auto-updated Homebrew!
==> Updated Homebrew from 40d2f5c23a to 5517894523.
Updated 2 taps (homebrew/core and homebrew/cask).
==> New Formulae
morpheus
==> New Casks
flox

You have 3 outdated formulae installed.

==> Fetching dependencies for px4/px4/px4-sim-gazebo: webp, qt, pyqt and osrf/simulation/gazebo11
==> Fetching webp
==> Downloading https://ghcr.io/v2/homebrew/core/webp/manifests/1.4.0
################################################################################### 100.0%
==> Downloading https://ghcr.io/v2/homebrew/core/webp/blobs/sha256:600311045d5469c75d84d6b
################################################################################### 100.0%
==> Fetching qt
==> Downloading https://ghcr.io/v2/homebrew/core/qt/manifests/6.6.2_2
Already downloaded: /Users/mch/Library/Caches/Homebrew/downloads/cf36261aa27289d96576c5b783876d6d099619320e54260030da953401be5e7b--qt-6.6.2_2.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/qt/blobs/sha256:97ae8277d876f50e1606dd8b5
Already downloaded: /Users/mch/Library/Caches/Homebrew/downloads/f9d275431f704a16398bac063a3d6aebd739ed615944eba3a30dddb742986070--qt--6.6.2_2.sonoma.bottle.tar.gz
==> Fetching pyqt
==> Downloading https://ghcr.io/v2/homebrew/core/pyqt/manifests/6.6.1
Already downloaded: /Users/mch/Library/Caches/Homebrew/downloads/2c411cb2b2b791a5d2844faf277f1149a2b14660757f8666c6cb15acab97a7eb--pyqt-6.6.1.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/pyqt/blobs/sha256:b073875b968656e535ecbfe
Already downloaded: /Users/mch/Library/Caches/Homebrew/downloads/e43b38c34dcccefb74fc0d4b7dcae5c0b028554ede75e56dd75d9d1c11e3c4b5--pyqt--6.6.1.sonoma.bottle.tar.gz
==> Fetching osrf/simulation/gazebo11
==> Downloading https://osrf-distributions.s3.amazonaws.com/bottles-simulation/gazebo11-11
Already downloaded: /Users/mch/Library/Caches/Homebrew/downloads/01e03cdb447de84fba8de9196dd5b687d50eefb381a6a3e5cc53066dbb691f58--gazebo11-11.14.0_18.ventura.bottle.tar.gz
==> Fetching px4/px4/px4-sim-gazebo
==> Downloading https://raw.githubusercontent.com/PX4/PX4-Autopilot/master/Tools/px4.py
Already downloaded: /Users/mch/Library/Caches/Homebrew/downloads/be4aff9b473b8b0c6bae0ff08a6611885de9d56ee3a477c05453bb6bbcaab246--px4.py
==> Installing px4-sim-gazebo from px4/px4
==> Installing dependencies for px4/px4/px4-sim-gazebo: webp, qt, pyqt and osrf/simulation/gazebo11
==> Installing px4/px4/px4-sim-gazebo dependency: webp
==> Downloading https://ghcr.io/v2/homebrew/core/webp/manifests/1.4.0
Already downloaded: /Users/mch/Library/Caches/Homebrew/downloads/4fd3ea2923bb7885c84ba3eee4738315675acd20fdff772153a63d8087daa4ad--webp-1.4.0.bottle_manifest.json
==> Pouring webp--1.4.0.sonoma.bottle.tar.gz
🍺  /usr/local/Cellar/webp/1.4.0: 63 files, 2.5MB
==> Installing px4/px4/px4-sim-gazebo dependency: qt
==> Downloading https://ghcr.io/v2/homebrew/core/qt/manifests/6.6.2_2
Already downloaded: /Users/mch/Library/Caches/Homebrew/downloads/cf36261aa27289d96576c5b783876d6d099619320e54260030da953401be5e7b--qt-6.6.2_2.bottle_manifest.json
==> Pouring qt--6.6.2_2.sonoma.bottle.tar.gz
🍺  /usr/local/Cellar/qt/6.6.2_2: 14,413 files, 624.2MB
==> Installing px4/px4/px4-sim-gazebo dependency: pyqt
==> Downloading https://ghcr.io/v2/homebrew/core/pyqt/manifests/6.6.1
Already downloaded: /Users/mch/Library/Caches/Homebrew/downloads/2c411cb2b2b791a5d2844faf277f1149a2b14660757f8666c6cb15acab97a7eb--pyqt-6.6.1.bottle_manifest.json
==> Pouring pyqt--6.6.1.sonoma.bottle.tar.gz
🍺  /usr/local/Cellar/pyqt/6.6.1: 1,220 files, 33.5MB
==> Installing px4/px4/px4-sim-gazebo dependency: osrf/simulation/gazebo11
==> Pouring gazebo11-11.14.0_18.ventura.bottle.tar.gz
🍺  /usr/local/Cellar/gazebo11/11.14.0_18: 1,297 files, 134.5MB
==> Installing px4/px4/px4-sim-gazebo
==> PX4 Gazebo simulation installed
Warning: Formula gst-libav was renamed to gstreamer.
Warning: Formula gst-plugins-bad was renamed to gstreamer.
Warning: Formula gst-plugins-base was renamed to gstreamer.
Warning: Formula gst-plugins-good was renamed to gstreamer.
Warning: Formula gst-plugins-ugly was renamed to gstreamer.
Warning: Formula gst-libav was renamed to gstreamer.
Warning: Formula gst-plugins-bad was renamed to gstreamer.
Warning: Formula gst-plugins-base was renamed to gstreamer.
[... 18 lines cut ...]
HOMEBREW_VERSION: 4.2.18-29-g5517894
ORIGIN: https://github.com/Homebrew/brew
HEAD: 5517894523620d2000f96013aa770b51dd73effc
Last commit: 85 minutes ago
Core tap HEAD: 5aacf8df4ff2465cded3b61e2be9b2369fc57e98
Core tap last commit: 6 weeks ago
Core tap JSON: 16 Apr 01:36 UTC
Core cask tap HEAD: ea9d870841f7a6be26a9537245c38bdbe40876c1
Core cask tap last commit: 6 weeks ago
Core cask tap JSON: 16 Apr 01:36 UTC
HOMEBREW_PREFIX: /usr/local
HOMEBREW_CASK_OPTS: []
HOMEBREW_DISPLAY: /private/tmp/com.apple.launchd.ieQqhddOTc/org.xquartz:0
HOMEBREW_MAKE_JOBS: 10
HOMEBREW_SORBET_RUNTIME: set
Homebrew Ruby: 3.1.4 => /usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/3.1.4/bin/ruby
CPU: 10-core 64-bit westmere
Clang: 15.0.0 build 1500
Git: 2.39.3 => /Library/Developer/CommandLineTools/usr/bin/git
Curl: 8.4.0 => /usr/bin/curl
macOS: 14.3.1-x86_64
CLT: 15.1.0.0.1.1700200546
Xcode: N/A
Rosetta 2: true
```

scpeters, 2024-04-16:
> May I ask if I have used the latest fixed version?

> ```
> ==> Pouring gazebo11-11.14.0_18.ventura.bottle.tar.gz
> 🍺  /usr/local/Cellar/gazebo11/11.14.0_18: 1,297 files, 134.5MB
> ```

you have installed revision 18, which was the revision bump in https://github.com/osrf/homebrew-simulation/pull/2636, so yes you have the latest version