osrf/homebrew-simulation #2646: building gazebo11 failed on Mac M1

user-2048, 2024-04-29:
Hi I'm trying to build gazebo11 on my Mac and it gets most of the way and then comes up with errors. Can I get any advice on ways to get gazebo (or similar ROS1 simulator) working? I'm fairly new to this, so sorry if I haven't made it clear what the issue is.

**After running with --verbose**
Error: osrf/simulation/ogre1.9 1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9 did not build
Logs:
     /Users/KarolinaMungai/Library/Logs/Homebrew/ogre1.9/00.options.out
     /Users/KarolinaMungai/Library/Logs/Homebrew/ogre1.9/01.cmake
     /Users/KarolinaMungai/Library/Logs/Homebrew/ogre1.9/01.cmake.cc
     /Users/KarolinaMungai/Library/Logs/Homebrew/ogre1.9/02.make
     /Users/KarolinaMungai/Library/Logs/Homebrew/ogre1.9/02.make.cc
     /Users/KarolinaMungai/Library/Logs/Homebrew/ogre1.9/build

**This is what comes up at the end when I try download.**
==> make install
Last 15 lines from /Users/KarolinaMungai/Library/Logs/Homebrew/ogre1.9/02.make:
20 errors generated.
20 errors generated.
make[2]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreAlignedAllocator.cpp.o] Error 1
20 errors generated.
20 errors generated.
20 errors generated.
make[2]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreAnimable.cpp.o] Error 1
make[2]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreAnimationState.cpp.o] Error 1
make[2]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreAtomicScalar.cpp.o] Error 1
make[2]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreAnimationTrack.cpp.o] Error 1
make[2]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreArchiveManager.cpp.o] Error 1
20 errors generated.
make[2]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreAnimation.cpp.o] Error 1
make[1]: *** [OgreMain/CMakeFiles/OgreMain.dir/all] Error 2
make: *** [all] Error 2

diysumit, 2024-05-03:
Getting same errors

`(base) ➜  ~ curl -ssL http://get.gazebosim.org | sh
=======================================================
GAZEBO INSTALLATION SCRIPT
=======================================================

This script is installing the latest stable version of
Gazebo Simulator available from your package manager

No receipt for 'org.macosforge.xquartz.pkg' found at '/'.
package-id: org.xquartz.X11
version: 2.8.5
volume: /
location: 
install-time: 1714714299
==> Updating Homebrew...
Already up-to-date.
==> Fetching dependencies for osrf/simulation/gazebo11: dartsim, jasper, graphviz, osrf/simulation/ignition-cmake2, pybind11, python@3.11, ruby, osrf/simulation/ignition-math6, osrf/simulation/ignition-common3, osrf/simulation/ignition-tools, osrf/simulation/ignition-msgs5, osrf/simulation/ignition-fuel-tools4, osrf/simulation/ignition-transport8, libtar, osrf/simulation/ogre1.9, qwt-qt5, osrf/simulation/tinyxml1, osrf/simulation/sdformat9 and osrf/simulation/simbody
==> Fetching dartsim
==> Downloading https://ghcr.io/v2/homebrew/core/dartsim/manifests/6.13.2_3
######################################################################### 100.0%
==> Downloading https://ghcr.io/v2/homebrew/core/dartsim/blobs/sha256:d482ac5d4f
######################################################################### 100.0%
==> Fetching jasper
==> Downloading https://ghcr.io/v2/homebrew/core/jasper/manifests/4.2.4
######################################################################### 100.0%
==> Downloading https://ghcr.io/v2/homebrew/core/jasper/blobs/sha256:a9c32c908bd
######################################################################### 100.0%
==> Fetching graphviz
==> Downloading https://ghcr.io/v2/homebrew/core/graphviz/manifests/11.0.0
######################################################################### 100.0%
==> Downloading https://ghcr.io/v2/homebrew/core/graphviz/blobs/sha256:29eb1e32e
######################################################################### 100.0%
==> Fetching osrf/simulation/ignition-cmake2
==> Downloading https://osrf-distributions.s3.amazonaws.com/ign-cmake/releases/i
######################################################################### 100.0%
==> Fetching pybind11
==> Downloading https://ghcr.io/v2/homebrew/core/pybind11/manifests/2.12.0
######################################################################### 100.0%
==> Downloading https://ghcr.io/v2/homebrew/core/pybind11/blobs/sha256:567ce43ee
######################################################################### 100.0%
==> Fetching python@3.11
==> Downloading https://ghcr.io/v2/homebrew/core/python/3.11/manifests/3.11.9
######################################################################### 100.0%
==> Downloading https://ghcr.io/v2/homebrew/core/python/3.11/blobs/sha256:d986c0
######################################################################### 100.0%
==> Fetching ruby
==> Downloading https://ghcr.io/v2/homebrew/core/ruby/manifests/3.3.1
Already downloaded: /Users/diysumit/Library/Caches/Homebrew/downloads/844452a66e10bb7cb2dcc8942491cf29cf39ee3e4f47e379bd22f1ff461e30e4--ruby-3.3.1.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/ruby/blobs/sha256:737689c80a9c9
Already downloaded: /Users/diysumit/Library/Caches/Homebrew/downloads/6f561c6071a423eb7984d3e11b970ae2240521b054512507929705550f39ad2a--ruby--3.3.1.arm64_sonoma.bottle.tar.gz
==> Fetching osrf/simulation/ignition-math6
==> Downloading https://osrf-distributions.s3.amazonaws.com/ign-math/releases/ig
######################################################################### 100.0%
==> Fetching osrf/simulation/ignition-common3
==> Downloading https://osrf-distributions.s3.amazonaws.com/ign-common/releases/
######################################################################### 100.0%
==> Fetching osrf/simulation/ignition-tools
==> Downloading https://osrf-distributions.s3.amazonaws.com/ign-tools/releases/i
######################################################################### 100.0%
==> Fetching osrf/simulation/ignition-msgs5
==> Downloading https://github.com/gazebosim/gz-msgs/commit/0c0926c37042ac8f5aeb
######################################################################### 100.0%
==> Downloading https://osrf-distributions.s3.amazonaws.com/ign-msgs/releases/ig
######################################################################### 100.0%
==> Fetching osrf/simulation/ignition-fuel-tools4
==> Downloading https://osrf-distributions.s3.amazonaws.com/ign-fuel-tools/relea
######################################################################### 100.0%
==> Fetching osrf/simulation/ignition-transport8
[... 161 lines cut ...]
patching file 'OgreMain/src/OgrePlatformInformation.cpp'
patching file 'OgreMain/src/OgreSIMDHelper.h'
==> cmake -DCMAKE_OSX_ARCHITECTURES='x86_64' -DOGRE_BUILD_LIBS_AS_FRAMEWORKS=OFF
==> make install
Last 15 lines from /Users/diysumit/Library/Logs/Homebrew/ogre1.9/02.make:
make[2]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreAnimationTrack.cpp.o] Error 1
make[2]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreBillboard.cpp.o] Error 1
make[2]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreAnimationState.cpp.o] Error 1
20 errors generated.
20 errors generated.
20 errors generated.
make[2]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreAtomicScalar.cpp.o] Error 1
make[2]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreAlignedAllocator.cpp.o] Error 1
20 errors generated.
make[2]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreAnimable.cpp.o] Error 1
make[2]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreAnimation.cpp.o] Error 1
20 errors generated.
make[2]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreAutoParamDataSource.cpp.o] Error 1
make[1]: *** [OgreMain/CMakeFiles/OgreMain.dir/all] Error 2
make: *** [all] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/homebrew-core):
  https://github.com/osrf/homebrew-simulation/issues

`

marvinosswald, 2024-05-05:
Verbose output of this error:
<details><summary>Verbose Output</summary>
<p>

```bash
In file included from /opt/homebrew/include/boost/system/system_error.hpp:50: error: expected function body after function declarator
:35:
In file included from /tmp/ogre1.9-20240505-71123-lksjit/sinbad-ogre-108ab0bcc696/OgreMain/include/Threading/OgreThreadHeaders.h:34:
In file included from /tmp/ogre1.9-20240505-71123-lksjit/sinbad-ogre-108ab0bcc696/OgreMain/include/Threading/OgreThreadHeadersBoost.h:48::
9In file included from inline error_category const & generic_category() noexcept BOOST_SYMBOL_VISIBLE;/opt/homebrew/include/boost/thread/recursive_mutex.hpp:/opt/homebrew/include/boost/system/detail/error_category.hpp:
:106:40: 16error: :
expected ';' at end of declaration list

                                                 ^
In file included from /opt/homebrew/include/boost/system/errc.hpp:14:
In file included from /opt/homebrew/include/boost/system/detail/error_code.hpp:15:
In file included from /opt/homebrew/include/boost/system/detail/error_condition.hpp:14In file included from /opt/homebrew/include/boost/thread/pthread/recursive_mutex.hpp:11:
In file included from /opt/homebrew/include/boost/thread/exceptions.hpp:22:
In file included from /opt/homebrew/include/boost/system/system_error.hpp/opt/homebrew/include/boost/system/detail/generic_category.hpp:
/opt/homebrew/include/boost/system/detail/generic_category.hpp::38:52: error: expected ';' at end of declaration list
    virtual bool failed( int ev ) const noexcept
                                       ^
In file included from     BOOST_SYSTEM_CONSTEXPR generic_error_category() noexcept:
                                                   ^
100:50: error: expected function body after function declarator
inline error_category const & generic_category() noexcept
                                                 ^
/tmp/ogre1.9-20240505-71123-lksjit/sinbad-ogre-108ab0bcc696/OgreMain/src/OgreAtomicScalar.cpp:/opt/homebrew/include/boost/system/detail/generic_category.hpp29::
49:In file included from 73/tmp/ogre1.9-20240505-71123-lksjit/sinbad-ogre-108ab0bcc696/OgreMain/include/OgreStableHeaders.h:: 44error: :
expected ';' at end of declaration listIn file included from
/tmp/ogre1.9-20240505-71123-lksjit/sinbad-ogre-108ab0bcc696/OgreMain/include/OgreCamera.h:35:
In file included from /tmp/ogre1.9-20240505-71123-lksjit/sinbad-ogre-108ab0bcc696/OgreMain/include/OgreMovableObject.h:37:
In file included from /tmp/ogre1.9-20240505-71123-lksjit/sinbad-ogre-108ab0bcc696/OgreMain/include/OgreShadowCaster.h:32:
In file included from     char const * message( int ev, char * buffer, std::size_t len ) const noexcept BOOST_OVERRIDE;/tmp/ogre1.9-20240505-71123-lksjit/sinbad-ogre-108ab0bcc696/OgreMain/include/OgreRenderable.h
:                                                                        ^36
:
In file included from /tmp/ogre1.9-20240505-71123-lksjit/sinbad-ogre-108ab0bcc696/OgreMain/include/OgreMaterial.h:33:
In file included from /tmp/ogre1.9-20240505-71123-lksjit/sinbad-ogre-108ab0bcc696/OgreMain/include/OgreResource.h:34:
In file included from /tmp/ogre1.9-20240505-71123-lksjit/sinbad-ogre-108ab0bcc696/OgreMain/include/OgreStringInterface.h:35:
In file included from /tmp/ogre1.9-20240505-71123-lksjit/sinbad-ogre-108ab0bcc696/OgreMain/include/Threading/OgreThreadHeaders.h:34:
In file included from /tmp/ogre1.9-20240505-71123-lksjit/sinbad-ogre-108ab0bcc696/OgreMain/include/Threading/OgreThreadHeadersBoost.h:48:
In file included from /opt/homebrew/include/boost/thread/recursive_mutex.hpp:16:
In file included from /opt/homebrew/include/boost/thread/pthread/recursive_mutex.hpp:11:
In file included from /opt/homebrew/include/boost/thread/exceptions.hpp:22:
In file included from /opt/homebrew/include/boost/system/system_error.hpp:9:
In file included from /opt/homebrew/include/boost/system/errc.hpp:14:
In file included from /opt/homebrew/include/boost/system/detail/error_code.hpp:15:
In file included from /opt/homebrew/include/boost/system/detail/error_condition.hpp:14:
In file included from /opt/homebrew/include/boost/system/detail/generic_category.hpp:14:
/opt/homebrew/include/boost/system/detail/generic_category_message.hpp:71:94: error: expected function body after function declarator
/opt/homebrew/include/boost/system/detail/generic_category.hpp:58:101: error: expected function body after function declarator
inline char const * generic_error_category_message( int ev, char * buffer, std::size_t len ) noexcept
                                                                                             ^
inline char const * generic_error_category::message( int ev, char * buffer, std::size_t len ) const noexcept
                                                                                                    ^
In file included from /tmp/ogre1.9-20240505-71123-lksjit/sinbad-ogre-108ab0bcc696/OgreMain/src/OgreAlignedAllocator.cpp:28:
In file included from /opt/homebrew/include/boost/system/detail/generic_category.hpp/tmp/ogre1.9-20240505-71123-lksjit/sinbad-ogre-108ab0bcc696/OgreMain/include/OgreStableHeaders.h::9744::
50In file included from :/tmp/ogre1.9-20240505-71123-lksjit/sinbad-ogre-108ab0bcc696/OgreMain/include/OgreCamera.h :error: 35expected function body after function declarator:

In file included from /tmp/ogre1.9-20240505-71123-lksjit/sinbad-ogre-108ab0bcc696/OgreMain/include/OgreMovableObject.h:37:
[... cut ...]
     /Users/marvinosswald/Library/Logs/Homebrew/ogre1.9/00.options.out
     /Users/marvinosswald/Library/Logs/Homebrew/ogre1.9/01.cmake
     /Users/marvinosswald/Library/Logs/Homebrew/ogre1.9/01.cmake.cc
     /Users/marvinosswald/Library/Logs/Homebrew/ogre1.9/02.make
     /Users/marvinosswald/Library/Logs/Homebrew/ogre1.9/02.make.cc
     /Users/marvinosswald/Library/Logs/Homebrew/ogre1.9/build
If reporting this issue please do so at (not Homebrew/brew or Homebrew/homebrew-core):

```
</p>
</details>

scpeters, 2024-05-10:
this may actually be caused by boost 1.85.0: https://github.com/gazebosim/gazebo-classic/issues/3385

it's not always easy to see in the logs due to all the compiler warnings

I'm working on a patch in https://github.com/gazebosim/gazebo-classic/pull/3386

scpeters, 2024-05-13:
a similar patch to https://github.com/gazebosim/gazebo-classic/pull/3386 may be needed for `ogre1.9`

scpeters, 2024-05-13:
thank you @marvinosswald; I was able to reproduce the same failure to build `ogre1.9`. I will retitle this issue to refer to `ogre1.9` and that it affects Intel CPU as well

scpeters, 2024-05-13:
I believe the c++ standard version is not being set by cmake, so it is defaulting to c++20 for the current versions of Clang / Xcode. Since `noexcept` has been removed from c++20, the ogre1.9 code is failing to compile. I think setting the cxx standard in cmake could fix this

scpeters, 2024-05-13:
I tried building with the following patch:

~~~
diff --git a/Formula/ogre1.9.rb b/Formula/ogre1.9.rb
index fabd3dc..8a1b437 100644
--- a/Formula/ogre1.9.rb
+++ b/Formula/ogre1.9.rb
@@ -97,6 +97,7 @@ class Ogre19 < Formula
 
   def install
     cmake_args = [
+      "-DCMAKE_CXX_STANDARD='11'",
       "-DCMAKE_OSX_ARCHITECTURES='x86_64'",
       "-DOGRE_BUILD_LIBS_AS_FRAMEWORKS=OFF",
       "-DOGRE_FULL_RPATH:BOOL=FALSE",
~~~

and it gets farther in the build but still fails with a linking error:

~~~
RenderSystems/GL/src/OSX/OgreOSXWindow.cpp:326:21: error: assigning to 'CGError' from incompatible type 'CGLError' (aka '_CGLError')
            cgErr = CGLSetParameter(mCGLContextObj, kCGLCPSwapInterval, &swapInterval);
                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
~~~

marvinosswald, 2024-05-14:
@scpeters thanks for looking into this 👍 

i can confirm your patch results in the same error on my mac m1

```
/tmp/ogre1.9-20240514-73090-tr2xz7/sinbad-ogre-108ab0bcc696/RenderSystems/GL/src/OSX/OgreOSXWindow.cpp:326:21: error: assigning to 'CGError' from incompatible type 'CGLError' (aka '_CGLError')
            cgErr = CGLSetParameter(mCGLContextObj, kCGLCPSwapInterval, &swapInterval);
```

marvinosswald, 2024-05-14:
So i have some progress, i've [patched `OgreOSXWindow.cpp`](https://gist.github.com/marvinosswald/0e981e918aeee4403e9a0790c6f32cdd):
```diff
diff --git a/RenderSystems/GL/src/OSX/OgreOSXWindow.cpp b/RenderSystems/GL/src/OSX/OgreOSXWindow.cpp
index 44231ab5a..9412b01d7 100644
--- a/RenderSystems/GL/src/OSX/OgreOSXWindow.cpp
+++ b/RenderSystems/GL/src/OSX/OgreOSXWindow.cpp
@@ -323,7 +323,7 @@ namespace Ogre
             cgErr = CGLSetParameter(mCGLContextObj, kCGLCPSwapInterval, &swapInterval);
 #else
             GLint swapInterval = 1;
-            cgErr = CGLSetParameter(mCGLContextObj, kCGLCPSwapInterval, &swapInterval);
+            cgErr = (CGError)CGLSetParameter(mCGLContextObj, kCGLCPSwapInterval, &swapInterval);
 #endif
             CG_CHECK_ERROR(cgErr)
```

This allowd me to actually complete the install. And run `gz sim -s`

marvinosswald, 2024-05-14:
using this [getting started guide ](https://gazebosim.org/docs/harmonic/getstarted#macos) gazebo seemed to have started:
(not entirely sure about the empty ogre window but i'll take it)
[image]

scpeters, 2024-05-14:
nice work! I think I've seen that empty Ogre window before, not sure why it's there

I think this example world may be using ogre2 instead of ogre1.9. You could try running the [dem_volcano.sdf](https://github.com/gazebosim/gz-sim/blob/6af74459f4b0f5a2a33fa112e24ac50633a90d3b/examples/worlds/dem_volcano.sdf#L24) world since it explicitly requests to use `ogre`

but I guess if `ogre1.9` is the only package blocking things, we could remove it as a dependency from gz-rendering and gz-sim. I'll try your patch first though

marvinosswald, 2024-05-15:
Good catch, indeed i've now tried the `dem_volcano.sdf` and while the server boots fine the gui doesn't.

<details><summary>Logs</summary>

```
[GUI] [Dbg] [MinimalScene.cc:749] Create scene [scene]
[GUI] [Err] [BaseCamera.hh:840] RenderTextureMetalId is not supported by current render engine
[GUI] [Dbg] [MinimalScene.cc:1052] Creating texture node render interface for Metal
Stack trace (most recent call last) in thread 6348877824:
#27   Object "QtCore", at 0x107b7dd73, in QObject::event(QEvent*) + 595
#26   Object "libMinimalScene.dylib", at 0x142734c07, in gz::gui::plugins::RenderThread::RenderNext(gz::gui::plugins::RenderSync*) + 35
#25   Object "libMinimalScene.dylib", at 0x142740d4f, in gz::gui::plugins::RenderThreadRhiMetal::RenderNext(gz::gui::plugins::RenderSync*) + 95
#24   Object "libMinimalScene.dylib", at 0x142731627, in gz::gui::plugins::GzRenderer::Render(gz::gui::plugins::RenderSync*, gz::gui::plugins::RenderThreadRhi&) + 227
#23   Object "libgz-rendering8-ogre.8.1.1.dylib", at 0x1680170e3, in gz::rendering::v8::BaseCamera<gz::rendering::v8::OgreSensor>::PreRender() + 283
#22   Object "libgz-rendering8-ogre.8.1.1.dylib", at 0x16808711b, in virtual thunk to gz::rendering::v8::BaseRenderTarget<gz::rendering::v8::OgreObject>::PreRender(std::__1::shared_ptr<gz::rendering::v8::Camera> const&) + 47
#21   Object "libgz-rendering8-ogre.8.1.1.dylib", at 0x168084e8f, in gz::rendering::v8::OgreRenderTarget::PreRender() + 59
#20   Object "libgz-rendering8-ogre.8.1.1.dylib", at 0x168086fa7, in gz::rendering::v8::BaseRenderTarget<gz::rendering::v8::OgreObject>::Rebuild() + 35
#19   Object "libgz-rendering8-ogre.8.1.1.dylib", at 0x168085173, in virtual thunk to gz::rendering::v8::OgreRenderTarget::RebuildImpl() + 39
#18   Object "libgz-rendering8-ogre.8.1.1.dylib", at 0x168085ddf, in virtual thunk to gz::rendering::v8::OgreRenderTexture::RebuildTarget() + 43
#17   Object "libgz-rendering8-ogre.8.1.1.dylib", at 0x168085ee3, in gz::rendering::v8::OgreRenderTexture::DestroyTarget() + 235
#16   Object "libOgreMain.1.9.0.dylib", at 0x1683f3e4f, in Ogre::ResourceManager::remove(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&) + 79
#15   Object "libOgreMain.1.9.0.dylib", at 0x1682f7347, in Ogre::SharedPtr<Ogre::Resource>::~SharedPtr() + 67
#14   Object "libOgreMain.1.9.0.dylib", at 0x168324bcb, in Ogre::SharedPtrInfoDelete<Ogre::Resource>::~SharedPtrInfoDelete() + 51
#13   Object "RenderSystem_GL.1.9.0.dylib", at 0x168f0ad3b, in Ogre::GLTexture::~GLTexture() + 11
#12   Object "RenderSystem_GL.1.9.0.dylib", at 0x168f0acf7, in Ogre::GLTexture::~GLTexture() + 63
#11   Object "libOgreMain.1.9.0.dylib", at 0x168477867, in Ogre::Texture::freeInternalResources() + 35
#10   Object "RenderSystem_GL.1.9.0.dylib", at 0x168f0c4c7, in Ogre::GLTexture::freeInternalResourcesImpl() + 39
#9    Object "RenderSystem_GL.1.9.0.dylib", at 0x168f0c8d7, in Ogre::SharedPtr<Ogre::HardwarePixelBuffer>::~SharedPtr() + 67
#8    Object "libOgreMain.1.9.0.dylib", at 0x1683385f3, in Ogre::SharedPtrInfoDelete<Ogre::HardwarePixelBuffer>::~SharedPtrInfoDelete() + 51
#7    Object "RenderSystem_GL.1.9.0.dylib", at 0x168efb64b, in Ogre::GLTextureBuffer::~GLTextureBuffer() + 11
#6    Object "RenderSystem_GL.1.9.0.dylib", at 0x168efb62f, in Ogre::GLTextureBuffer::~GLTextureBuffer() + 151
#5    Object "RenderSystem_GL.1.9.0.dylib", at 0x168ef643b, in Ogre::GLFBORenderTexture::~GLFBORenderTexture() + 35
#4    Object "RenderSystem_GL.1.9.0.dylib", at 0x168ef69b3, in Ogre::GLFrameBufferObject::~GLFrameBufferObject() + 75
#3    Object "libsystem_platform.dylib", at 0x18eeaf583, in _sigtramp + 55
#2    Object "libgz-tools2-backward.2.0.1.dylib", at 0x10249ae9b, in backward::SignalHandling::sig_handler(int, __siginfo*, void*) + 19
#1    Object "libgz-tools2-backward.2.0.1.dylib", at 0x10249aeff, in backward::SignalHandling::handleSignal(int, __siginfo*, void*) + 71
#0    Object "libgz-tools2-backward.2.0.1.dylib", at 0x10249b03f, in backward::StackTraceImpl<backward::system_tag::darwin_tag>::load_from(void*, unsigned long, void*, void*) + 43
[1]    24282 segmentation fault  gz sim -v 4 -g
```
</details>

skol101, 2024-06-14:
Still not working on m1 as per https://github.com/osrf/homebrew-simulation/issues/1628#issuecomment-2167803844