osrf/homebrew-simulation #2650: Installation failure: Multiple errors during make install

mondiriariana, 2024-05-12:
Attempting to install Ogre 1.9 via Homebrew, encountered numerous errors during the make install process. Here are the last 15 lines from the log:
~~~

~~~
brew install gazebo11
~~~

`make install
Last 15 lines from /Users/arianamondiri/Library/Logs/Homebrew/ogre1.9/02.make:
20 errors generated.
20 errors generated.
make[2]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreAnimation.cpp.o] Error 1
make[2]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreAxisAlignedBox.cpp.o] Error 1
make[2]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreAnimationState.cpp.o] Error 1
20 errors generated.
make[2]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreAnimationTrack.cpp.o] Error 1
20 errors generated.
20 errors generated.
make[2]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreAnimable.cpp.o] Error 1
make[2]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreArchiveManager.cpp.o] Error 1
20 errors generated.
make[2]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreAutoParamDataSource.cpp.o] Error 1
make[1]: *** [OgreMain/CMakeFiles/OgreMain.dir/all] Error 2
make: *** [all] Error 2`

Seems like the compilation process is failing consistently with various files encountering errors. This issue prevents successful installation of Ogre 1.9. Any insights or guidance on resolving these errors would be greatly appreciated.

scpeters, 2024-05-13:
this looks like #2646 to me

scpeters, 2024-05-13:
do you mind if we consolidate this discussion with that issue?

closed 2024-05-13 by scpeters