osrf/homebrew-simulation #2693: brew install  ogre1.9 fails on Apple M1 Max silicon

nihavend, 2024-06-28:
HOMEBREW_VERSION: 4.3.7-36-ga40f327
ORIGIN: https://github.com/Homebrew/brew
HEAD: a40f32776f5d4020df4d11bb6acef9e845b70fb1
Last commit: 4 hours ago
Core tap JSON: 28 Jun 12:03 UTC
Core cask tap JSON: 28 Jun 12:03 UTC
HOMEBREW_PREFIX: /opt/homebrew
HOMEBREW_CASK_OPTS: []
HOMEBREW_DISPLAY: /private/tmp/com.apple.launchd.XzhOFS1xIq/org.xquartz:0
HOMEBREW_GITHUB_API_TOKEN: set
HOMEBREW_MAKE_JOBS: 10
HOMEBREW_SORBET_RUNTIME: set
Homebrew Ruby: 3.3.3 => /opt/homebrew/Library/Homebrew/vendor/portable-ruby/3.3.3/bin/ruby
CPU: 10-core 64-bit arm_firestorm_icestorm
Clang: 15.0.0 build 1500
Git: 2.39.3 => /Applications/Xcode.app/Contents/Developer/usr/bin/git
Curl: 8.6.0 => /usr/bin/curl
macOS: 14.5-arm64
CLT: 15.3.0.0.1.1708646388
Xcode: 15.4
Rosetta 2: false
~~~

[logs](https://gist.github.com/nihavend/fe3b00afa52b8135ebf347801707f28)

scpeters, 2024-07-01:
I can't view your logs link; perhaps the permissions are set to private?

I'm guessing this may be a duplicate of https://github.com/osrf/homebrew-simulation/issues/2478

nihavend, 2024-07-02:
> I can't view your logs link; perhaps the permissions are set to private?
> 
> I'm guessing this may be a duplicate of #2478

Ooops. I fixed the link thanks for warning and I guess it is similar bug but not sure that it is duplicate. If you are than I can close it.

nihavend, 2024-07-06:
closing as this is the duplicate of #2699

closed 2024-07-06 by nihavend