osrf/homebrew-simulation #2699: brew install  ogre1.9 fails on Apple M1 Max silicon

neoskye24, 2024-07-04:
brew install gazebo11
==> Downloading https://formulae.brew.sh/api/formula.jws.json
############################################################################################################################################################################################################### 100.0%
==> Downloading https://formulae.brew.sh/api/cask.jws.json
############################################################################################################################################################################################################### 100.0%
==> Fetching dependencies for osrf/simulation/gazebo11: osrf/simulation/ogre1.9, qt@5, qwt-qt5, osrf/simulation/tinyxml1, osrf/simulation/sdformat9 and osrf/simulation/simbody
==> Fetching osrf/simulation/ogre1.9
==> Downloading https://gist.githubusercontent.com/marvinosswald/0e981e918aeee4403e9a0790c6f32cdd/raw/5d295499255867064ace8c4c41b2f31fd48d4576/gistfile1.txt
Already downloaded: /Users/akashsahoo/Library/Caches/Homebrew/downloads/812d1904a40fe1c469fe6d0846788e981df3e0b14378c7eff5c6f4d0e2a0ace1--gistfile1.txt
==> Downloading https://gist.github.com/NikolausDemmel/2b11d1b49b35cd27a102/raw/bf4a4d16020821218f73db0d56aa111ab2fde679/fix-1.9-HEAD.diff
Already downloaded: /Users/akashsahoo/Library/Caches/Homebrew/downloads/e7c203b88a808ca752ebac27f8b00b295445f8651766e688eabac24ca91f2af8--fix-1.9-HEAD.diff
==> Downloading https://gist.githubusercontent.com/NikolausDemmel/927bd7bb3f14c1788599/raw/c9a5ba88b758e80d3f46511629c4e8026b92c462/ogre1.9.patch
Already downloaded: /Users/akashsahoo/Library/Caches/Homebrew/downloads/ed6176508d9571d70a8a8b9c812c24e9fa7e56e4bfb6284e836cbafbfaa9c98c--ogre1.9.patch
==> Downloading https://gist.githubusercontent.com/iche033/68349eebfc436e484b70e6e3508ae27b/raw/d96227cd79ed1d63f051689a18c95a1c174a4efa/ogre-1.9-ois.patch
Already downloaded: /Users/akashsahoo/Library/Caches/Homebrew/downloads/4cf7ce49769de5262fd9a6a9858ff48c9104aaa3873fdd80419964a928372486--ogre-1.9-ois.patch
==> Downloading https://gist.githubusercontent.com/iche033/b73766fac9ab3d628a79b5ed986677cd/raw/878d0902704c7fb51511163052c95294361f1dbe/ogre-1.9-agl.patch
Already downloaded: /Users/akashsahoo/Library/Caches/Homebrew/downloads/decf335241bb718c38a96a2a2a70cfa096762c4beb4dc2f3e4a16f889ed36002--ogre-1.9-agl.patch
==> Downloading https://gist.githubusercontent.com/iche033/e0080a592c890cc9a4fce31f6863a5ed/raw/875ae8ad1d9f0eaa271fd44eab8e0979bac74119/ogre-1.9-cocoa_window_scale.patch
Already downloaded: /Users/akashsahoo/Library/Caches/Homebrew/downloads/c15e631644dee75637c0cc48b026154a91280d8c057a25bfb1b3eda50e7499c8--ogre-1.9-cocoa_window_scale.patch
==> Downloading https://gist.githubusercontent.com/iche033/e2b152d9df080b21f71ba3b65aa39922/raw/5397fa149c7570c17f2d78421d44b12dfa175387/ogre-1.9-cxx_flags.patch
Already downloaded: /Users/akashsahoo/Library/Caches/Homebrew/downloads/d170e4463cc8b76d13766d512b11bdab092bbf4407be160a479c718b9a0a099b--ogre-1.9-cxx_flags.patch
==> Downloading https://gist.githubusercontent.com/iche033/2b5e2ba31436881f1db29f9b60c7a5b2/raw/b6ab953ebd82127ad1177744f367a36e059312a9/ogre-1.9-findogre.patch
Already downloaded: /Users/akashsahoo/Library/Caches/Homebrew/downloads/685334483037480ddd52ca2f1dbad6eb35a573818150b82cc2de714cb99becaf--ogre-1.9-findogre.patch
==> Downloading https://github.com/OGRECave/ogre/commit/cade48b4c2215dd77ea74b7aa219a22c2a5d6654.patch?full_index=1
Already downloaded: /Users/akashsahoo/Library/Caches/Homebrew/downloads/d1d1b1546f3d7d50f8a0a37c28e7221c498c83ce5db8a239d770b7dc74bd1633--cade48b4c2215dd77ea74b7aa219a22c2a5d6654.patch
==> Downloading https://github.com/OGRECave/ogre/commit/2371c8d001725190a9cda62dc5df282cde78f951.patch?full_index=1
Already downloaded: /Users/akashsahoo/Library/Caches/Homebrew/downloads/b9b188483f25c506521c31da22540bd8c596fc9c38039cf0c96375a271f391b3--2371c8d001725190a9cda62dc5df282cde78f951.patch
==> Downloading https://github.com/scpeters/ogre/commit/8fcfe0885e6bd98d971250d4a3fb9a23e3743dfd.patch?full_index=1
Already downloaded: /Users/akashsahoo/Library/Caches/Homebrew/downloads/92546ae2ac08eb68f3076eaff910d6f37892a71d4035ead212b1f827d4d9e3a3--8fcfe0885e6bd98d971250d4a3fb9a23e3743dfd.patch
==> Downloading https://github.com/scpeters/ogre/commit/c39b6df49618773357da0b54437f351b57168d12.patch?full_index=1
Already downloaded: /Users/akashsahoo/Library/Caches/Homebrew/downloads/32a0b2d5366cc6e476bfee9f50ec7207164ead6287ce326019beaa38b875eed9--c39b6df49618773357da0b54437f351b57168d12.patch
==> Downloading https://osrf-distributions.s3.amazonaws.com/ogre/releases/sinbad-ogre-108ab0bcc696.tar.bz2
Already downloaded: /Users/akashsahoo/Library/Caches/Homebrew/downloads/af051d2d04619fdb9791203d7d4d20e63622bbd39401ccf50e3df5ff4bdebff4--sinbad-ogre-108ab0bcc696.tar.bz2
==> Fetching qt@5
==> Downloading https://ghcr.io/v2/homebrew/core/qt/5/manifests/5.15.13_1
Already downloaded: /Users/akashsahoo/Library/Caches/Homebrew/downloads/adf1b7cb33cc562d83b89f2da9bd2a5da157cf7ef6558756d9de91c05b772b20--qt@5-5.15.13_1.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/qt/5/blobs/sha256:3a497f2b6e427f72c52e52e979f655cea54e0d6eddffd1fbf3c4a25bb5eae101
Already downloaded: /Users/akashsahoo/Library/Caches/Homebrew/downloads/eaafa7396711707b7a7be02588b9387c8a5961c7ce997992061e38410bb87816--qt@5--5.15.13_1.arm64_sonoma.bottle.tar.gz
==> Fetching qwt-qt5
==> Downloading https://ghcr.io/v2/homebrew/core/qwt-qt5/manifests/6.3.0
Already downloaded: /Users/akashsahoo/Library/Caches/Homebrew/downloads/53f90f0abd89abf216c37c580889ecdaabdcb782d11ed5d9ea42ca712a01e519--qwt-qt5-6.3.0.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/qwt-qt5/blobs/sha256:894d743586ad9e9dcbbb45f9448c6f2fc945aae526f50b2d17e2162b82527402
Already downloaded: /Users/akashsahoo/Library/Caches/Homebrew/downloads/27b5ea4caf891d86a3a7a2be1fa62be3c8e97a7a887f828b61fa31629eaa52cb--qwt-qt5--6.3.0.arm64_sonoma.bottle.tar.gz
==> Fetching osrf/simulation/tinyxml1
==> Downloading https://raw.githubusercontent.com/robotology/yarp/59eedfbaa1069aa5f03a4a9980d984d59decd55c/extern/tinyxml/patches/enforce-use-stl.patch
Already downloaded: /Users/akashsahoo/Library/Caches/Homebrew/downloads/bf6a247beb0ca453bb54126e92998c3154c5ee977cbc9c73eecb617a50617796--enforce-use-stl.patch
==> Downloading https://raw.githubusercontent.com/robotology/yarp/59eedfbaa1069aa5f03a4a9980d984d59decd55c/extern/tinyxml/patches/entity-encoding.patch
Already downloaded: /Users/akashsahoo/Library/Caches/Homebrew/downloads/e08a1fd1e2a22d987472c36469b262fb99e02a8383894dd8fb17b21656afed82--entity-encoding.patch
==> Downloading https://gist.githubusercontent.com/scpeters/6325123/raw/cfb079be67997cb19a1aee60449714a1dedefed5/tinyxml_CMakeLists.patch
Already downloaded: /Users/akashsahoo/Library/Caches/Homebrew/downloads/988b8e49e31db1f8b98cf7f29553982a2af39808670e60aceca2f61b99f53f98--tinyxml_CMakeLists.patch
==> Downloading https://downloads.sourceforge.net/project/tinyxml/tinyxml/2.6.2/tinyxml_2_6_2.tar.gz
Already downloaded: /Users/akashsahoo/Library/Caches/Homebrew/downloads/6d7dbf5cbe5e453348a7155d88f10038ce9c9e0662b3603a9ab8bffd235e0166--tinyxml_2_6_2.tar.gz
==> Fetching osrf/simulation/sdformat9
==> Downloading https://osrf-distributions.s3.amazonaws.com/sdformat/releases/sdformat-9.10.1.tar.bz2
Already downloaded: /Users/akashsahoo/Library/Caches/Homebrew/downloads/0e42562c1686ae0eee3d36006c27cde577b05804274738fb7ee310f059473b36--sdformat-9.10.1.tar.bz2
==> Fetching osrf/simulation/simbody
==> Downloading https://github.com/simbody/simbody/archive/refs/tags/Simbody-3.7.tar.gz
Already downloaded: /Users/akashsahoo/Library/Caches/Homebrew/downloads/7844df9fee82cc6770cddae0d6d96361ae40d17a292c06148d0ecde2e825ac76--simbody-Simbody-3.7.tar.gz
==> Fetching osrf/simulation/gazebo11
==> Downloading https://github.com/gazebosim/gazebo-classic/commit/e4b4d0fb752c7e43e34ab97d0e01a2a3eaca1ed4.patch?full_index=1
Already downloaded: /Users/akashsahoo/Library/Caches/Homebrew/downloads/6cca052704d3760483ae0e3028a64abb77fcc5871c081876657b7da1701fcdfa--e4b4d0fb752c7e43e34ab97d0e01a2a3eaca1ed4.patch
==> Downloading https://github.com/gazebosim/gazebo-classic/commit/d04e1e0e9473e3bbbc0ae6e3f62dcad682bf9fd9.patch?full_index=1
Already downloaded: /Users/akashsahoo/Library/Caches/Homebrew/downloads/23e9502345234e8bfd5b8e282e7032c0339508339a07b1f2d5027d632ea64cdd--d04e1e0e9473e3bbbc0ae6e3f62dcad682bf9fd9.patch
[... cut ...]
Call Stack (most recent call first):
  /opt/homebrew/Cellar/cmake/3.30.0/share/cmake/Modules/CMakeTestCCompiler.cmake:26 (CMAKE_DETERMINE_COMPILER_ABI)
  CMakeLists.txt:51 (project)

-- Configuring incomplete, errors occurred!

If reporting this issue please do so at (not Homebrew/brew or Homebrew/homebrew-core):
  https://github.com/osrf/homebrew-simulation/issues

These open issues may also help:
brew install  ogre1.9 fails on Apple M1 Max silicon https://github.com/osrf/homebrew-simulation/issues/2693

eslamhabashy, 2024-07-05:
how to solve this problem

neoskye24, 2024-07-05:
I have gone through the system change and different installations. Did not work. If anyone has a solution, Highly appreciate it. Thanks

awalcz, 2024-07-07:
Remove this line 

https://github.com/osrf/homebrew-simulation/blob/364c90d6b17703d0c4438a52e4107861f8c2ba07/Formula/ogre1.9.rb#L106C2-L106C44

Install from the modified formula:

```bash
brew install -s ~/Formulas/ogre1.9.rb
```

christiancp100, 2024-07-08:
Having the same problem here!

scpeters, 2024-07-08:
> Remove this line
> 
> https://github.com/osrf/homebrew-simulation/blob/364c90d6b17703d0c4438a52e4107861f8c2ba07/Formula/ogre1.9.rb#L106C2-L106C44
> 
> Install from the modified formula:
> 
> ```shell
> brew install -s ~/Formulas/ogre1.9.rb
> ```

I've opened https://github.com/osrf/homebrew-simulation/pull/2700 to remove this line. I don't have an M1 machine to test with, but it doesn't seem to hurt anything on my intel machine

scpeters, 2024-07-08:
I just merged #2700. If it's still broken, please upload new logs and I'll take a look

christiancp100, 2024-07-09:
> I just merged #2700. If it's still broken, please upload new logs and I'll take a look

It works for me now! Thank you very much

closed 2024-07-11 by scpeters

nihavend, 2024-07-18:
this fixed my gazebo installation thanks