osrf/homebrew-simulation #2793: Homebrew-simulation - macos 15.0 installation issues

prawinsankar-ta, 2024-09-20:
Please supply your brew configuration:

~~~
$ brew config
HOMEBREW_VERSION: 4.3.23
ORIGIN: https://github.com/Homebrew/brew
HEAD: 42c11a4dedf352229271593de831161071603c66
Last commit: 3 days ago
Core tap JSON: 20 Sep 06:13 UTC
Core cask tap JSON: 20 Sep 06:13 UTC
HOMEBREW_PREFIX: /opt/homebrew
HOMEBREW_CASK_OPTS: []
HOMEBREW_DISPLAY: /private/tmp/com.apple.launchd.gOXWWMAMy8/org.xquartz:0
HOMEBREW_MAKE_JOBS: 8
Homebrew Ruby: 3.3.4 => /opt/homebrew/Library/Homebrew/vendor/portable-ruby/3.3.4_1/bin/ruby
CPU: octa-core 64-bit arm_firestorm_icestorm
Clang: 16.0.0 build 1600
Git: 2.46.1 => /opt/homebrew/bin/git
Curl: 8.7.1 => /usr/bin/curl
macOS: 15.0-arm64
CLT: 16.0.0.0.1.1724870825
Xcode: N/A
Rosetta 2: false
~~~

I'm trying to run the general command as per documentation and receiving the following error,
~~~
brew tap osrf/simulation
brew install gz-harmonic
~~~

Command line errors:
~~~
   error   104< | c#s t d ieor>r otrr i<ecds tidnicol>u dtirnige d< sitndcilou.dhi>n gb u<ts tddiidon.'ht>  fbiuntd  dliidbnc'+t+ 'fsi n<ds tldiiboc.+h+>' sh e<asdtedri.o .\h
>       | h    ^e
ader. \
      |     ^
:64:39: error: field has incomplete type 'exception_ptr'
   64 |   _LIBCPP_HIDE_FROM_ABI exception_ptr(nullptr_t) _NOEXCEPT : __ptr_() {}
      |                                       ^
/Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk/usr/include/c++/v1/__exception/exception_ptr.h:54:33: note: definition of 'std::exception_ptr' is not complete until the closing '}'
   54 | class _LIBCPP_EXPORTED_FROM_ABI exception_ptr {
      |                                 ^
/Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk/usr/include/c++/v1/__exception/exception_ptr.h:64:3: error: '__abi_tag__' attribute only applies to structs, variables, functions, and namespaces
   64 |   _LIBCPP_HIDE_FROM_ABI exception_ptr(nullptr_t) _NOEXCEPT : __ptr_() {}
      |   ^
/Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk/usr/include/c++/v1/__config:874:26: note: expanded from macro '_LIBCPP_HIDE_FROM_ABI'
  874 |           __attribute__((__abi_tag__(_LIBCPP_TOSTRING(_LIBCPP_ODR_SIGNATURE))))
      |                          ^
In file included from /tmp/ogre1.9-20240920-10745-ttkh05/sinbad-ogre-108ab0bcc696/OgreMain/src/OgreAnimation.cpp:28:
In file included from /tmp/ogre1.9-20240920-10745-ttkh05/sinbad-ogre-108ab0bcc696/OgreMain/include/OgreStableHeaders.h:41:
In file included from /tmp/ogre1.9-20240920-10745-ttkh05/sinbad-ogre-108ab0bcc696/OgreMain/include/OgreArchive.h:31:
In file included from /tmp/ogre1.9-20240920-10745-ttkh05/sinbad-ogre-108ab0bcc696/OgreMain/include/OgrePrerequisites.h:32:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk/usr/include/c++/v1/string:594:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk/usr/include/c++/v1/__memory_resource/polymorphic_allocator.h:20:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk/usr/include/c++/v1/tuple:1455:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk/usr/include/c++/v1/exception:82:
/Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk/usr/include/c++/v1/__exception/exception_ptr.h:64:49: error: expected ';' at end of declaration list
   64 |   _LIBCPP_HIDE_FROM_ABI exception_ptr(nullptr_t) _NOEXCEPT : __ptr_() {}
      |                                                 ^
In file included from /tmp/ogre1.9-20240920-10745-ttkh05/sinbad-ogre-108ab0bcc696/OgreMain/src/OgreAnimable.cpp:28:
In file included from /tmp/ogre1.9-20240920-10745-ttkh05/sinbad-ogre-108ab0bcc696/OgreMain/include/OgreStableHeaders.h:41:
In file included from /tmp/ogre1.9-20240920-10745-ttkh05/sinbad-ogre-108ab0bcc696/OgreMain/include/OgreArchive.h:31:
In file included from /tmp/ogre1.9-20240920-10745-ttkh05/sinbad-ogre-108ab0bcc696/OgreMain/include/OgrePrerequisites.h:32:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk/usr/include/c++/v1/string:600:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk/usr/include/c++/v1/__string/char_traits.h:26:
/Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk/usr/include/c++/v1/cstdio:104:5: error: <cstdio> tried including <stdio.h> but didn't find libc++'s <stdio.h> header.           This usually means that your header search paths are not configured properly.           The header search paths should contain the C++ Standard Library headers before           any C Standard Library, and you are probably using compiler flags that make that           not be the case.
  104 | #   error <cstdio> tried including <stdio.h> but didn't find libc++'s <stdio.h> header. \
      |     ^
In file included from /tmp/ogre1.9-20240920-10745-ttkh05/sinbad-ogre-108ab0bcc696/OgreMain/src/OgreAnimation.cpp:28:
[... 60 lines cut ...]
HOMEBREW_CC: clang
HOMEBREW_CXX: clang++
MAKEFLAGS: -j8
CMAKE_PREFIX_PATH: /opt/homebrew/opt/icu4c:/opt/homebrew
CMAKE_INCLUDE_PATH: /Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk/System/Library/Frameworks/OpenGL.framework/Versions/Current/Headers
CMAKE_LIBRARY_PATH: /Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk/System/Library/Frameworks/OpenGL.framework/Versions/Current/Libraries
PKG_CONFIG_PATH: /opt/homebrew/opt/icu4c/lib/pkgconfig:/opt/homebrew/opt/xz/lib/pkgconfig:/opt/homebrew/opt/lz4/lib/pkgconfig:/opt/homebrew/opt/zstd/lib/pkgconfig:/opt/homebrew/opt/libpng/lib/pkgconfig:/opt/homebrew/opt/freetype/lib/pkgconfig:/opt/homebrew/opt/libxau/lib/pkgconfig:/opt/homebrew/opt/libxdmcp/lib/pkgconfig:/opt/homebrew/opt/libxcb/lib/pkgconfig:/opt/homebrew/opt/libx11/lib/pkgconfig:/opt/homebrew/opt/libzzip/lib/pkgconfig:/opt/homebrew/opt/hwloc/lib/pkgconfig:/opt/homebrew/opt/tbb/lib/pkgconfig:/opt/homebrew/opt/xorgproto/share/pkgconfig
PKG_CONFIG_LIBDIR: /usr/lib/pkgconfig:/opt/homebrew/Library/Homebrew/os/mac/pkgconfig/15
HOMEBREW_MAKE_JOBS: 8
HOMEBREW_GIT: git
HOMEBREW_SDKROOT: /Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk
ACLOCAL_PATH: /opt/homebrew/share/aclocal
PATH: /opt/homebrew/Library/Homebrew/shims/mac/super:/opt/homebrew/opt/cmake/bin:/opt/homebrew/opt/icu4c/bin:/opt/homebrew/opt/xz/bin:/opt/homebrew/opt/lz4/bin:/opt/homebrew/opt/zstd/bin:/opt/homebrew/opt/doxygen/bin:/opt/homebrew/opt/libpng/bin:/opt/homebrew/opt/freetype/bin:/opt/homebrew/opt/libzzip/bin:/opt/homebrew/opt/hwloc/bin:/usr/bin:/bin:/usr/sbin:/sbin

Error: osrf/simulation/ogre1.9 1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9 did not build
~~~

I have tried,

- unlinking Ogre 1.9 module which is present and tried installing again.
- uninstalling Ogre 1.9 and installing from source again

but both the above approaches hasn't helped much. Any suggestions or help would be appreciated.

Thanks in advance!

scpeters, 2024-09-22:
issues with ogre1.9 have also been reported in https://github.com/osrf/homebrew-simulation/issues/2478. Can you try installing `ogre2.3` to see if it has issues?

prawinsankar-ta, 2024-09-23:
Hey @scpeters, thanks for reverting. Should I try installing only Ogre2.3 and building homebrew-simulation?

prawinsankar-ta, 2024-09-23:
I tried removing and installing ogre2.3 on terminal in standalone. It got installed. 

~~~
brew install ogre2.3
~~~

scpeters, 2024-09-23:
you can try removing the `depends_on "ogre1.9"` line from [gz-rendering8](https://github.com/osrf/homebrew-simulation/blob/master/Formula/gz-rendering8.rb#L27), so see if that helps with building gz-harmonic

ta-jetson, 2024-11-10:
I tried rebuilding as you had pointed out, but still running into errors. Is it needed to comment `depends_on "ogre1.9` else apart from [gz-rendering8](https://github.com/osrf/homebrew-simulation/blob/master/Formula/gz-rendering8.rb#L27)?

Attached below is the screenshot for your ref.

[image]

prawinsankar-ta, 2024-11-10:
I tried rebuilding as you had pointed out, but still running into errors. Is it needed to comment depends_on "ogre1.9 else apart from [gz-rendering8](https://github.com/osrf/homebrew-simulation/blob/master/Formula/gz-rendering8.rb#L27) ?

Attached below is the screenshot for your ref.

[image]

scpeters, 2024-11-11:
Please look at and considering the entire contents of the 02.make file,which only has the last 15 lines shown in that screenshot. The relevant error is earlier in that file

prawinsankar-ta, 2024-11-11:
I reviewed the 02.make file. It seems there are various build flags raised during the installation. Not sure if it's a ogre dependency issue as well.

[02_make.txt](https://github.com/user-attachments/files/17707033/02_make.txt)

scpeters, 2024-11-11:
the log file is a bit garbled due to parallel build errors writing to the log file at the same time, but I was able to see the following:

~~~
           <cstddef> tried including <stddef.h> but didn't find libc++'s <stddef.h> header.
           This usually means that your header search paths are not configured properly.
           The header search paths should contain the C++ Standard Library headers before
           any C Standard Library, and you are probably using compiler flags that make that
           not be the case.
~~~

Perhaps try uninstalling the Xcode command-line tools and reinstalling?

prawinsankar-ta, 2024-11-28:
[image]

@scpeters I tried reinstalling `xcode-command-line-tools`, but I'm still getting the same error while building hombrew. 

Even I tried installing installing homebrew-simulation using,

`brew install gz-harmonic --verbose`

but still having same error.

DavidKalajdzic, 2025-04-18:
Same issue

scpeters, 2025-05-09:
We just updated the ogre1.9 formula in #3029, can you try again?

scpeters, 2025-05-12:
> We just updated the ogre1.9 formula in [#3029](https://github.com/osrf/homebrew-simulation/pull/3029), can you try again?

I'm going to close this since I haven't heard any updates. Please comment if the issue persists and we can reopen.