osrf/homebrew-simulation #2837: Rebuild bottles broken by protobuf 28.3

scpeters, 2024-10-26:
Please supply your brew configuration:

### Brew configuration

~~~
$ brew config
HOMEBREW_VERSION: 4.4.2
ORIGIN: https://github.com/Homebrew/brew
HEAD: 35ebf4a047b00b2b3de33ba0081d71293cdadede
Last commit: 4 days ago
Core tap HEAD: 1aec28a1865044928ce004585565cca06b38d8f0
Core tap last commit: 5 minutes ago
Core tap JSON: 18 Sep 11:25 UTC
Core cask tap HEAD: 4c94e5822bb023992785d6b106961dd1ce78c4ab
Core cask tap last commit: 5 minutes ago
Core cask tap JSON: 18 Sep 11:18 UTC
HOMEBREW_PREFIX: /usr/local
HOMEBREW_CASK_OPTS: []
HOMEBREW_GIT_EMAIL: osrfbuild@osrfoundation.org
HOMEBREW_GIT_NAME: OSRF Build Bot
HOMEBREW_MAKE_JOBS: 12
HOMEBREW_NO_AUTO_UPDATE: set
HOMEBREW_NO_INSTALL_FROM_API: set
HOMEBREW_SORBET_RUNTIME: set
HOMEBREW_UPDATE_TO_TAG: set
Homebrew Ruby: 3.3.5 => /usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/3.3.5/bin/ruby
CPU: dodeca-core 64-bit kabylake
Clang: 15.0.0 build 1500
Git: 2.47.0 => /usr/local/bin/git
Curl: 8.1.2 => /usr/bin/curl
macOS: 13.6.3-x86_64
CLT: 15.1.0.0.1.1700200546
Xcode: N/A
~~~

### Brew linkage test result

From the following jenkins build:

* [image] https://build.osrfoundation.org/view/gz-ionic/job/gz_msgs11-install_bottle-homebrew-amd64/315/

~~~
$ brew linkage --test gz-msgs11
Broken dependencies:
  /usr/local/opt/protobuf/lib/libprotobuf.28.2.0.dylib (protobuf)
  /usr/local/opt/protobuf/lib/libprotoc.28.2.0.dylib (protobuf)
~~~

### Checklist

- [X] Identify change in homebrew-core that broke the bottle
    - https://github.com/Homebrew/homebrew-core/pull/195295
- [X] Identify list of affected formulae
    - [X] gz-msgs* and dependents
- [ ] Remove broken bottles: for each affected formula, run `brew bump-revision --remove-bottle-block`
- [ ] Rebuild broken bottles
    - [ ] Ionic
    - [ ] Harmonic
    - [ ] Garden
    - [ ] Fortress
    - [ ] Citadel
    - [ ] gazebo11

Crola1702, 2024-11-04:
Hey @scpeters I see [gz-transport13](https://build.osrfoundation.org/job/gz_transport-ci-gz-transport13-homebrew-amd64/94/#showFailuresLink) and [gz-msgs10](https://build.osrfoundation.org/job/gz_msgs-ci-gz-msgs10-homebrew-amd64/98/) are failing.

The log output says something about protobuf version mismatch. Is this related to these broken bottles?

<details>
<summary>
Log output
</summary>

```
15: ==================================== ERRORS ====================================
15: ______________ ERROR collecting gz-msgs/python/test/basic_TEST.py ______________
15: ../../gz-msgs/python/test/basic_TEST.py:15: in <module>
15:     from gz.msgs10.vector3d_pb2 import Vector3d
15: /usr/local/Cellar/gz-msgs10/HEAD/lib/python/gz/msgs10/vector3d_pb2.py:12: in <module>
15:     _runtime_version.ValidateProtobufRuntimeVersion(
15: /Users/jenkins/Library/Python/3.13/lib/python/site-packages/google/protobuf/runtime_version.py:106: in ValidateProtobufRuntimeVersion
15:     _ReportVersionError(
15: /Users/jenkins/Library/Python/3.13/lib/python/site-packages/google/protobuf/runtime_version.py:47: in _ReportVersionError
15:     raise VersionError(msg)
15: E   google.protobuf.runtime_version.VersionError: Detected incompatible Protobuf Gencode/Runtime versions when loading gz/msgs/vector3d.proto: gencode 5.28.3 runtime 5.28.2. Runtime version cannot be older than the linked gencode version. See Protobuf version guarantees at https://protobuf.dev/support/cross-version-runtime-guarantee.
15: - generated xml file: /Users/jenkins/jenkins-agent/workspace/gz_msgs-ci-gz-msgs10-homebrew-amd64/build/test_results/basic_TEST.xml -
15: =========================== short test summary info ============================
15: ERROR ../../gz-msgs/python/test/basic_TEST.py - google.protobuf.runtime_version.VersionError: Detected incompatible Protobuf Gencode/Runtime versions when loading gz/msgs/vector3d.proto: gencode 5.28.3 runtime 5.28.2. Runtime version cannot be older than the linked gencode version. See Protobuf version guarantees at https://protobuf.dev/support/cross-version-runtime-guarantee.
15: !!!!!!!!!!!!!!!!!!!! Interrupted: 1 error during collection !!!!!!!!!!!!!!!!!!!!
15: =============================== 1 error in 0.14s ===============================
15/15 Test #15: basic_TEST.py ..................................***Failed    0.34 sec
```

</details>

scpeters, 2024-11-04:
> Hey @scpeters I see [gz-transport13](https://build.osrfoundation.org/job/gz_transport-ci-gz-transport13-homebrew-amd64/94/#showFailuresLink) and [gz-msgs10](https://build.osrfoundation.org/job/gz_msgs-ci-gz-msgs10-homebrew-amd64/98/) are failing.
> 
> The log output says something about protobuf version mismatch. Is this related to these broken bottles?
> 
> Log output

it's because we install protobuf python support via `pip` and that version is persisting across builds and getting out of date. See https://github.com/gazebo-tooling/release-tools/pull/1193 for a discussion about how to fix this