osrf/homebrew-simulation #2889: `make[2]: *** [lib/macosx/libOgreMain.1.9.0.dylib] Error 1` when running `brew install gz-harmonic` on M1 Mac

r-bt, 2024-11-19:
I'm trying to install gz-harmonic on an M1 Macbook Pro

~~~
$ brew config
HOMEBREW_VERSION: 4.4.6
ORIGIN: https://github.com/Homebrew/brew
HEAD: cc0b5d6d05d1f3af5b2f53e9fbb1e7bfa0c332ab
Last commit: 2 days ago
Core tap JSON: 19 Nov 03:04 UTC
Core cask tap JSON: 19 Nov 02:55 UTC
HOMEBREW_PREFIX: /opt/homebrew
HOMEBREW_CASK_OPTS: []
HOMEBREW_DISPLAY: /private/tmp/com.apple.launchd.Z5Qm3gsh6r/org.xquartz:0
HOMEBREW_MAKE_JOBS: 8
Homebrew Ruby: 3.3.6 => /opt/homebrew/Library/Homebrew/vendor/portable-ruby/3.3.6/bin/ruby
CPU: octa-core 64-bit arm_firestorm_icestorm
Clang: 15.0.0 build 1500
Git: 2.39.3 => /Library/Developer/CommandLineTools/usr/bin/git
Curl: 8.6.0 => /usr/bin/curl
macOS: 14.5-arm64
CLT: 15.3.0.0.1.1708646388
Xcode: 16.1
Rosetta 2: false
~~~

However, it is failing when trying to install `ogre` I believe, with the following error:

```
Last 15 lines from /Users/richard/Library/Logs/Homebrew/ogre1.9/02.make:
  "boost::thread::start_thread_noexcept()", referenced from:
      boost::thread::start_thread() in OgreDefaultWorkQueueStandard.cpp.o
  "boost::thread::detach()", referenced from:
      boost::thread::~thread() in OgreDefaultWorkQueueStandard.cpp.o
  "typeinfo for boost::detail::thread_data_base", referenced from:
      typeinfo for boost::detail::thread_data<Ogre::DefaultWorkQueueBase::WorkerFunc> in OgreDefaultWorkQueueStandard.cpp.o
  "vtable for boost::detail::thread_data_base", referenced from:
      boost::detail::thread_data_base::thread_data_base() in OgreDefaultWorkQueueStandard.cpp.o
   NOTE: a missing vtable usually means the first non-inline virtual member function has no definition.
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
r/src/OgreSerializer.cpp.o CMakeFiles/OgreMain.dir/src/OgreShadowCameraSetup.cpp.o CMakeFiles/OgreMain.dir/src/OgreShadowCameraSetupFocused.cpp.o CMakeFiles/OgreMain.dir/src/OgreShadowCameraSetupLiSPSM.cpp.o CMakeFiles/OgreMain.dir/src/OgreShadowCameraSetupPSSM.cpp.o CMakeFiles/OgreMain.dir/src/OgreShadowCameraSetupPlaneOptimal.cpp.o CMakeFiles/OgreMain.dir/src/OgreShadowCaster.cpp.o CMakeFiles/OgreMain.dir/src/OgreShadowTextureManager.cpp.o CMakeFiles/OgreMain.dir/src/OgreShadowVolumeExtrudeProgram.cpp.o CMakeFiles/OgreMain.dir/src/OgreSimpleRenderable.cpp.o CMakeFiles/OgreMain.dir/src/OgreSimpleSpline.cpp.o CMakeFiles/OgreMain.dir/src/OgreSkeleton.cpp.o CMakeFiles/OgreMain.dir/src/OgreSkeletonInstance.cpp.o CMakeFiles/OgreMain.dir/src/OgreSkeletonManager.cpp.o CMakeFiles/OgreMain.dir/src/OgreSkeletonSerializer.cpp.o CMakeFiles/OgreMain.dir/src/OgreSmallVector.cpp.o CMakeFiles/OgreMain.dir/src/OgreStaticGeometry.cpp.o CMakeFiles/OgreMain.dir/src/OgreStreamSerialiser.cpp.o CMakeFiles/OgreMain.dir/src/OgreString.cpp.o CMakeFiles/OgreMain.dir/src/OgreStringConverter.cpp.o CMakeFiles/OgreMain.dir/src/OgreStringInterface.cpp.o CMakeFiles/OgreMain.dir/src/OgreSubEntity.cpp.o CMakeFiles/OgreMain.dir/src/OgreSubMesh.cpp.o CMakeFiles/OgreMain.dir/src/OgreTagPoint.cpp.o CMakeFiles/OgreMain.dir/src/OgreTangentSpaceCalc.cpp.o CMakeFiles/OgreMain.dir/src/OgreTechnique.cpp.o CMakeFiles/OgreMain.dir/src/OgreTexture.cpp.o CMakeFiles/OgreMain.dir/src/OgreTextureManager.cpp.o CMakeFiles/OgreMain.dir/src/OgreTextureUnitState.cpp.o CMakeFiles/OgreMain.dir/src/OgreUTFString.cpp.o CMakeFiles/OgreMain.dir/src/OgreUnifiedHighLevelGpuProgram.cpp.o CMakeFiles/OgreMain.dir/src/OgreUserObjectBindings.cpp.o CMakeFiles/OgreMain.dir/src/OgreVector2.cpp.o CMakeFiles/OgreMain.dir/src/OgreVector3.cpp.o CMakeFiles/OgreMain.dir/src/OgreVector4.cpp.o CMakeFiles/OgreMain.dir/src/OgreVertexIndexData.cpp.o CMakeFiles/OgreMain.dir/src/OgreViewport.cpp.o CMakeFiles/OgreMain.dir/src/OgreWindowEventUtilities.cpp.o CMakeFiles/OgreMain.dir/src/OgreWireBoundingBox.cpp.o CMakeFiles/OgreMain.dir/src/OgreWorkQueue.cpp.o CMakeFiles/OgreMain.dir/src/OgreFreeImageCodec.cpp.o CMakeFiles/OgreMain.dir/src/OgreDDSCodec.cpp.o CMakeFiles/OgreMain.dir/src/OgreZip.cpp.o CMakeFiles/OgreMain.dir/src/OSX/OgreConfigDialog.mm.o CMakeFiles/OgreMain.dir/src/OSX/OgreErrorDialog.mm.o CMakeFiles/OgreMain.dir/src/OSX/OgreFileSystemLayer.cpp.o CMakeFiles/OgreMain.dir/src/OSX/OgreTimer.cpp.o CMakeFiles/OgreMain.dir/src/OSX/macUtils.mm.o CMakeFiles/OgreMain.dir/src/OgreSearchOps.cpp.o CMakeFiles/OgreMain.dir/src/Threading/OgreDefaultWorkQueueStandard.cpp.o   -L/usr/local/lib  -Wl,-rpath,/usr/local/lib /usr/local/lib/libboost_thread-mt.a /usr/local/lib/libboost_date_time-mt.a /opt/homebrew/lib/libfreeimage.dylib /opt/homebrew/lib/libzzip.dylib /Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk/usr/lib/libz.tbd
make[2]: *** [lib/macosx/libOgreMain.1.9.0.dylib] Error 1
make[1]: *** [OgreMain/CMakeFiles/OgreMain.dir/all] Error 2
make: *** [all] Error 2
```

I've seen other issues where gazebo can't be installed due to `ogre issues`:
- https://github.com/osrf/homebrew-simulation/issues/2793
- https://github.com/osrf/homebrew-simulation/issues/2478

However those errors don't match mine exactly, so I wanted to open a new issue. 

I tried install `ogre2.3` with `brew install ogre2.3` however that also gave me errors

I've put the logs from `brew install gz-harmonic --verbose` here: https://gist.github.com/r-bt/114dfc0952a043f95f406b72b734a2a9

scpeters, 2024-11-19:
can you confirm the boost version that you have installed? I'm guessing 1.86 since that's the latest version:

* https://github.com/Homebrew/homebrew-core/blob/master/Formula/b/boost.rb#L11

Are there any `brew doctor` complaints?

Also, I see that you have `freetype` installed in `/usr/local`. Is that build from source? I would have expected that to be found in `/opt/homebrew`; that's a little suspicious

r-bt, 2024-11-19:
@scpeters thanks for the quick reply

I'm using boost 1.86

```
$ brew info boost
==> boost: stable 1.86.0 (bottled), HEAD
```

I've attached my brew doctor log here: https://gist.github.com/r-bt/f86838aa2ece9d6dabe66d3d9180b441

I used to have brew running under Rosetta where I believe it installs to `/usr/local` I reinstalled `freetype` and now it's under `/opt/homebrew`

```
$ brew info freetype
==> freetype: stable 2.13.3 (bottled)
Software library to render fonts
https://www.freetype.org/
Installed
/opt/homebrew/Cellar/freetype/2.13.3 (68 files, 2.5MB) *
  Poured from bottle using the formulae.brew.sh API on 2024-11-19 at 13:34:03
From: https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/f/freetype.rb
License: FTL
==> Dependencies
Build: pkg-config ✔
Required: libpng ✔
==> Analytics
install: 105,723 (30 days), 401,870 (90 days), 1,073,654 (365 days)
install-on-request: 6,067 (30 days), 23,949 (90 days), 55,073 (365 days)
build-error: 9 (30 days)
```

However I get the same error when trying to install `gz-harmonic`

scpeters, 2024-11-19:
at the very end of [this line](https://gist.github.com/r-bt/114dfc0952a043f95f406b72b734a2a9#file-log-txt-L863), I see references to boost libraries in `/usr/local`

~~~
/usr/local/lib/libboost_thread-mt.a
/usr/local/lib/libboost_date_time-mt.a
~~~

try removing those as well

r-bt, 2024-11-19:
Brilliant! I had `boost` installed under Rosetta brew, so I uninstalled that and it worked

scpeters, 2024-11-19:
great; glad it's working now

closed 2024-11-19 by scpeters