osrf/homebrew-simulation #2916: Rebuild bottles broken by boost 1.87.0

scpeters, 2024-12-17:
Please supply your brew configuration:

### Brew configuration

~~~
$ brew config

~~~

### Brew linkage test result

~~~
$ brew linkage --test gz-common5

~~~

### Checklist

- [ ] Identify change in homebrew-core that broke the bottle https://github.com/Homebrew/homebrew-core/pull/198115
- [ ] Identify list of affected formulae
    - [ ] dartsim@6.10.0
    - [ ] gazebo11
- [ ] Remove broken bottles: for each affected formula, run `brew bump-revision --remove-bottle-block`
- [ ] Rebuild broken bottles

scpeters, 2024-12-17:
gazebo11 is failing to build with the new boost:

* [image] https://build.osrfoundation.org/job/gazebo-install-one_liner-homebrew-amd64/3371/

~~~
Last 15 lines from /Users/jenkins/Library/Logs/Homebrew/gazebo11/02.make:
In file included from /tmp/gazebo11-20241217-86675-upeq8d/gazebo-11.15.0/plugins/rest_web/RestUiWidget.hh:27:
In file included from /tmp/gazebo11-20241217-86675-upeq8d/gazebo-11.15.0/gazebo/gazebo.hh:20:
In file included from /tmp/gazebo11-20241217-86675-upeq8d/gazebo-11.15.0/gazebo/gazebo_core.hh:21:
In file included from /tmp/gazebo11-20241217-86675-upeq8d/gazebo-11.15.0/build/gazebo/transport/transport.hh:5:
/tmp/gazebo11-20241217-86675-upeq8d/gazebo-11.15.0/gazebo/transport/IOManager.hh:45:28: error: no type named 'io_service' in namespace 'boost::asio'
      public: boost::asio::io_service &GetIO();
              ~~~~~~~~~~~~~^
2 errors generated.
make[2]: *** [plugins/rest_web/CMakeFiles/RestUiPlugin.dir/RestUiLoginDialog.cc.o] Error 1
2 errors generated.
make[2]: *** [plugins/rest_web/CMakeFiles/RestUiPlugin.dir/RestUiPlugin.cc.o] Error 1
2 errors generated.
make[2]: *** [plugins/rest_web/CMakeFiles/RestUiPlugin.dir/RestUiWidget.cc.o] Error 1
make[1]: *** [plugins/rest_web/CMakeFiles/RestUiPlugin.dir/all] Error 2
make: *** [all] Error 2
~~~

scpeters, 2024-12-17:
> gazebo11 is failing to build with the new boost:
> 
> 
> ```
> /tmp/gazebo11-20241217-86675-upeq8d/gazebo-11.15.0/gazebo/transport/IOManager.hh:45:28: error: no type named 'io_service' in namespace 'boost::asio'
>       public: boost::asio::io_service &GetIO();
>               ~~~~~~~~~~~~~^
> ```

apparently `io_service` has been deprecated in favor of `io_context` and was just recently removed: https://github.com/boostorg/asio/issues/110

scpeters, 2025-01-06:
dartsim@6.10.0 is only used by gz-physics2, which is now EOL, so I think we won't try to fix that

scpeters, 2025-01-06:
I experimented with added a keg-only versioned formula for `boost@1.85.0` for gazebo11 to use, but this would require creating a second version of `ogre1.9` that uses this old version of boost.

scpeters, 2025-01-10:
from https://github.com/osrf/homebrew-simulation/issues/2927#issuecomment-2584494030 by @xdire:

> did you guys found any temporary solution to the problem? Trying to install gazebo11 on m3 and this error basically blocks me out.

scpeters, 2025-01-10:
as I noted in https://github.com/osrf/homebrew-simulation/issues/2916#issuecomment-2574017026, I experimented with a keg-only versioned formula for `boost@1.85.0` for gazebo11, but this would also require a second keg-only version of `ogre1.9` that uses that version of boost

I'll make a branch with these changes for you to test

xdire, 2025-01-10:
@scpeters, thank you. And if possible, some instructions for the branch, as I didn't get that many HB problems prior to this one, and seems no manual linking can solve those problems :) I appreciate the support for this case.

scpeters, 2025-01-10:
the more proper solution is to modify gazebo to support boost 1.87 (see https://github.com/gazebosim/gazebo-classic/issues/3414), but that is more complicated

scpeters, 2025-01-10:
> I'll make a branch with these changes for you to test

try the [scpeters/fork_boost branch](https://github.com/osrf/homebrew-simulation/tree/scpeters/fork_boost)

scpeters, 2025-01-10:
> > I'll make a branch with these changes for you to test
> 
> try the [scpeters/fork_boost branch](https://github.com/osrf/homebrew-simulation/tree/scpeters/fork_boost)

instructions:

~~~
brew remove gazebo11
cd `brew --repo osrf/simulation`
git fetch origin
git checkout scpeters/fork_boost
brew install gazebo11
~~~

I'm currently testing this now; it will take a while to compile boost, ogre, and gazebo

xdire, 2025-01-11:
@scpeters, thank you for the thorough instructions; I tried your branch, but I got the same error.

scpeters, 2025-01-13:
I just rebuilt the gazebo11 bottles to use `boost@1.85.0` in https://github.com/osrf/homebrew-simulation/pull/2935

closed 2025-01-13 by scpeters