osrf/homebrew-simulation #2926: gz-utils2 fails to install on MacOS M1

mbampi, 2024-12-28:
~~~
$ brew config

HOMEBREW_VERSION: 4.4.13
ORIGIN: https://github.com/Homebrew/brew
HEAD: 6ea9df68df4868ce12ea7c51c85b57dce514f2e8
Last commit: 5 days ago
Branch: stable
Core tap JSON: 28 Dec 02:38 UTC
Core cask tap JSON: 28 Dec 02:38 UTC
HOMEBREW_PREFIX: /opt/homebrew
HOMEBREW_CASK_OPTS: []
HOMEBREW_DISPLAY: /private/tmp/com.apple.launchd.mh3N4SK7Iq/org.xquartz:0
HOMEBREW_MAKE_JOBS: 8
Homebrew Ruby: 3.3.6 => /opt/homebrew/Library/Homebrew/vendor/portable-ruby/3.3.6/bin/ruby
CPU: octa-core 64-bit arm_firestorm_icestorm
Clang: 16.0.0 build 1600
Git: 2.39.5 => /Library/Developer/CommandLineTools/usr/bin/git
Curl: 8.7.1 => /usr/bin/curl
macOS: 15.1.1-arm64
CLT: 16.1.0.0.1.1729049160
Xcode: 16.2
Rosetta 2: false
~~~

When trying to run `brew install gz-harmonic`, it fails due to gz-utils2 failing to be installed.

Trying to run it by itself it gives (as expected) the same error. Here is the command with the debug flag.
~~~
$ brew install --debug gz-utils2

/opt/homebrew/Library/Homebrew/brew.rb (Formulary::FromNameLoader): loading gz-utils2
/opt/homebrew/Library/Homebrew/brew.rb (Formulary::FromNameLoader): loading gz-utils2
/opt/homebrew/Library/Homebrew/brew.rb (Cask::CaskLoader::NullLoader): loading gz-utils2
/opt/homebrew/Library/Homebrew/brew.rb (Formulary::FromAPILoader): loading cmake
/opt/homebrew/Library/Homebrew/brew.rb (Formulary::FromAPILoader): loading pkgconf
/opt/homebrew/Library/Homebrew/brew.rb (Formulary::FromNameLoader): loading gz-cmake3
/opt/homebrew/Library/Homebrew/brew.rb (Formulary::FromAPILoader): loading cmake
/opt/homebrew/Library/Homebrew/brew.rb (Formulary::FromAPILoader): loading pkgconf
/opt/homebrew/Library/Homebrew/brew.rb (Formulary::FromNameLoader): loading gz-cmake3
/opt/homebrew/Library/Homebrew/brew.rb (Formulary::FromAPILoader): loading cmake
/opt/homebrew/Library/Homebrew/brew.rb (Formulary::FromAPILoader): loading pkgconf
/opt/homebrew/Library/Homebrew/brew.rb (Formulary::FromAPILoader): loading cmake
/opt/homebrew/Library/Homebrew/brew.rb (Formulary::FromAPILoader): loading pkgconf
/opt/homebrew/Library/Homebrew/brew.rb (Formulary::FromTapLoader): loading osrf/simulation/gz-cmake3
/opt/homebrew/Library/Homebrew/brew.rb (Formulary::FromAPILoader): loading cmake
/opt/homebrew/Library/Homebrew/brew.rb (Formulary::FromAPILoader): loading pkgconf
/opt/homebrew/Library/Homebrew/brew.rb (Formulary::FromTapLoader): loading osrf/simulation/gz-cmake3
/opt/homebrew/Library/Homebrew/brew.rb (Formulary::FromAPILoader): loading cmake
/opt/homebrew/Library/Homebrew/brew.rb (Formulary::FromAPILoader): loading pkgconf
/opt/homebrew/Library/Homebrew/brew.rb (Formulary::FromTapLoader): loading osrf/simulation/gz-cmake3
/usr/bin/env /opt/homebrew/Library/Homebrew/shims/shared/git --version
/opt/homebrew/Library/Homebrew/brew.rb (Formulary::FromAPILoader): loading cmake
/opt/homebrew/Library/Homebrew/brew.rb (Formulary::FromAPILoader): loading cmake
/opt/homebrew/Library/Homebrew/brew.rb (Formulary::FromAPILoader): loading cmake
/opt/homebrew/Library/Homebrew/brew.rb (Formulary::FromAPILoader): loading pkgconf
/opt/homebrew/Library/Homebrew/brew.rb (Formulary::FromAPILoader): loading pkgconf
/opt/homebrew/Library/Homebrew/brew.rb (Formulary::FromAPILoader): loading pkgconf
/opt/homebrew/Library/Homebrew/brew.rb (Formulary::FromNameLoader): loading gz-cmake3
/opt/homebrew/Library/Homebrew/brew.rb (Formulary::FromNameLoader): loading gz-cmake3
/opt/homebrew/Library/Homebrew/brew.rb (Formulary::FromNameLoader): loading gz-cmake3
/opt/homebrew/Library/Homebrew/brew.rb (Formulary::FromAPILoader): loading cmake
/opt/homebrew/Library/Homebrew/brew.rb (Formulary::FromAPILoader): loading cmake
/opt/homebrew/Library/Homebrew/brew.rb (Formulary::FromAPILoader): loading cmake
/opt/homebrew/Library/Homebrew/brew.rb (Formulary::FromAPILoader): loading pkgconf
/opt/homebrew/Library/Homebrew/brew.rb (Formulary::FromAPILoader): loading pkgconf
/opt/homebrew/Library/Homebrew/brew.rb (Formulary::FromAPILoader): loading pkgconf
/opt/homebrew/Library/Homebrew/brew.rb (Formulary::FromAPILoader): loading cmake
/opt/homebrew/Library/Homebrew/brew.rb (Formulary::FromAPILoader): loading cmake
/opt/homebrew/Library/Homebrew/brew.rb (Formulary::FromAPILoader): loading pkgconf
[... 100 lines cut ...]
/opt/homebrew/Library/Homebrew/formula.rb:3347:in `block (2 levels) in stage'
/opt/homebrew/Library/Homebrew/extend/kernel.rb:539:in `with_env'
/opt/homebrew/Library/Homebrew/formula.rb:3346:in `block in stage'
/opt/homebrew/Library/Homebrew/resource.rb:114:in `block (2 levels) in unpack'
/opt/homebrew/Library/Homebrew/download_strategy.rb:115:in `chdir'
/opt/homebrew/Library/Homebrew/download_strategy.rb:115:in `chdir'
/opt/homebrew/Library/Homebrew/download_strategy.rb:102:in `stage'
/opt/homebrew/Library/Homebrew/resource.rb:110:in `block in unpack'
/opt/homebrew/Library/Homebrew/mktemp.rb:90:in `block in run'
/opt/homebrew/Library/Homebrew/mktemp.rb:90:in `chdir'
/opt/homebrew/Library/Homebrew/mktemp.rb:90:in `run'
/opt/homebrew/Library/Homebrew/resource.rb:237:in `stage_resource'
/opt/homebrew/Library/Homebrew/resource.rb:109:in `unpack'
/opt/homebrew/Library/Homebrew/resource.rb:83:in `stage'
/opt/homebrew/Library/Homebrew/vendor/portable-ruby/3.3.6/lib/ruby/3.3.0/forwardable.rb:240:in `stage'
/opt/homebrew/Library/Homebrew/formula.rb:3324:in `stage'
/opt/homebrew/Library/Homebrew/formula.rb:1616:in `brew'
/opt/homebrew/Library/Homebrew/build.rb:133:in `block in install'
/opt/homebrew/Library/Homebrew/extend/kernel.rb:539:in `with_env'
/opt/homebrew/Library/Homebrew/build.rb:125:in `install'
/opt/homebrew/Library/Homebrew/build.rb:232:in `<main>'
~~~

Any idea what the issue could be? And how to complete the installation on my system?
Thank you in advance!

codevilot, 2024-12-30:
In my case, I installed Xcode and tried again, and it worked.

mbampi, 2024-12-30:
The solution was just reinstalling Xcode, as suggested by Namheon Baek.

I did it by executing the following:

`sudo rm -rf /Library/Developer/CommandLineTools`

`xcode-select --install`

Working fine now. Thanks ✅

closed 2024-12-30 by mbampi