osrf/homebrew-simulation #2927: macos15.2 m1pro

ngducmiinh, 2025-01-02:
/opt/homebrew/Cellar/qwt-qt5/6.3.0: 2,213 files, 33.3MB
==> Installing osrf/simulation/gazebo11
==> Patching
==> Applying brew_python_fix.patch
==> cmake .. -DQWT_WIN_INCLUDE_DIR=/opt/homebrew/opt/qwt-qt5/lib/qwt.framework/Headers -DQWT_WIN_LIBRARY_DIR=/
==> make install
Last 15 lines from /Users/ngducmiinh/Library/Logs/Homebrew/gazebo11/02.make:
In file included from /tmp/gazebo11-20250102-91305-tpbco6/gazebo-11.15.0/plugins/rest_web/RestUiWidget.hh:27:
In file included from /tmp/gazebo11-20250102-91305-tpbco6/gazebo-11.15.0/gazebo/gazebo.hh:20:
In file included from /tmp/gazebo11-20250102-91305-tpbco6/gazebo-11.15.0/gazebo/gazebo_core.hh:21:
In file included from /tmp/gazebo11-20250102-91305-tpbco6/gazebo-11.15.0/build/gazebo/transport/transport.hh:5:
/tmp/gazebo11-20250102-91305-tpbco6/gazebo-11.15.0/gazebo/transport/IOManager.hh:45:28: error: no type named 'io_service' in namespace 'boost::asio'
   45 |       public: boost::asio::io_service &GetIO();
      |               ~~~~~~~~~~~~~^
2 errors generated.
make[2]: *** [plugins/rest_web/CMakeFiles/RestUiPlugin.dir/RestUiLoginDialog.cc.o] Error 1
2 errors generated.
make[2]: *** [plugins/rest_web/CMakeFiles/RestUiPlugin.dir/RestUiPlugin.cc.o] Error 1
2 errors generated.
make[2]: *** [plugins/rest_web/CMakeFiles/RestUiPlugin.dir/RestUiWidget.cc.o] Error 1
make[1]: *** [plugins/rest_web/CMakeFiles/RestUiPlugin.dir/all] Error 2
make: *** [all] Error 2

scpeters, 2025-01-08:
gazebo11 does not currently build with boost 1.87

* https://github.com/gazebosim/gazebo-classic/issues/3414
* https://github.com/osrf/homebrew-simulation/issues/2916

xdire, 2025-01-10:
@scpeters @ngducmiinh did you guys found any temporary solution to the problem? Trying to install gazebo11 on m3 and this error basically blocks me out.

scpeters, 2025-01-10:
I'm going to close this as a duplicate of #2916 to move discussion to that issue to avoid repeating things in multiple places

closed 2025-01-10 by scpeters