osrf/homebrew-simulation #2984: Failed to build B2 build engine when installing boost@1.85.0 and gazebo11 on macOS (Apple M1)

EasonHU507, 2025-02-16:
Dear osrf/homebrew-simulation team,
I am encountering an issue when trying to install osrf/simulation/boost@1.85.0 and subsequently osrf/simulation/gazebo11 on my macOS system. The installation consistently fails with the error "Failed to build B2 build engine".
System Information:
MacBook Pro Model: MacBook Pro (2020, Apple M1)
macOS Version: macOS (Please specify your exact version - I entered 15.3 but this might be incorrect. Please confirm your version, e.g., macOS Ventura 13.x or macOS Sonoma 14.x)
Homebrew Version: Homebrew 4.4.20
Xcode Command Line Tools: Installed and up to date (confirmed via xcode-select --install).
Problem Description:
When attempting to install osrf/simulation/boost@1.85.0 (either directly or as a dependency of gazebo11), the build process fails during the Boost.Build (B2) build engine compilation with the following error:
MacBook-Pro:~ eason$ brew install osrf/simulation/boost@1.85.0 --build-from-source --cc=gcc-14
==> Auto-updating Homebrew...
Adjust how often this is run with HOMEBREW_AUTO_UPDATE_SECS or disable with
HOMEBREW_NO_AUTO_UPDATE. Hide these hints with HOMEBREW_NO_ENV_HINTS (see `man brew`).
Warning: You passed `--cc=gcc-14`.
It is expected behaviour that some formulae will fail to build in this unsupported configuration.
It is expected behaviour that Homebrew will be buggy and slow.
Do not create any issues about this on Homebrew's GitHub repositories.
Do not create any issues even if you think this message is unrelated.
Any opened issues will be immediately closed without response.
Do not ask for help from Homebrew or its maintainers on social media.
You may ask for help in Homebrew's discussions but are unlikely to receive a response.
Try to figure out the problem yourself and submit a fix as a pull request.
We will review it but may or may not accept it.

==> Fetching osrf/simulation/boost@1.85.0
==> Downloading https://github.com/boostorg/boost/releases/download/boost-1.85.0/boost-1.85.0-b2-nodocs.tar.xz
Already downloaded: /Users/eason/Library/Caches/Homebrew/downloads/b89b260e2b1f089e89ffc282d37ada0e1e6b1e969eea343f304bb8631cb58a8d--boost-1.85.0-b2-nodocs.tar.xz
==> Installing boost@1.85.0 from osrf/simulation
==> ./bootstrap.sh --prefix=/opt/homebrew/Cellar/boost@1.85.0/1.85.0 --libdir=/opt/homebrew/Cellar/boost@1.85.0/1.
==> ./b2 headers
Last 15 lines from /Users/eason/Library/Logs/Homebrew/boost@1.85.0/02.b2:
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

'
/private/tmp/boostA1.85.0-20250216-97073-2xgh5v/boost-1.85.0/tools/build/src/tools/clang-linux.jam:112: in get-full-version from module clang-linux
/private/tmp/boostA1.85.0-20250216-97073-2xgh5v/boost-1.85.0/tools/build/src/tools/clang-linux.jam:117: in clang-linux.get-short-version from module clang-linux
/private/tmp/boostA1.85.0-20250216-97073-2xgh5v/boost-1.85.0/tools/build/src/tools/clang-darwin.jam:71: in get-short-version from module clang-darwin
/private/tmp/boostA1.85.0-20250216-97073-2xgh5v/boost-1.85.0/tools/build/src/tools/clang-darwin.jam:48: in clang-darwin.init from module clang-darwin
/private/tmp/boostA1.85.0-20250216-97073-2xgh5v/boost-1.85.0/tools/build/src/build/toolset.jam:42: in toolset.using from module toolset
/private/tmp/boostA1.85.0-20250216-97073-2xgh5v/boost-1.85.0/tools/build/src/tools/clang.jam:28: in clang.init from module clang
/private/tmp/boostA1.85.0-20250216-97073-2xgh5v/boost-1.85.0/tools/build/src/build/toolset.jam:42: in toolset.using from module toolset
/private/tmp/boostA1.85.0-20250216-97073-2xgh5v/boost-1.85.0/tools/build/src/build/project.jam:1283: in using from module project-rules
project-config.jam:12: in modules.load from module project-config
/private/tmp/boostA1.85.0-20250216-97073-2xgh5v/boost-1.85.0/tools/build/src/build-system.jam:255: in load-config from module build-system
/private/tmp/boostA1.85.0-20250216-97073-2xgh5v/boost-1.85.0/tools/build/src/build-system.jam:486: in load-configuration-files from module build-system
/private/tmp/boostA1.85.0-20250216-97073-2xgh5v/boost-1.85.0/tools/build/src/build-system.jam:607: in module scope from module build-system

Do not report this issue to Homebrew/brew or Homebrew/homebrew-core!

Do not report this issue: you are running in an unsupported configuration.
MacBook-Pro:~ eason$ export CC=/opt/homebrew/bin/gcc-14
MacBook-Pro:~ eason$ export CXX=/opt/homebrew/bin/g++-14
MacBook-Pro:~ eason$ brew install osrf/simulation/boost@1.85.0 --build-from-source
==> Fetching osrf/simulation/boost@1.85.0
==> Downloading https://github.com/boostorg/boost/releases/download/boost-1.85.0/boost-1.85.0-b2-nodocs.tar.xz
Already downloaded: /Users/eason/Library/Caches/Homebrew/downloads/b89b260e2b1f089e89ffc282d37ada0e1e6b1e969eea343f304bb8631cb58a8d--boost-1.85.0-b2-nodocs.tar.xz
==> Installing boost@1.85.0 from osrf/simulation
==> ./bootstrap.sh --prefix=/opt/homebrew/Cellar/boost@1.85.0/1.85.0 --libdir=/opt/homebrew/Cellar/boost@1.85.0/1.
Last 15 lines from /Users/eason/Library/Logs/Homebrew/boost@1.85.0/01.bootstrap.sh:

Options:
    --help                  Show this help message.
    --verbose               Show messages about what this script is doing.
    --debug                 Build b2 with debug information, and no
                            optimizations.
    --guess-toolset         Print the toolset we can detect for building.
    --cxx=CXX               The compiler exec to use instead of the detected
                            compiler exec.
    --cxxflags=CXXFLAGS     The compiler flags to use in addition to the
                            flags for the detected compiler.

Failed to build B2 build engine

This error prevents the installation of Boost and subsequently Gazebo 11.
Attempted Solutions:
I have tried the following approaches to resolve the issue, but none have been successful:
Attempted to specify GCC compiler using --toolset option:       brew install osrf/simulation/boost@1.85.0 --build-from-source --toolset=gcc --cxx=gcc-14 --cxxflags="-std=c++11"
Result: Error: invalid option: --toolset=gcc

Attempted to specify GCC compiler using --cc option and --cxxflags:       brew install osrf/simulation/boost@1.85.0 --build-from-source --cc=gcc-14 --cxxflags="-std=c++11"
Result: Error: invalid option: --cxxflags=-std=c++11

Attempted to specify GCC compiler using --cc option only:       brew install osrf/simulation/boost@1.85.0 --build-from-source --cc=gcc-14
Result: Still "Failed to build B2 build engine" error (same as original error).

Attempted to force GCC compiler via environment variables and install without compiler options:       export CC=/opt/homebrew/bin/gcc-14
export CXX=/opt/homebrew/bin/g++-14
brew install osrf/simulation/boost@1.85.0 --build-from-source
Result: Still "Failed to build B2 build engine" error (same as original error).

Observations:
The brew install --help output indicates that --toolset and --cxxflags are not valid options for brew install.
The --cc=gcc-14 option, while seemingly accepted by brew install, does not appear to resolve the B2 build failure, and the error logs still suggest that Clang is being used during the initial B2 setup.
The error message "unsupported configuration" hints at a potential incompatibility between the Boost.Build system and my macOS environment, possibly related to compiler detection or toolchain setup.
Request:
Could you please investigate this "Failed to build B2 build engine" error when installing osrf/simulation/boost@1.85.0 on macOS Apple Silicon? Any guidance or workarounds to resolve this issue and successfully install Gazebo 11 would be greatly appreciated.
Thank you for your time and assistance.
Sincerely,
Eason

sudheer0404, 2025-03-12:
did you find the solution?

scpeters, 2025-03-26:
sorry, gazebo11 has reached end-of-life, so we don't have any plans to fix this. Please consider posting on https://robotics.stackexchange.com/

closed 2025-03-26 by scpeters