osrf/homebrew-simulation #3003: Compatibility with CMake < 3.5 has been removed from CMake.

atinfinity, 2025-04-02:
~~~
$ brew config
HOMEBREW_VERSION: 4.4.27
ORIGIN: https://github.com/Homebrew/brew
HEAD: 9b1efcd944408db0bd85f82fe4724ccae04a812f
Last commit: 3 days ago
Branch: stable
Core tap JSON: 02 Apr 15:41 UTC
Core cask tap JSON: 02 Apr 15:41 UTC
HOMEBREW_PREFIX: /opt/homebrew
HOMEBREW_CASK_OPTS: []
HOMEBREW_DISPLAY: /private/tmp/com.apple.launchd.qmCm6XaxQG/org.xquartz:0
HOMEBREW_MAKE_JOBS: 8
Homebrew Ruby: 3.3.7 => /opt/homebrew/Library/Homebrew/vendor/portable-ruby/3.3.7/bin/ruby
CPU: octa-core 64-bit arm_ibiza
Clang: 15.0.0 build 1500
Git: 2.39.3 => /Applications/Xcode.app/Contents/Developer/usr/bin/git
Curl: 8.4.0 => /usr/bin/curl
macOS: 14.4-arm64
CLT: 15.3.0.0.1.1708646388
Xcode: 15.4
Rosetta 2: false
~~~

~~~
brew tap osrf/simulation
brew install gz-harmonic
~~~

~~~
Last 15 lines from /Users/dandelion/Library/Logs/Homebrew/ogre1.9/01.cmake:
-DBUILD_TESTING=OFF
-DCMAKE_OSX_SYSROOT=/Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk
..

CMake Error at CMakeLists.txt:17 (cmake_minimum_required):
  Compatibility with CMake < 3.5 has been removed from CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

  Or, add -DCMAKE_POLICY_VERSION_MINIMUM=3.5 to try configuring anyway.

-- Configuring incomplete, errors occurred!

If reporting this issue please do so at (not Homebrew/brew or Homebrew/homebrew-core):
  https://github.com/osrf/homebrew-simulation/issues

These open issues may also help:
ogre1.9 - macOS 15.0 from source installation issues https://github.com/osrf/homebrew-simulation/issues/2793
~~~

<https://www.kitware.com/cmake-4-0-0-available-for-download/>

spark404, 2025-04-07:
This is due to CMAKE 4.x.x indeed. Compatibility for cmake version has changed to throw an error on a project that supports versions before 3.5. I've locally applied the following patch to orge1.9 and that seems to allow the package to install correctly.

```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 8413428..0d14a45 100755
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -14,7 +14,7 @@
 # and provide build configuration options.
 ######################################################################
 
-cmake_minimum_required(VERSION 2.6.2)
+cmake_minimum_required(VERSION 3.11)
 set(CMAKE_ALLOW_LOOSE_LOOP_CONSTRUCTS TRUE)
 cmake_policy(SET CMP0003 NEW)
 
```

and the change to `brew edit ogre1.9`:
```
  patch do
    url "file:///Users/whereever/ogre-cmake-patch.diff"
    sha256 "b6b47aa49bd668d561fff3c6a1472c436982a39a45fb603f6bc32391dd02a9cb"
  end
```

atinfinity, 2025-04-13:
@m0sviatoslav I apply <https://github.com/osrf/homebrew-simulation/issues/3003#issuecomment-2784275810>. As a result, I could install `gz-harmonic`.

And, `ignition-fortress` have same problem. I apply patch to `ogre1.9` and `ignition-tools`. As a result, I could install `gz-fortress`.

<details>

```bash
brew tap osrf/simulation
brew edit ogre1.9
```

And, I add the following description.

```rb
  patch do
    url "https://gist.githubusercontent.com/atinfinity/19911c781319ec471d52ad2525922cb6/raw/a2bca19b8156f38a00e1b0ea5cfd919d98413399/ogre-cmake-patch.diff"
    sha256 "466ae0277babc45d74066325d22742b491d104714ec91be7d90c3c0fdc417b67"
  end
```

```bash
brew edit ignition-tools
```

And, I add the following description.

```rb
  patch do
    url "https://gist.githubusercontent.com/atinfinity/7ebce5aedc90655445355cb1e1f6ba24/raw/5490dded4cf859d08e6bb29b41537041a714cc18/ignition-tools-cmake-patch.diff"
    sha256 "67a0654f0154f4c5311eeb62b33aea3c5bb98987eb9f44d8ab232364a989fb66"
  end
```

I installed `ignition-fortress`.

```bash
brew install ignition-fortress
```

</details>

m0sviatoslav, 2025-04-13:
@atinfinity tried it but it didn't work for me. Did you place the patch in some specific place? I tried to paste it before first and after the last patch in the file.

here is my error after `edit orge1.9`:

```
==> make install
Last 15 lines from /Users/m0sviatoslav/Library/Logs/Homebrew/ogre1.9/02.make:
/Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk/usr/include/c++/v1/string:2912:3: error: implicit instantiation of undefined template 'std::char_traits<unsigned short>'
 2912 |   traits_type::assign(*__p, __c);
      |   ^
/private/tmp/ogre1.9-20250413-25832-hk2zbm/sinbad-ogre-108ab0bcc696/OgreMain/src/OgreUTFString.cpp:705:9: note: in instantiation of member function 'std::basic_string<unsigned short>::push_back' requested here
  705 |                 mData.push_back( static_cast<code_point>( val ) );
      |                       ^
/Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk/usr/include/c++/v1/__fwd/string.h:23:29: note: template is declared here
   23 | struct _LIBCPP_TEMPLATE_VIS char_traits;
      |                             ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make[2]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreUTFString.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [OgreMain/CMakeFiles/OgreMain.dir/all] Error 2
make: *** [all] Error 2
```

atinfinity, 2025-04-14:
@m0sviatoslav I described the result of `brew config` in <https://github.com/osrf/homebrew-simulation/issues/3003#issue-2967775288>. Is your version of Homebrew the same as mine?

m0sviatoslav, 2025-04-14:
@atinfinity 
```
brew config
HOMEBREW_VERSION: 4.4.30-4-g9f49db1
ORIGIN: https://github.com/Homebrew/brew
HEAD: 9f49db135c0ab6651b6958de6fdfe956c7d4dcbd
Last commit: 2 hours ago
Branch: HEAD
Core tap JSON: 14 Apr 20:27 UTC
Core cask tap JSON: 14 Apr 20:27 UTC
HOMEBREW_PREFIX: /opt/homebrew
HOMEBREW_CASK_OPTS: []
HOMEBREW_DISPLAY: /private/tmp/com.apple.launchd.U0RG0N9y6o/org.xquartz:0
HOMEBREW_MAKE_JOBS: 8
HOMEBREW_SORBET_RUNTIME: set
Homebrew Ruby: 3.3.8 => /opt/homebrew/Library/Homebrew/vendor/portable-ruby/3.3.8/bin/ruby
CPU: octa-core 64-bit arm_firestorm_icestorm
Clang: 17.0.0 build 1700
Git: 2.39.5 => /Library/Developer/CommandLineTools/usr/bin/git
Curl: 8.7.1 => /usr/bin/curl
macOS: 15.4-arm64
CLT: 16.3.0.0.1.1742442376
Xcode: 16.3
Rosetta 2: false
```

without patch 
```
Last 15 lines from /Users/m0sviatoslav/Library/Logs/Homebrew/ogre1.9/01.cmake:
-DBUILD_TESTING=OFF
-DCMAKE_OSX_SYSROOT=/Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk
..

CMake Error at CMakeLists.txt:17 (cmake_minimum_required):
  Compatibility with CMake < 3.5 has been removed from CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

  Or, add -DCMAKE_POLICY_VERSION_MINIMUM=3.5 to try configuring anyway.

-- Configuring incomplete, errors occurred!

If reporting this issue please do so at (not Homebrew/brew or Homebrew/homebrew-core):
  https://github.com/osrf/homebrew-simulation/issues

These open issues may also help:
ogre1.9 - macOS 15.0 from source installation issues https://github.com/osrf/homebrew-simulation/issues/2793
```

after I added patch
```
Last 15 lines from /Users/m0sviatoslav/Library/Logs/Homebrew/ogre1.9/02.make:
/Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk/usr/include/c++/v1/string:2912:3: error: implicit instantiation of undefined template 'std::char_traits<unsigned short>'
 2912 |   traits_type::assign(*__p, __c);
      |   ^
/private/tmp/ogre1.9-20250414-19193-csy54z/sinbad-ogre-108ab0bcc696/OgreMain/src/OgreUTFString.cpp:705:9: note: in instantiation of member function 'std::basic_string<unsigned short>::push_back' requested here
  705 |                 mData.push_back( static_cast<code_point>( val ) );
      |                       ^
/Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk/usr/include/c++/v1/__fwd/string.h:23:29: note: template is declared here
   23 | struct _LIBCPP_TEMPLATE_VIS char_traits;
      |                             ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make[2]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreUTFString.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [OgreMain/CMakeFiles/OgreMain.dir/all] Error 2
make: *** [all] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/homebrew-core):
  https://github.com/osrf/homebrew-simulation/issues

These open issues may also help:
ogre1.9 - macOS 15.0 from source installation issues https://github.com/osrf/homebrew-simulation/issues/2793
```

DavidKalajdzic, 2025-04-18:
Got the same error

argenos, 2025-04-24:
I'm getting the same error as @m0sviatoslav after applying the patch from @spark404 suggested here: https://github.com/osrf/homebrew-simulation/issues/3003#issuecomment-2784275810

My brew config: 
```
HOMEBREW_VERSION: 4.4.32-46-gcecbb43
ORIGIN: https://github.com/Homebrew/brew
HEAD: cecbb4350eb45a65abf50781efc369f522fa472e
Last commit: 2 hours ago
Branch: master
Core tap JSON: 24 Apr 12:18 UTC
Core cask tap JSON: 24 Apr 12:18 UTC
HOMEBREW_PREFIX: /opt/homebrew
HOMEBREW_CASK_OPTS: []
HOMEBREW_EDITOR: vim
HOMEBREW_MAKE_JOBS: 12
HOMEBREW_SORBET_RUNTIME: set
Homebrew Ruby: 3.4.3 => /opt/homebrew/Library/Homebrew/vendor/portable-ruby/3.4.3/bin/ruby
CPU: dodeca-core 64-bit arm_lobos
Clang: 17.0.0 build 1700
Git: 2.47.0 => /opt/homebrew/bin/git
Curl: 8.7.1 => /usr/bin/curl
macOS: 15.4-arm64
CLT: 16.3.0.0.1.1742442376
Xcode: 16.3
Rosetta 2: false
```

jgillick, 2025-04-29:
After using [the suggested patch](https://github.com/osrf/homebrew-simulation/issues/3003#issuecomment-2784275810), I get the following error during installation:

```
Last 15 lines from /Users/jeremy/Library/Logs/Homebrew/ogre1.9/02.make:
In file included from /usr/local/include/boost/date_time/posix_time/posix_time_duration.hpp:15:
In file included from /usr/local/include/boost/numeric/conversion/cast.hpp:33:
In file included from /usr/local/include/boost/numeric/conversion/converter.hpp:13:
In file included from /usr/local/include/boost/numeric/conversion/conversion_traits.hpp:13:
In file included from /usr/local/include/boost/numeric/conversion/detail/conversion_traits.hpp:18:
In file included from /usr/local/include/boost/numeric/conversion/detail/int_float_mixture.hpp:19:
In file included from /usr/local/include/boost/mpl/integral_c.hpp:32:
/usr/local/include/boost/mpl/aux_/integral_wrapper.hpp:73:31: error: integer value -1 is outside the valid range of values [0, 3] for the enumeration type 'int_float_mixture_enum' [-Wenum-constexpr-conversion]
/usr/local/include/boost/mpl/aux_/static_cast.hpp:24:47: note: expanded from macro 'BOOST_MPL_AUX_STATIC_CAST'
   24 | #   define BOOST_MPL_AUX_STATIC_CAST(T, expr) static_cast<T>(expr)
      |                                               ^
2 errors generated.
make[2]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreAutoParamDataSource.cpp.o] Error 1
make[1]: *** [OgreMain/CMakeFiles/OgreMain.dir/all] Error 2
make: *** [all] Error 2
```

mrpollo, 2025-04-29:
I'm working on a PR to fix this issue, see #3029, I'm getting close but not yet done

metehansenyer, 2025-04-30:
Hi, I'm encountering a build failure when trying to install `gz-ionic`, specifically during the build of its dependency `gz-rendering9`. This seems related to the `ogre1.9` issues discussed here, but manifests as a linking error in the downstream package.

**Environment:**

* macOS: 15.4-arm64
* Xcode: 16.3
* Homebrew: 4.5.0

**Error:**

The build fails during `make install` for `gz-rendering9` with the following error (from `/Users/metehansenyer/Library/Logs/Homebrew/gz-rendering9/02.make`):

```
make[2]: *** No rule to make target `OGRE_LIBRARY_OgreMain-NOTFOUND', needed by `lib/libgz-rendering9-ogre.9.1.0.dylib'. Stop.
make[1]: *** [ogre/src/CMakeFiles/gz-rendering9-ogre.dir/all] Error 2
make: *** [all] Error 2
```

**Context:**

This error `OGRE_LIBRARY_OgreMain-NOTFOUND` suggests that while `ogre1.9` might install correctly (potentially with the patches being developed in #3029), the CMake configuration for the dependent package `gz-rendering9` is unable to locate the necessary `OgreMain` library from the installed `ogre1.9` (Updated with [Commit 1f98a09](https://github.com/mrpollo/homebrew-simulation/commit/1f98a0941e27c190b65731ea7c29ec0948e07ae0)).

This could indicate either:

1.  There's still an issue with how the patched `ogre1.9` installs or registers itself that affects downstream detection.
2.  There's a separate issue within `gz-rendering9`'s build system in finding `ogre1.9` on newer macOS/Xcode setups.

Could the fixes being worked on for `ogre1.9` also address how it's found by dependent packages, or should this be investigated separately in `gz-rendering9`?

I can provide the full `02.make` log if needed.

mrpollo, 2025-04-30:
@metehansenyer as far as I can tell the fixes in #3029 will address the linking issues as well, let's see once it get's merged.

Teddy13111995, 2025-04-30:
@mrpollo Please share the solution once #3029 is resolved.

Zezesi, 2025-05-02:
Last 15 lines from /Users/zexzexa/Library/Logs/Homebrew/ogre1.9-with-boost1.85/01.cmake:
-DBUILD_TESTING=OFF
-DCMAKE_OSX_SYSROOT=/Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk
..

CMake Error at CMakeLists.txt:17 (cmake_minimum_required):
  Compatibility with CMake < 3.5 has been removed from CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

  Or, add -DCMAKE_POLICY_VERSION_MINIMUM=3.5 to try configuring anyway.

-- Configuring incomplete, errors occurred!

Zezesi, 2025-05-02:
I have the same problem with

Homebrew 4.5.0-77-g93e0e85

Xcode 16.3
Build version 16E140

macOS Sequoia 15.4.1

MeethiKishori, 2025-05-07:
> This is due to CMAKE 4.x.x indeed. Compatibility for cmake version has changed to throw an error on a project that supports versions before 3.5. I've locally applied the following patch to orge1.9 [...]

Hi, Could you please help with the Url file name, is it Users/wherever or something specific to the file location, because i searched for the ogre-cmake-patch.diff, but i didn't find in my pc.

nofear, 2025-05-07:
This issue stems from the use of std::basic_string with unsigned short or unsigned char types, which are not standard-compliant. The stricter compiler in Xcode 16.2 no longer tolerates such usage.

A downgrade Xcode to 16.1 (Xcode and Command Line Tools) did the trick for me:

https://developer.apple.com/download/all/?q=xcode%2016.1

Teddy13111995, 2025-05-08:
@nofear
even after downloading and installing xcode16.1, I am getting same error. Here is my log:

(base) srija@MacBook-Air ~ % xcodebuild -version                              

Xcode 16.1
Build version 16B40
(base) srija@MacBook-Air ~ % brew tap osrf/simulation
brew install gz-ionic

==> Fetching dependencies for osrf/simulation/gz-ionic: osrf/simulation/ogre1.9, rapidjson, osrf/simulation/ogre2.3, osrf/simulation/gz-rendering9, osrf/simulation/gz-transport14, osrf/simulation/gz-gui9, google-benchmark, osrf/simulation/sdformat15, osrf/simulation/gz-physics8, osrf/simulation/gz-sensors9, osrf/simulation/gz-sim9 and osrf/simulation/gz-launch8
==> Fetching osrf/simulation/ogre1.9
==> Downloading https://gist.githubusercontent.com/marvinosswald/0e981e918aeee44
Already downloaded: /Users/srija/Library/Caches/Homebrew/downloads/812d1904a40fe1c469fe6d0846788e981df3e0b14378c7eff5c6f4d0e2a0ace1--gistfile1.txt
==> Downloading https://gist.github.com/NikolausDemmel/2b11d1b49b35cd27a102/raw/
Already downloaded: /Users/srija/Library/Caches/Homebrew/downloads/e7c203b88a808ca752ebac27f8b00b295445f8651766e688eabac24ca91f2af8--fix-1.9-HEAD.diff
==> Downloading https://gist.githubusercontent.com/NikolausDemmel/927bd7bb3f14c1
Already downloaded: /Users/srija/Library/Caches/Homebrew/downloads/ed6176508d9571d70a8a8b9c812c24e9fa7e56e4bfb6284e836cbafbfaa9c98c--ogre1.9.patch
==> Downloading https://gist.githubusercontent.com/iche033/68349eebfc436e484b70e
Already downloaded: /Users/srija/Library/Caches/Homebrew/downloads/4cf7ce49769de5262fd9a6a9858ff48c9104aaa3873fdd80419964a928372486--ogre-1.9-ois.patch
==> Downloading https://gist.githubusercontent.com/iche033/b73766fac9ab3d628a79b
Already downloaded: /Users/srija/Library/Caches/Homebrew/downloads/decf335241bb718c38a96a2a2a70cfa096762c4beb4dc2f3e4a16f889ed36002--ogre-1.9-agl.patch
==> Downloading https://gist.githubusercontent.com/iche033/e0080a592c890cc9a4fce
Already downloaded: /Users/srija/Library/Caches/Homebrew/downloads/c15e631644dee75637c0cc48b026154a91280d8c057a25bfb1b3eda50e7499c8--ogre-1.9-cocoa_window_scale.patch
==> Downloading https://gist.githubusercontent.com/iche033/e2b152d9df080b21f71ba
Already downloaded: /Users/srija/Library/Caches/Homebrew/downloads/d170e4463cc8b76d13766d512b11bdab092bbf4407be160a479c718b9a0a099b--ogre-1.9-cxx_flags.patch
==> Downloading https://gist.githubusercontent.com/iche033/2b5e2ba31436881f1db29
Already downloaded: /Users/srija/Library/Caches/Homebrew/downloads/685334483037480ddd52ca2f1dbad6eb35a573818150b82cc2de714cb99becaf--ogre-1.9-findogre.patch
==> Downloading https://github.com/OGRECave/ogre/commit/cade48b4c2215dd77ea74b7a
Already downloaded: /Users/srija/Library/Caches/Homebrew/downloads/d1d1b1546f3d7d50f8a0a37c28e7221c498c83ce5db8a239d770b7dc74bd1633--cade48b4c2215dd77ea74b7aa219a22c2a5d6654.patch
==> Downloading https://github.com/OGRECave/ogre/commit/2371c8d001725190a9cda62d
Already downloaded: /Users/srija/Library/Caches/Homebrew/downloads/b9b188483f25c506521c31da22540bd8c596fc9c38039cf0c96375a271f391b3--2371c8d001725190a9cda62dc5df282cde78f951.patch
==> Downloading https://github.com/scpeters/ogre/commit/8fcfe0885e6bd98d971250d4
Already downloaded: /Users/srija/Library/Caches/Homebrew/downloads/92546ae2ac08eb68f3076eaff910d6f37892a71d4035ead212b1f827d4d9e3a3--8fcfe0885e6bd98d971250d4a3fb9a23e3743dfd.patch
==> Downloading https://github.com/scpeters/ogre/commit/c39b6df49618773357da0b54
Already downloaded: /Users/srija/Library/Caches/Homebrew/downloads/32a0b2d5366cc6e476bfee9f50ec7207164ead6287ce326019beaa38b875eed9--c39b6df49618773357da0b54437f351b57168d12.patch
==> Downloading https://osrf-distributions.s3.amazonaws.com/ogre/releases/sinbad
Already downloaded: /Users/srija/Library/Caches/Homebrew/downloads/af051d2d04619fdb9791203d7d4d20e63622bbd39401ccf50e3df5ff4bdebff4--sinbad-ogre-108ab0bcc696.tar.bz2
==> Fetching rapidjson
==> Downloading https://ghcr.io/v2/homebrew/core/rapidjson/manifests/1.1.0-3
Already downloaded: /Users/srija/Library/Caches/Homebrew/downloads/a1862c9bc81880b766f7b3201eb610f09cf7c460a95e459bf3bc395ce76d5db0--rapidjson-1.1.0-3.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/rapidjson/blobs/sha256:ba505c9a
Already downloaded: /Users/srija/Library/Caches/Homebrew/downloads/dd764bd637a69ee70dd63e547d66cd83ed5b95afcd4640cd5616e6040f82dc75--rapidjson--1.1.0.all.bottle.3.tar.gz
==> Fetching osrf/simulation/ogre2.3
==> Downloading https://github.com/scpeters/ogre-next/commit/b7439ae047489aa104a
Already downloaded: /Users/srija/Library/Caches/Homebrew/downloads/12aa07ae3e0419f6b40ce55a0baeb1ac0e8dcdafbbfdb5170e56ed13b93f90ec--b7439ae047489aa104a6775a99a9e93294c3d5b5.patch
==> Downloading https://github.com/OGRECave/ogre-next/archive/refs/tags/v2.3.1.t
Already downloaded: /Users/srija/Library/Caches/Homebrew/downloads/934d985a51c822920a6ba97a12c2c0c3cb158c69da77b31188e86eb50f4075e4--ogre-next-2.3.1.tar.gz
==> Fetching osrf/simulation/gz-rendering9
==> Downloading https://osrf-distributions.s3.amazonaws.com/gz-rendering/release
Already downloaded: /Users/srija/Library/Caches/Homebrew/downloads/aecbe2e94d1fcca05729d0979566779d9cd27fe697c7ee4ca04c48be7d776a2a--gz-rendering-9.1.0.tar.bz2
==> Fetching osrf/simulation/gz-transport14
==> Downloading https://osrf-distributions.s3.amazonaws.com/gz-transport/release
Already downloaded: /Users/srija/Library/Caches/Homebrew/downloads/eb0140dae136afeb2a067f4b955cd7d8c384ec0a89b5f5e04bf4491a2d9fcb92--gz-transport-14.0.1.tar.bz2
[... cut ...]
  Or, add -DCMAKE_POLICY_VERSION_MINIMUM=3.5 to try configuring anyway.

-- Configuring incomplete, errors occurred!

If reporting this issue please do so at (not Homebrew/* repositories):
  https://github.com/osrf/homebrew-simulation/issues

These open issues may also help:
ogre1.9: Compatibility with CMake < 3.5 has been removed from CMake. https://github.com/osrf/homebrew-simulation/issues/3003
ogre1.9 - macOS 15.0 from source installation issues https://github.com/osrf/homebrew-simulation/issues/2793

nofear, 2025-05-08:
@Teddy13111995 
I overlooked that my local code included a fix for the compatibility issue with the CMAKE version:

- "brew edit ogre1.9"
- add  "-DCMAKE_POLICY_VERSION_MINIMUM=3.11," to the cmake_args.

```
cmake_args = [
  "-DCMAKE_POLICY_VERSION_MINIMUM=3.11",
  "-DCMAKE_CXX_STANDARD='14'",
  "-DOGRE_BUILD_DOCS:BOOL=FALSE",
  "-DOGRE_BUILD_LIBS_AS_FRAMEWORKS=OFF",
  "-DOGRE_BUILD_SAMPLES:BOOL=FALSE",
  "-DOGRE_FULL_RPATH:BOOL=FALSE",
  "-DOGRE_INSTALL_DOCS:BOOL=FALSE",
  "-DOGRE_INSTALL_SAMPLES:BOOL=FALSE",
  "-DOGRE_INSTALL_SAMPLES_SOURCE:BOOL=FALSE",
]
```

It's all a bit 'hacky' atm, but I just wanted to get it to work. Based on what I've seen so far on this issue it's probably not too difficult to fix, it's just a matter finding some time and figuring out how the PR / release process is organized for this project.

Teddy13111995, 2025-05-08:
@nofear
I applied the changes and still had issues. Here is the log:

Installing osrf/simulation/gz-ionic dependency: osrf/simulation/ogre1.9
==> Patching
==> Applying gistfile1.txt
==> Applying fix-1.9-HEAD.diff
==> Applying ogre1.9.patch
==> Applying ogre-1.9-ois.patch
==> Applying ogre-1.9-agl.patch
==> Applying ogre-1.9-cocoa_window_scale.patch
==> Applying ogre-1.9-cxx_flags.patch
==> Applying ogre-1.9-findogre.patch
==> Applying cade48b4c2215dd77ea74b7aa219a22c2a5d6654.patch
==> Applying 2371c8d001725190a9cda62dc5df282cde78f951.patch
==> Applying 8fcfe0885e6bd98d971250d4a3fb9a23e3743dfd.patch
==> Applying c39b6df49618773357da0b54437f351b57168d12.patch
==> cmake -DCMAKE_POLICY_VERSION_MINIMUM=3.11 -DCMAKE_CXX_STANDARD='14' -DOGRE_B
==> make install
Last 15 lines from /Users/srija/Library/Logs/Homebrew/ogre1.9/02.make:
   23 | struct _LIBCPP_TEMPLATE_VIS char_traits;
      |                             ^
In file included from /private/tmp/ogre1.9-20250508-19927-uextsd/sinbad-ogre-108ab0bcc696/OgreMain/src/OgreUTFString.cpp:28:
In file included from /private/tmp/ogre1.9-20250508-19927-uextsd/sinbad-ogre-108ab0bcc696/OgreMain/include/OgreStableHeaders.h:41:
In file included from /private/tmp/ogre1.9-20250508-19927-uextsd/sinbad-ogre-108ab0bcc696/OgreMain/include/OgreArchive.h:31:
In file included from /private/tmp/ogre1.9-20250508-19927-uextsd/sinbad-ogre-108ab0bcc696/OgreMain/include/OgrePrerequisites.h:32:
/Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk/usr/include/c++/v1/string:3290:16: error: incomplete definition of type 'std::char_traits<unsigned short>'
 3290 |     traits_type::assign(*__get_long_pointer(), value_type());
      |     ~~~~~~~~~~~^~
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make[2]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreUTFString.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [OgreMain/CMakeFiles/OgreMain.dir/all] Error 2
make: *** [all] Error 2

If reporting this issue please do so at (not Homebrew/* repositories):
  https://github.com/osrf/homebrew-simulation/issues

These open issues may also help:
ogre1.9: Compatibility with CMake < 3.5 has been removed from CMake. https://github.com/osrf/homebrew-simulation/issues/3003
ogre1.9 - macOS 15.0 from source installation issues https://github.com/osrf/homebrew-simulation/issues/2793

nofear, 2025-05-08:
@Teddy13111995 
I have seen this error too. To make 100% sure that I had Xcode 16.1 with its CommandLineTools (in the same version) and the SDK 15 (for that version). I did the following: 
- rm -rf /Library/Developer/CommandLineTools
- install the Xcode Command Line Tools 16.1
- remove the /Applications/Xcode.app
- add the downloaded 16.1 version.

Do 'xcodebuild -version' in a terminal, it should say
>Xcode 16.1
>Build version 16B40

I've tried many things (with the help of ChatGPT); like setting HOMEBREW_SDKROOT/SDKROOT or adding CMAKE_OSX_SYSROOT, use xcode-select, etc. It all failed. I think the key is that the build system is using the "/Library/Developer/CommandLineTools/" at the core and that needs to be the correct version with matching SDK.

commit b0b64169 (2025-05-09): ogre1.9: patch for latest clang (#3029)

macOS’s libc++ lacks std::char_traits for uint16 and unsigned int,
which breaks OgreUTFString’s use of basic_string<code_point>.  Inject
specializations for both uint16 (UTF-16) and uint32 (UTF-32) so that
std::basic_string<code_point> instantiates cleanly without touching
Ogre’s overloads or core logic.

Fixes #3003.

* ogre1.9: enable rpath
* ogre1.9: revision bump
* ogre1.9: update homepage
  Certificate has expired at ogre3d.org
* ignition-rendering6: revision bump for ogre1.9
* gz-rendering8: revision bump for ogre1.9
* gz-rendering9: revision bump [... cut]
closed 2025-05-09 by scpeters