osrf/homebrew-simulation #3014: Ogre installation error:  incomplete definition of type 'std::char_traits<unsigned short>'

NDHANA94, 2025-04-14:
HI, I am trying to install `gz-harmonic`. However, I have been facing this error and I cannot proceed the installation of `gz-harmonic`. 
</br>

Here is my `brew config`
~~~
$ brew config
HOMEBREW_VERSION: 4.4.30-2-g2664c05
ORIGIN: https://github.com/Homebrew/brew
HEAD: 2664c0568dc854e498ca257b8484a07fcfb0eae0
Last commit: 3 hours ago
Branch: master
Core tap JSON: 14 Apr 17:30 UTC
Core cask tap JSON: 14 Apr 17:30 UTC
HOMEBREW_PREFIX: /opt/homebrew
HOMEBREW_CASK_OPTS: []
HOMEBREW_DISPLAY: /private/tmp/com.apple.launchd.wglgVBsMi9/org.xquartz:0
HOMEBREW_EDITOR: vim
HOMEBREW_MAKE_JOBS: 11
HOMEBREW_SORBET_RUNTIME: set
Homebrew Ruby: 3.3.8 => /opt/homebrew/Library/Homebrew/vendor/portable-ruby/3.3.8/bin/ruby
CPU: 11-core 64-bit arm_lobos
Clang: 16.0.0 build 1600
Git: 2.39.5 => /Applications/Xcode.app/Contents/Developer/usr/bin/git
Curl: 8.7.1 => /usr/bin/curl
macOS: 15.4-arm64
CLT: 16.3.0.0.1.1742442376
Xcode: 16.0
Rosetta 2: false
~~~

This is the build log
~~~
brew install ogre2.3 --verbose

...

[ 41%] Building CXX object OgreMain/CMakeFiles/OgreMain.dir/src/OgreVertexShadowMapHelper.cpp.o
cd /private/tmp/ogre2.3-20250414-57853-nmprwt/ogre-next-2.3.1/build/OgreMain && /opt/homebrew/Library/Homebrew/shims/mac/super/clang++ -DFREEIMAGE_LIB -DOGRE_NONCLIENT_BUILD -DOgreMain_EXPORTS -D_MT -D_USRDLL -I/private/tmp/ogre2.3-20250414-57853-nmprwt/ogre-next-2.3.1/OgreMain/include -I/private/tmp/ogre2.3-20250414-57853-nmprwt/ogre-next-2.3.1/build/include -I/private/tmp/ogre2.3-20250414-57853-nmprwt/ogre-next-2.3.1/OgreMain/include/OSX -I/private/tmp/ogre2.3-20250414-57853-nmprwt/ogre-next-2.3.1/RenderSystems/Metal/include/Windowing/OSX -I/usr/local/include -I/usr/local/include/freetype2 -F/Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk/System/Library/Frameworks -I/private/tmp/ogre2.3-20250414-57853-nmprwt/ogre-next-2.3.1 -I/private/tmp/ogre2.3-20250414-57853-nmprwt/ogre-next-2.3.1/OgreMain/include/Threading -I/private/tmp/ogre2.3-20250414-57853-nmprwt/ogre-next-2.3.1/OgreMain/src/nedmalloc -Wall -Winit-self -Wno-overloaded-virtual -Wcast-qual -Wwrite-strings -Wextra -Wno-unused-parameter -Wshadow -Wno-missing-field-initializers -Wno-long-long  -msse -msse2 -O3 -DNDEBUG -std=gnu++11 -arch arm64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.0.sdk -fPIC -DOGRE_GCC_VISIBILITY -fvisibility=hidden -fvisibility-inlines-hidden -MD -MT OgreMain/CMakeFiles/OgreMain.dir/src/OgreVertexShadowMapHelper.cpp.o -MF CMakeFiles/OgreMain.dir/src/OgreVertexShadowMapHelper.cpp.o.d -o CMakeFiles/OgreMain.dir/src/OgreVertexShadowMapHelper.cpp.o -c /private/tmp/ogre2.3-20250414-57853-nmprwt/ogre-next-2.3.1/OgreMain/src/OgreVertexShadowMapHelper.cpp
In file included from /private/tmp/ogre2.3-20250414-57853-nmprwt/ogre-next-2.3.1/OgreMain/src/OgreUTFString.cpp:28:
In file included from /private/tmp/ogre2.3-20250414-57853-nmprwt/ogre-next-2.3.1/OgreMain/include/OgreStableHeaders.h:44:
In file included from /private/tmp/ogre2.3-20250414-57853-nmprwt/ogre-next-2.3.1/OgreMain/include/OgreCommon.h:32:
In file included from /private/tmp/ogre2.3-20250414-57853-nmprwt/ogre-next-2.3.1/OgreMain/include/OgrePrerequisites.h:32:
/Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk/usr/include/c++/v1/string:821:42: error: implicit instantiation of undefined template 'std::char_traits<unsigned short>'
  821 |   static_assert(is_same<_CharT, typename traits_type::char_type>::value,
      |                                          ^
/private/tmp/ogre2.3-20250414-57853-nmprwt/ogre-next-2.3.1/OgreMain/include/OgreUTFString.h:235:13: note: in instantiation of template class 'std::basic_string<unsigned short>' requested here
  235 |             dstring::iterator mIter;
      |             ^
/Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk/usr/include/c++/v1/__fwd/string.h:23:29: note: template is declared here
   23 | struct _LIBCPP_TEMPLATE_VIS char_traits;
      |                             ^
In file included from /private/tmp/ogre2.3-20250414-57853-nmprwt/ogre-next-2.3.1/OgreMain/src/OgreUTFString.cpp:28:
In file included from /private/tmp/ogre2.3-20250414-57853-nmprwt/ogre-next-2.3.1/OgreMain/include/OgreStableHeaders.h:44:
In file included from /private/tmp/ogre2.3-20250414-57853-nmprwt/ogre-next-2.3.1/OgreMain/include/OgreCommon.h:32:
In file included from /private/tmp/ogre2.3-20250414-57853-nmprwt/ogre-next-2.3.1/OgreMain/include/OgrePrerequisites.h:32:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk/usr/include/c++/v1/string:648:
/Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk/usr/include/c++/v1/string_view:300:42: error: implicit instantiation of undefined template 'std::char_traits<unsigned short>'
  300 |   static_assert(is_same<_CharT, typename traits_type::char_type>::value,
      |                                          ^
/Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk/usr/include/c++/v1/__type_traits/is_convertible.h:22:102: note: in instantiation of template class 'std::basic_string_view<unsigned short>' requested here
   22 | struct _LIBCPP_TEMPLATE_VIS is_convertible : public integral_constant<bool, __is_convertible(_T1, _T2)> {};
      |                                                                                                      ^
/Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk/usr/include/c++/v1/string:746:29: note: in instantiation of template class 'std::is_convertible<const unsigned short *const &, std::basic_string_view<unsigned short>>' requested here
  746 |     : public _BoolConstant< is_convertible<const _Tp&, basic_string_view<_CharT, _Traits> >::value &&
      |                             ^
/Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk/usr/include/c++/v1/string:1543:38: note: in instantiation of template class 'std::__can_be_converted_to_string_view<unsigned short, std::char_traits<unsigned short>, const unsigned short *>' requested here
 1543 |   template <class _Tp, __enable_if_t<__can_be_converted_to_string_view<_CharT, _Traits, _Tp>::value, int> = 0>
      |                                      ^
/Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk/usr/include/c++/v1/string:1545:3: note: while substituting prior template arguments into non-type template parameter [with _Tp = const code_point *]
 1545 |   insert(size_type __pos1, const _Tp& __t) {
[... 192 lines cut ...]
Clang: 16.0.0 build 1600
Git: 2.39.5 => /Applications/Xcode.app/Contents/Developer/usr/bin/git
Curl: 8.7.1 => /usr/bin/curl
macOS: 15.4-arm64
CLT: 16.3.0.0.1.1742442376
Xcode: 16.0
Rosetta 2: false

HOMEBREW_CC: clang
HOMEBREW_CXX: clang++
MAKEFLAGS: -j11
CMAKE_PREFIX_PATH: /opt/homebrew
CMAKE_INCLUDE_PATH: /Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk/System/Library/Frameworks/OpenGL.framework/Versions/Current/Headers
CMAKE_LIBRARY_PATH: /Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk/System/Library/Frameworks/OpenGL.framework/Versions/Current/Libraries
PKG_CONFIG_PATH: /opt/homebrew/opt/libpng/lib/pkgconfig:/opt/homebrew/opt/freetype/lib/pkgconfig:/opt/homebrew/opt/libxau/lib/pkgconfig:/opt/homebrew/opt/libxdmcp/lib/pkgconfig:/opt/homebrew/opt/libxcb/lib/pkgconfig:/opt/homebrew/opt/libx11/lib/pkgconfig:/opt/homebrew/opt/libzzip/lib/pkgconfig:/opt/homebrew/opt/rapidjson/lib/pkgconfig:/opt/homebrew/opt/hwloc/lib/pkgconfig:/opt/homebrew/opt/tbb/lib/pkgconfig:/opt/homebrew/opt/xorgproto/share/pkgconfig
PKG_CONFIG_LIBDIR: /usr/lib/pkgconfig:/opt/homebrew/Library/Homebrew/os/mac/pkgconfig/15
HOMEBREW_MAKE_JOBS: 11
HOMEBREW_GIT: git
HOMEBREW_SDKROOT: /Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk
ACLOCAL_PATH: /opt/homebrew/share/aclocal
PATH: /opt/homebrew/Library/Homebrew/shims/mac/super:/opt/homebrew/opt/cmake/bin:/opt/homebrew/opt/doxygen/bin:/opt/homebrew/opt/libpng/bin:/opt/homebrew/opt/freetype/bin:/opt/homebrew/opt/libzzip/bin:/opt/homebrew/opt/hwloc/bin:/usr/bin:/bin:/usr/sbin:/sbin

~~~

Has anyone solved this issue on MacOS 15 with M3 cpu?

scpeters, 2025-04-18:
I read a bit of https://github.com/facebook/react-native/issues/50411, which points to the [Deprecations section of the XCode 16.3 release notes](https://developer.apple.com/documentation/xcode-release-notes/xcode-16_3-release-notes#Deprecations):

> The base template for std::char_traits has been removed. If you are using std::char_traits with types other than char, wchar_t, char8_t, char16_t, char32_t or a custom character type for which you specialized std::char_traits, your code will stop working. The Standard does not mandate that a base template is provided, and such a base template is bound to be incorrect for some types, which could previously cause unexpected behavior while going undetected.

I see that your command line tools (CLT) version is 16.3.0.0.1.1742442376, so I'm guessing this is the issue

scpeters, 2025-04-18:
I reported this to the ogre-next developers in https://github.com/OGRECave/ogre-next/issues/519

scpeters, 2025-04-18:
I'll see if the patch in https://github.com/OGRECave/ogre-next/commit/497650c461281275b16af27c175bb608c2768127 applies cleanly to our 2.3.1 formula

scpeters, 2025-04-18:
I've opened https://github.com/osrf/homebrew-simulation/pull/3020 with a patch that should add support for Xcode 16.3

can anyone try testing it?

arorasarthak, 2025-04-20:
Hey @scpeters, thanks a lot for your amazing work! I was able to test it succesfully with `gz-ionic` and `gz-harmonic` on M1 Pro. However, I still needed to patch ogre1.9 as done in [#3020](https://github.com/osrf/homebrew-simulation/pull/3020). Had to patch [OgrePlatform.h](https://github.com/OGRECave/ogre/blob/master/OgreMain/include/OgrePlatform.h) to disable OGRE_UNICODE_SUPPORT.

NDHANA94, 2025-04-21:
> I've opened https://github.com/osrf/homebrew-simulation/pull/3020 with a patch that should add support for Xcode 16.3
> 
> can anyone try testing it?

Thanks a lot, I'll try it soon and provide you a feedback

scpeters, 2025-04-21:
thanks for the testing @arorasarthak 

let's track fixing ogre1.9 in a separate issue. there is already a problem related to cmake support in #3003

PR #3020 (2025-04-21): ogre2.3: patch for XCode 16.3
Fixes #3014.
closed 2025-04-21 by scpeters