osrf/homebrew-simulation #3035: error while installing due to failed patch download (http error 429)

mrpollo, 2025-05-09:
Please supply your brew configuration:

~~~
$ brew config
HOMEBREW_VERSION: 4.5.1-32-gc1d29f6
ORIGIN: https://github.com/Homebrew/brew
HEAD: c1d29f65a5443175d7b490c386ebc240ddcef5bc
Last commit: 28 hours ago
Branch: master
Core tap HEAD: 61c0f04f537cd504969f2086a9bb14aaf60ea144
Core tap last commit: 7 weeks ago
Core tap JSON: 08 May 19:53 UTC
Core cask tap JSON: 08 May 19:53 UTC
HOMEBREW_PREFIX: /opt/homebrew
HOMEBREW_CASK_OPTS: []
HOMEBREW_DISPLAY: /private/tmp/com.apple.launchd.a6bkYeTPWe/org.xquartz:0
HOMEBREW_EDITOR: mvim
HOMEBREW_MAKE_JOBS: 14
HOMEBREW_SORBET_RUNTIME: set
Homebrew Ruby: 3.4.3 => /opt/homebrew/Library/Homebrew/vendor/portable-ruby/3.4.3/bin/ruby
CPU: 14-core 64-bit arm_palma
Clang: 17.0.0 build 1700
Git: 2.39.5 => /Library/Developer/CommandLineTools/usr/bin/git
Curl: 8.7.1 => /usr/bin/curl
macOS: 15.4.1-arm64
CLT: 16.3.0.0.1.1742442376
Xcode: 16.3
Rosetta 2: false
~~~

Error while trying to install `osrf/simulation/gz-harmonic`

~~~
Run brew install --build-from-source px4-sim-gazebo
  brew install --build-from-source px4-sim-gazebo
  shell: /bin/bash -e {0}
  env:
    pythonLocation: /Users/runner/hostedtoolcache/Python/3.11.9/arm64
    PKG_CONFIG_PATH: /Users/runner/hostedtoolcache/Python/3.11.9/arm64/lib/pkgconfig
    Python_ROOT_DIR: /Users/runner/hostedtoolcache/Python/3.11.9/arm64
    Python2_ROOT_DIR: /Users/runner/hostedtoolcache/Python/3.11.9/arm64
    Python3_ROOT_DIR: /Users/runner/hostedtoolcache/Python/3.11.9/arm64
    export HOMEBREW_GITHUB_API_TOKEN: ***
  
==> Tapping osrf/simulation
Cloning into '/opt/homebrew/Library/Taps/osrf/homebrew-simulation'...
Tapped 104 formulae (195 files, 5MB).
==> Tapping discoteq/discoteq
Cloning into '/opt/homebrew/Library/Taps/discoteq/homebrew-discoteq'...
Tapped 1 formula (14 files, 8.9KB).
==> Fetching dependencies for px4/px4/px4-sim-gazebo: exiftool, gflags, glog, gettext, highway, imath, libdeflate, openexr, webp, jpeg-xl, libvmaf, aom, libavif, gd, jasper, netpbm, gts, gdk-pixbuf, fribidi, pango, librsvg, m4, libtool, graphviz, dav1d, faac, faad2, fdk-aac, aribb24, frei0r, lame, libunibreak, libass, libbluray, cjson, libmicrohttpd, mbedtls, librist, libsoxr, libssh, libvidstab, libogg, libvorbis, libvpx, opencore-amr, openjpeg, opus, rav1e, libsamplerate, flac, mpg123, lib
==> Fetching exiftool
==> Downloading https://ghcr.io/v2/homebrew/core/exiftool/manifests/13.25
==> Downloading https://ghcr.io/v2/homebrew/core/exiftool/blobs/sha256:37c990988f852110496a00cf326f0219ec692e63bd611070d05c006980f048f9
==> Fetching gflags
==> Downloading https://ghcr.io/v2/homebrew/core/gflags/manifests/2.2.2-2
==> Downloading https://ghcr.io/v2/homebrew/core/gflags/blobs/sha256:1b14d0fd5ab4d2d04ff229bf7cace191208d62a3dc67029151ce1140ecf81258
==> Fetching glog
==> Downloading https://ghcr.io/v2/homebrew/core/glog/manifests/0.6.0
==> Downloading https://ghcr.io/v2/homebrew/core/glog/blobs/sha256:c6eb9b8ce678f03a87a9864ea498434b44a206cd331322ed771824a71320a97a
==> Fetching gettext
==> Downloading https://ghcr.io/v2/homebrew/core/gettext/manifests/0.25
==> Downloading https://ghcr.io/v2/homebrew/core/gettext/blobs/sha256:19c917a48a53614a6f6a611c018cc2884466268638d0aabb4c2f596684b94e9f
==> Fetching highway
==> Downloading https://ghcr.io/v2/homebrew/core/highway/manifests/1.2.0
==> Downloading https://ghcr.io/v2/homebrew/core/highway/blobs/sha256:406c96cf28555eb84e1c67788db50223a6af2fd488ce91e831068e60981d128a
==> Fetching imath
==> Downloading https://ghcr.io/v2/homebrew/core/imath/manifests/3.1.12
==> Downloading https://ghcr.io/v2/homebrew/core/imath/blobs/sha256:787f2525c5b53bb1153e265774555f2a39c350bf927ce32fdb675987db179776
==> Fetching libdeflate
==> Downloading https://ghcr.io/v2/homebrew/core/libdeflate/manifests/1.23
==> Downloading https://ghcr.io/v2/homebrew/core/libdeflate/blobs/sha256:15e665a6443b3f652cb920892a936cf09af93fb518c2771fde48211f3cb81a11
==> Fetching openexr
==> Downloading https://ghcr.io/v2/homebrew/core/openexr/manifests/3.3.3
==> Downloading https://ghcr.io/v2/homebrew/core/openexr/blobs/sha256:120c5a42eb3a3ef8f77336da362cd413fdb0d1a5d2afb9057ad141e55995a9bd
==> Fetching webp
==> Downloading https://ghcr.io/v2/homebrew/core/webp/manifests/1.5.0
==> Downloading https://ghcr.io/v2/homebrew/core/webp/blobs/sha256:121c8d526d62724c65ca82ad99a2edfa56bf7aeb6a69a231399e0faab96cad1f
==> Fetching jpeg-xl
==> Downloading https://ghcr.io/v2/homebrew/core/jpeg-xl/manifests/0.11.1_1
==> Downloading https://ghcr.io/v2/homebrew/core/jpeg-xl/blobs/sha256:d512c7b32e5c14337646f39e27187d4a241b8d7efaad2003b5cca543012c43fd
==> Fetching libvmaf
==> Downloading https://ghcr.io/v2/homebrew/core/libvmaf/manifests/3.0.0-1
==> Downloading https://ghcr.io/v2/homebrew/core/libzzip/blobs/sha256:38d51c19c574872633dda1817897c67f3834fbf9a8bbae9ad3afeaebdeb0e171
==> Fetching osrf/simulation/ogre1.9
==> Downloading https://gist.githubusercontent.com/marvinosswald/0e981e918aeee4403e9a0790c6f32cdd/raw/5d295499255867064ace8c4c41b2f31fd48d4576/gistfile1.txt
==> Downloading https://gist.github.com/NikolausDemmel/2b11d1b49b35cd27a102/raw/bf4a4d16020821218f73db0d56aa111ab2fde679/fix-1.9-HEAD.diff
==> Downloading from https://gist.githubusercontent.com/NikolausDemmel/2b11d1b49b35cd27a102/raw/bf4a4d16020821218f73db0d56aa111ab2fde679/fix-1.9-HEAD.diff
==> Downloading https://gist.githubusercontent.com/NikolausDemmel/927bd7bb3f14c1788599/raw/c9a5ba88b758e80d3f46511629c4e8026b92c462/ogre1.9.patch
==> Downloading https://gist.githubusercontent.com/iche033/68349eebfc436e484b70e6e3508ae27b/raw/d96227cd79ed1d63f051689a18c95a1c174a4efa/ogre-1.9-ois.patch
==> Downloading https://gist.githubusercontent.com/iche033/b73766fac9ab3d628a79b5ed986677cd/raw/878d0902704c7fb51511163052c95294361f1dbe/ogre-1.9-agl.patch
==> Downloading https://gist.githubusercontent.com/iche033/e0080a592c890cc9a4fce31f6863a5ed/raw/875ae8ad1d9f0eaa271fd44eab8e0979bac74119/ogre-1.9-cocoa_window_scale.patch
==> Downloading https://gist.githubusercontent.com/iche033/e2b152d9df080b21f71ba3b65aa39922/raw/5397fa149c7570c17f2d78421d44b12dfa175387/ogre-1.9-cxx_flags.patch
==> Downloading https://gist.githubusercontent.com/iche033/2b5e2ba31436881f1db29f9b60c7a5b2/raw/b6ab953ebd82127ad1177744f367a36e059312a9/ogre-1.9-findogre.patch
==> Downloading https://github.com/OGRECave/ogre/commit/cade48b4c2215dd77ea74b7aa219a22c2a5d6654.patch?full_index=1
==> Downloading https://github.com/OGRECave/ogre/commit/2371c8d001725190a9cda62dc5df282cde78f951.patch?full_index=1
==> Downloading https://github.com/scpeters/ogre/commit/8fcfe0885e6bd98d971250d4a3fb9a23e3743dfd.patch?full_index=1
==> Downloading https://github.com/scpeters/ogre/commit/c39b6df49618773357da0b54437f351b57168d12.patch?full_index=1
==> Downloading https://gist.githubusercontent.com/mrpollo/70a1fa1413d52227d0f9b5b2aa934442/raw/974f144e4c17d623f4bf90258876b38a90ef6018/ogre1.9-macos15-utfstring-fix.patch
==> Downloading https://osrf-distributions.s3.amazonaws.com/ogre/releases/sinbad-ogre-108ab0bcc696.tar.bz2
==> Fetching rapidjson
==> Downloading https://ghcr.io/v2/homebrew/core/rapidjson/manifests/1.1.0-3
==> Downloading https://ghcr.io/v2/homebrew/core/rapidjson/blobs/sha256:ba505c9a587b3c26539eae5dd732ae36bca53b67daf94579fee177d88094fa52
==> Fetching osrf/simulation/ogre2.3
==> Downloading https://github.com/scpeters/ogre-next/commit/b7439ae047489aa104a6775a99a9e93294c3d5b5.patch?full_index=1
curl: (56) The requested URL returned error: 429
Error: px4-sim-gazebo: Failed to download resource "ogre2.3--patch"
Download failed: https://github.com/scpeters/ogre-next/commit/b7439ae047489aa104a6775a99a9e93294c3d5b5.patch?full_index=1

Error: Process completed with exit code 1.
~~~

I have seen this fail in multiple formulas. Still, most consistently while downloading patches for Ogre 2.3, perhaps we should consider hosting patches in this repository to avoid requests with raw content (blob) access from GitHub, since those are not authenticated and going through their CDN, which are way more aggressive at enforcing rate limits, especially on CI runners.

scpeters, 2025-05-19:
related: https://github.blog/changelog/2025-05-08-updated-rate-limits-for-unauthenticated-requests/

scpeters, 2025-05-19:
one way this could be improved upstream at `brew` is to not `curl` if the patch is already downloaded. I prefer opening pull requests with potential fixes rather than bug reports to brew, so I'll wait on that until I have a chance to investigate it more specifically