osrf/homebrew-simulation #3182: CMake Error at CmakeLists.txt:1

Bartichaud, 2025-09-20:
**ERROR :** 

==> Patching
==> Applying enforce-use-stl.patch
==> Applying entity-encoding.patch
==> Applying tinyxml_CMakeLists.patch
==> cmake .
Last 15 lines from /Users/brthadm/Library/Logs/Homebrew/tinyxml1/01.cmake.log:
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /opt/homebrew/Library/Homebrew/shims/mac/super/clang++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 has been removed from CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

  Or, add -DCMAKE_POLICY_VERSION_MINIMUM=3.5 to try configuring anyway.

-- Configuring incomplete, errors occurred!

What should I do ???

scpeters, 2025-09-25:
The tinyxml1 formula is no longer maintained. It was used by gazebo-classic (gazebo11), but that is end-of-life. Do you know which package is depending on `tinyxml1`?

scpeters, 2025-09-29:
closing since this formula is deprecated

closed 2025-09-29 by scpeters